raspberrypi/linux #2631: hostapd.service not found

raspbitech, 2018-07-31:
I am trying to make a Raspberry Pi 2 as a wifi access point in a standalone network. 

I followed this tutorial: https://docs.google.com/document/d/16hJS7fnPyKUwbvR33_nMku_b6XcYaBJXVpeCIMoJzcQ/pub

At one point in the tutorial it teaches how to download a hostapd that works in particular with Edimax wifi dongles. I have bought one for this project. 

These are the commands I followed to install the hostapd:

sudo wget https://github.com/jenssegers/RTL8188-hostapd/archive/v2.0.tar.gz

tar -zxvf v2.0.tar.gz

cd RTL8188-hostapd-2.0/hostapd

sudo make install

sudo service hostapd restart

Everything was working perfectly fine except when I got to the last command. The cmd prompt was returning "hostapd.service not found"
I don't understand what is wrong. I repeated this process multiple times, and every time I got the same error message.

Any help is appreciated. :)

popcornmix, 2018-07-31:
Sorry, this issue tracker is for reporting kernel bugs.
For general debian configuration help I'd suggest to post on the [forum](https://www.raspberrypi.org/forums).

closed 2018-07-31 by popcornmix

raspbitech, 2018-07-31:
ok, tnx