raspberrypi/linux #2632: I2C detection issue since kernel i2c drivers changed for the kernel 4.4.14

lnjuanj, 2018-07-31:
I have a PIC that communicates via I2C correctly with any RPi with 4.4.13 firmware.

However, when I change to any firmware that is 4.4.14 or newer (up to the last one I installed, 4.14.52), I seldom detect the I2C address and cannot work with the device anymore. It looks like something changed with version 4.4.14 in the kernel. I have reviewed the firmware repo and there are no direct changes to i2c_dev nor to i2c_bcm2708 / i2c_bcm2*** (I thought I never detected the address, but using 

`watch -d=cumulative i2cdetect -y 1`

 shows the address once in a while e.g. 20 minutes).

As I was saying, the 4.4.14 firmware does not introduce code changes to i2c, but there are i2c_dev.ko and i2c_bcm2708.ko kernel drivers compiled in [that version](https://github.com/raspberrypi/firmware/commit/20c6fda4e1551e0cbd8af41d5aaeecf9604c2b26)

The commits in the kernel around the time the firmware was updated to 4.4.14 [can be found here](https://github.com/raspberrypi/linux/commits/rpi-4.4.y?after=04c8e47067d4873c584395e5cb260b4f170a99ea+2634)

Any help will be greatly appreciated!

lategoodbye, 2018-07-31:
Can you provide a logic analyzer trace of good and bad case?

lnjuanj, 2018-08-01:
Here's the logic analyzer for an i2cdetect with the address 0x15, the PIC's address:
[image]
It's the same with 4.4.13 and 4.14.52. Could the problem be that the PIC keeps the SDA line low for a while after the ACK and the RPi detects that as an error since 4.4.14?

lategoodbye, 2018-08-01:
Yellow (1) is SCK and Cyan (2) is SDA?
SCK is running at 100 kHz?

lnjuanj, 2018-08-08:
Hi,
As you can see in the oscilloscope, the device performs clock stretching on SDA right after it has ack'ed. May that be the cause of the lack of detection of the PIC? As far as I know, I cannot avoid that clock stretching with the Microchip PIC (as explained [on this site](http://www.extremeelectronics.co.uk/rpi-i2c-interface-embedded-clock-stretching/) ).
Thanks in advance for your help.

6by9, 2018-08-08:
(a) Clock stretching is not supported due to a fairly well documented and understood bug in the hardware. 

(b) We haven't been updating 4.4 for a long time, although it is a useful data point that 4.4.14ish is when things seemed to change. I'm not checking the history, but 4.4.14 sounds a bit early for us to have adopted 4.4 as the standard kernel. Are you building your own kernels or downloading them from somewhere?

(c) There was a switch from the downstream i2c_bcm2708 driver to the mainline i2c_bcm2835 driver. I can't recall when that happened and I'm not spending the time hunting it down, but it would be useful to confirm which driver you are using. ```lsmod``` will tell you.
i2c_bcm2835 will always combine a write/read sequence with a repeated start (see https://github.com/raspberrypi/firmware/issues/828). If i2c-detect is trying to do a read of your device then that change may well be upsetting it. With the latest kernel tree there is an option to switch back to the i2c_bcm2708 by adding ```dtoverlay=i2c1-bcm2708,combine=1```, and that will therefore not automatically combine the two transactions.

lnjuanj, 2018-08-08:
Hi,
About (b), I just do a rpi-update to the firmware version 4.4.13
About (c), fw versión 4.4.14 keeps i2c_bcm2708.
It looks like my problem is tightly related to (a).

macmpi, 2018-08-08:
@lnjuanj 
Do you have same problem if you do `i2cdetect` under `sudo`?

I experienced similar issues on a custom built Raspbian, and this got fixed once I properly set user access to i2c & gpio, as it is done in Foundation Raspbian release.
Things like `usermod -a -G gpio,i2c,spi,input <your group>` and  [install `raspberrypi-sys-mods`](https://github.com/RPi-Distro/raspberrypi-sys-mods)

lnjuanj, 2018-08-11:
Hi @macmpi and all,
We have solved the problem by re-implementing the I2C in the PIC. We had an expert who reviewed the code and some things have been changed. The problem was related to PICs bad responsiveness.
The code in the PIC was changed so it is more robust and responds much faster than before.
It didn't have anything to do with permissions and versions 4.4.14 and newer ones (i.e. latest one as of this writing) communicate with the PIC.
Why and how the functionality changed from firmware version 4.4.13 to 4.4.14 is something that is yet (and probably will be) elusive to me.

closed 2018-08-11 by lnjuanj

apatlpo, 2020-04-29:
@Injuanj, I realize this is an old problem but it would have been nice to see what had to be changed in I2C PIC implementation in order to make this work.

lnjuanj, 2020-07-27:
You are right @apatlpo : I should have told what I did. But it was hardware-related and it did not have anything to do with the rpi operating system.

The previous implementation relied on a constant polling of the line and we changed it to an interruption-based implementation. I cannot see the hardware (I do not work on that project anymore), so I cannot say for sure, though I think we sticked to i2c_bcm2708. Again, not 100% sure though.