raspberrypi/linux #2633: Connecting to a BT device using console tools

vanfanel, 2018-08-01:
Hi there,

I have this project where DBUS is not an option, but bluetooth connections are needed.
I was able to setup the BT hardware on the Pi by loading it's modules, and attaching it to hci:
```

sudo modprobe hci_uart
sudo modprobe btbcm
sudo hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -
```
I can see it's working without problems with:
`sudo hciconfig -a`

I can easily bring up the device like this:
`sudo hciconfig hci0 up`

I can even scan devices with `sudo hcitool scan`, and I can see the target devices in range. Great so far!

The problem comes when I try to connect the Pi3b+ to them. There used to be "hidd --connect <address>", but the hidd command has been deprecated in favor of some mumbo-jumbo using DBUS and systemd services. I guess a service is needed to manage connections, etc, but all I need here is to stablish a BT connection manually. Simple.
So, what should I use in place of "hidd --connect <address>" now? I have tried `sudo hcitool cc <address>`, but it doesn't stablish any connections (connections can be verified with `hcitool con`).

maxnet, 2018-08-05:
>I have this Pi project where DBUS is not an option

Don't think bluez works properly without dbus and bluetoothd daemon.
And hciattach/hcitool/hciconfig are also deprecated.

vanfanel, 2018-08-06:
@maxnet : There's no way I will believe a service is necessary to stablish a connection. It's as if other network connections would need services to be stablished, that would make no sense. A service would take care of restoring connection settings on boot, saving them on shutdown, etc
But there must be a command to stablish connections manually, or things are just gone an absurd path.

maxnet, 2018-08-06:
1) The htattach daemon that maintains the uart connection is depreceated, and not necessary if you run your own embedded OS.
You can let the kernel take care of setting that up for you. However the downstream kernel currently lacks the right device-tree entries for that.
So that currently requires switching to upstream kernel, or some manual work.

However that will not solve your current problem, so will not go into the details.

2) If the bluetooth device you want to talk to requires any kind of pairing.
You need the bluetoothd daemon, and that will not run without dbus daemon.

It is not that dissimilair to other type of network connections.
Think wpa_supplicant for wifi...

JamesH65, 2018-09-03:
@vanfanel Any furthur progress to report on this?

maxnet, 2018-09-03:
There is no dependency on systemd.
dbus != systemd