raspberrypi/linux #2635: mmc0 Timeout waiting for hardware interrupt

aqking, 2018-08-02:
I am now getting this error on **a new Pi3+ I just bought two weeks ago**.
The device ran fine, until I tried to add a new user (via PuTTY).     
Suddenly there was no response, so I plugged in a HDMI screen to see the console.
After a reboot,  the above message was issued, followed by
"DMA transfer could not complete...."  and a 100 or so lines of errors, until the system halted on a Kernel Panic.    
 I have re-written the Raspian Stretch ISO onto the only mini-SD card I have (8GB), and the above error still occurs.
Send the PI back to the dealer? (through Amazon) or is there some other trick or boot system that I can try?

brubbel, 2018-08-03:
You should first try a new sd-card.

JamesH65, 2018-08-03:
Agreed. This could happen if the card if defective or fake.

lategoodbye, 2018-08-03:
@aqking Could you please upload the kernel panic?

aqking, 2018-08-03:
**Thanx,guys!** I finally DID find a 16GB miniSD among my junk SD stuff.
This time I formatted it with the **recommended utility "SDCard Formatter"** and then installed NOOBS on it.
The Raspbian is now installed on it, and running!
When I have a chance, I'll re-format that fauty8GB SD the proper way, which I never knew about before.
**Thanx again!**

closed 2018-08-03 by aqking