raspberrypi/linux #2640: hangs on boot, kthread staved, reproducible

lesto, 2018-08-09:
Hello,
I'm having a strange issue (both raspian and arch linux). I have a LTE usb modem connected on the raspberry; if the modem is connected on boot, there are about 50% chance the boot will hang.
The boot will hang after initializing the Network Service. (see screenshot)
[image]

I am able to reproduce the issue, and i would like to fix it.
If you need more information I can spend some time debugging it.
[image]

JamesH65, 2018-08-09:
At a minimum we woul need to know what the device  model is, although whether we would have time to buy one and test it is debatable.

JamesH65, 2018-08-09:
Although a first question would be, do you have a decent power supply. These LTE modems can be quite power hungry.

lategoodbye, 2018-08-13:
Could you please provide a full kernel output (e.g. via serial adapter)?

JamesH65, 2018-09-03:
@lesto Is this still an issue? Does the device work on other Linux systems? e.g.  X86?

lesto, 2018-09-12:
didnt test on latest kernel, right now i don't have the hardware and the time, but the issue is not there on a all the desktop i tested