raspberrypi/linux #2641: Full support for CM3 needs to be added to raspberrypi3-64 target

stevep-korg, 2018-08-09:
The raspberrypi3-64 target does not support the CM3.  It should.  That way the de-facto 64-bit target can easily be built for and run on the industrial/commercial version of the R Pi 3B.

A few things are needed to implement this:
1. reinstate the entry for the CM3 dtb for the raspberrypi3-64 target in the device tree overlay list
2. add the proper dts source file for the CM3 under arch/arm64/boot/dts/broadcom
3. add any dtsi file(s) included by the CM3 .dts under arch/arm64/boot/dts/broadcom
4. test that the dtb will successfully build

I'd like to add a request to have this enhancement merged to sumo once it makes it to master.  I have been told that this can be achieved by a PR (Pull Request?).  I am not familiar with this process, but I'd like to initiate the steps to cause it to occur.  My organizations build in sumo branch as often as possible - due to its high stability and the time-consuming nature of a failed build.

pelwell, 2018-08-09:
The 64-bit builds are not officially supported, but a Pull Request making the changes you suggest is very likely be merged.

stevep-korg, 2018-08-10:
https://github.com/raspberrypi/linux/pull/2644

pelwell, 2018-08-12:
Now that #2644 is merged, is it okay to close this issue?

stevep-korg, 2018-08-12:
Yes, I'll close it now.  Thanks much for the merge.

closed 2018-08-12 by stevep-korg