raspberrypi/linux #2642: [4.18] Intermittent ethernet link after boot on 3B+

HiassofT, 2018-08-10:
When running the current rpi-4.18.y kernel on a 3B+ I often have no network after boot (both cold boot and reboots).

dmesg reports that no link could be detected
```
[    7.580991] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
```

Quite interestingly though the switches indicate a Gbit link - tested with a HP 1810-24G v2 (J9803A) and a Netgear GS108. I also tested witth the green features (EEE, auto port power down) disabled on the HP switch but that didn't make a difference.

ethtool doesn't quite agree with the status from the switch, it reports a 100Mb link detected - and the 1000Mb entry is missing in the "Link partner advertised link modes" list

```
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbag
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

On kernel 4.14 ethernet works fine with both switches and the 3B+ correctly detects/negotiates a Gbit link

HiassofT, 2018-08-10:
The 3rd commit is already in 4.18 (commit dea39aca1d7a), testing with the other 2 added didn't help much. Maybe it got a bit better but as the issue is intermittent it's hard to tell.

Re-plugging the network cable also didn't help much. ethtool now showed a Gbit connection, but even after manually configuring the interface (ifconfig eth0 192.168.1.140 up) I couldn't ping any device in my lan. dmesg also didn't indicate that link status changed.

What seems to help though is disabling EEE via config.txt. In about 10 reboot and cold boot attempts I always got a link.

Maybe some EEE related patch is missing?

HiassofT, 2018-08-11:
It looks like the issue is cured by adding commit c2eb30683b43 "lan78xx: Move enabling of EEE into PHY init code" which is missing in the rpi-4.18.y branch. It wasn't a clean pick (LED code below changed) but rather easy to resolve.

ATM I'm testing with this tree: https://github.com/HiassofT/rpi-linux/tree/rpi-4.18.y-lan78xx-test which adds the following 4.14 commits
c2eb30683b43 lan78xx: Move enabling of EEE into PHY init code
1c70d2de5453 lan78xx: Fix link status notifications
d407fc229cdc net: lan78xx: fix rx handling before first packet is send

@6by9 @pelwell could you check if more lan78xx commits are missing in the 4.18 tree and pick them if needed?

The "lan78xx: Move enabling of EEE into PHY init code" commit doesn't seem to be upstreamed yet (next-20180810 still has the EEE code block in lan78xx_open()

lategoodbye, 2018-08-12:
A folded version of "net: lan78xx: fix rx handling before first packet is send" has been merged into 4.18 after rc7. I hadn't the EEE patch on my radar and today 4.18 comes out :-(

popcornmix, 2018-08-15:
@HiassofT I've cherry picked the "Move enabling of EEE into PHY init code" to 4.18.
I believe the other commits may already be present upstream.
Can you test if our updated 4.18 tree has the issue?

HiassofT, 2018-08-20:
Thanks, ethernet seems to be working fine in current 4.18 tree!

closed 2018-08-20 by HiassofT

lategoodbye, 2018-08-27:
@pelwell The merge window has been closed, do you send "lan78xx: Move enabling of EEE into PHY init code" to linux-netdev?

pelwell, 2018-08-28:
No - not yet,