raspberrypi/linux #2645: vchiq_arm

ShorTie8, 2018-08-13:
So I'm working on a LFS build in which everything is installed into buildroot directory for later use and untared into my chroot environment for current use.

When compiling the kernel, 'make zImage' compiles and installs into my chroot enviroment vchiq_arm and all the 'drivers/staging/vc04_services/interface/vchiq_arm' stuff like

`rpi34 build # locate libvchiq_arm.so
/opt/vc/lib/libvchiq_arm.so
/root/smoothwall-git/distrib/opt/vc/lib/libvchiq_arm.so
`
How can I get it installed into the buildroot/linux directory also, so it is included in the linux tarball being built that will be the actual final linux that will be running that gets installed ??

pelwell, 2018-08-13:
VCHIQ consists of a kernel driver and a user-space library. libvchiq_arm.so is the user-space library, which is built from the [userland repo](https://github.com/raspberrypi/userland/).

ShorTie8, 2018-08-13:
Although it may be built in the userland repo also, the kernel also builds the exact same thing while doing 'make zImage'.
But it is not included as part of linux, but installed by 'make zImage'.

rpi34 build # ls -l /opt/vc/lib/libvchiq_arm.so
-rw-r--r-- 1 root root 30288 Mar 13 19:20 /opt/vc/lib/libvchiq_arm.so
rpi34 build # ls -l /root/smoothwall-git/distrib/opt/vc/lib/libvchiq_arm.so
-rw-r--r-- 1 root root 30288 Jun  2 20:41 /root/smoothwall-git/distrib/opt/vc/lib/libvchiq_arm.so

If it is only built by userland, why is it in my chroot enviroment ??
I do not use the userland repo at all.

ShorTie8, 2018-08-13:
rpi34 build # md5sum /opt/vc/lib/libvchiq_arm.so
5cf2e93976cf5bfe9fa09d77b9b4b822  /opt/vc/lib/libvchiq_arm.so
rpi34 build # md5sum /root/smoothwall-git/distrib/opt/vc/lib/libvchiq_arm.so
5cf2e93976cf5bfe9fa09d77b9b4b822  /root/smoothwall-git/distrib/opt/vc/lib/libvchiq_arm.so

/root/smoothwall-git/distrib is my chroot root.
all compiles get install into /root/smoothwall-git/distrib/build/buildroot/...
then taredup and untared into / of the chroot.

pelwell, 2018-08-13:
> Although it may be built in the userland repo also, the kernel also builds the exact same thing while doing 'make zImage'.

Which makefile are you using? The kernel makefile will never build or install libvchiq_arm.so because the kernel tree doesn't include the source code or know anything about it. The buildroot makefiles may build the userland libraries or download them from the [firmware repo](https://github.com/raspberrypi/firmware), but that is outside the scope of a raspberrypi/linux issue.

ShorTie8, 2018-08-13:
all makefiles are of our own creation, buildroot is just a directory name, could be anything.
But the kernel build is doing it. look around line 2354 where it all starts,
or just find/search for  vchiq_arm

[Final.out.log](https://github.com/raspberrypi/linux/files/2282688/Final.out.log)

pelwell, 2018-08-13:
That's the kernel building the driver, not the userspace library.

ShorTie8, 2018-08-13:
Ok, sorry for my ignorance, how do I get it to include all the vchiq_arm driver/stuff then.
If it is not 'make zIamge' building libvchiq_arm.so shared library into the chroot root,
Any idea's on who is making it ??
Or where it is coming from ??
As everything else is just pure linux stuff, only raspberry specific stuff being built is the kernel from your sources.

ShorTie8, 2018-08-14:
My apologies,
After going to linux (4.14.61+),
the chroot libvchiq_arm.so is gone.
Don't know how it got there, but I mustof copied it at some point.

closed 2018-08-14 by ShorTie8