raspberrypi/linux #2646: uvcvideo driver cuts off edges of picture

Ae3NerdGod, 2018-08-14:
using latest rasbian and an "easycap" with a utv007 chipset (for an ambilight project) ive been unable to find a way to get my "grabber" to grab the very edge of the screen, ive tried negative crop values, various of the "quirks"  for the uvcvideo module none of them havedone what i need. in windows, and when using and actual tv, i can see the edges of the screen fine. for an old "easycap" module there was a solution for this  : " sudo sh -c "echo 'options easycap bars=0' >> /etc/modprobe.d/easycap.conf" " ive tried replacing easycap with uvcvideo too and nothing

Ae3NerdGod, 2018-08-16:
lsmod was how i discovered the uvcvideo module, so im pretty sure thats whats initializing the grabber. to confirm:
output of lsmod:
`$ lsmod
Module                  Size  Used by
bnep                   20480  2
hci_uart               24576  1
bluetooth             409600  23 hci_uart,bnep
ecdh_generic           36864  1 bluetooth
8021q                  32768  0
garp                   20480  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
iptable_nat            16384  0
nf_conntrack_ipv4      16384  1
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nf_nat_ipv4            16384  1 iptable_nat
nf_nat                 32768  1 nf_nat_ipv4
nf_conntrack          139264  3 nf_conntrack_ipv4,nf_nat_ipv4,nf_nat
evdev                  24576  1
uinput                 20480  1
iptable_mangle         16384  0
iptable_filter         16384  0
uvcvideo               94208  0
videobuf2_vmalloc      16384  1 uvcvideo
snd_usb_audio         180224  0
videobuf2_memops       16384  1 videobuf2_vmalloc
snd_hwdep              16384  1 snd_usb_audio
videobuf2_v4l2         28672  1 uvcvideo
snd_usbmidi_lib        36864  1 snd_usb_audio
videobuf2_core         49152  2 uvcvideo,videobuf2_v4l2
snd_rawmidi            32768  1 snd_usbmidi_lib
snd_seq_device         16384  1 snd_rawmidi
videodev              212992  3 uvcvideo,videobuf2_core,videobuf2_v4l2
media                  36864  2 uvcvideo,videodev
brcmfmac              262144  0
brcmutil               16384  1 brcmfmac
cfg80211              626688  1 brcmfmac
rfkill                 28672  5 bluetooth,cfg80211
snd_bcm2835            32768  0
snd_pcm               114688  2 snd_usb_audio,snd_bcm2835
snd_timer              36864  1 snd_pcm
snd                    86016  8 snd_hwdep,snd_usb_audio,snd_timer,snd_rawmidi,snd_usbmidi_lib,snd_seq_device,snd_bcm2835,snd_pcm
bcm2835_thermal        16384  0
lirc_rpi               16384  2
spi_bcm2835            20480  0
bcm2835_gpiomem        16384  0
lirc_dev               20480  1 lirc_rpi
rc_core                45056  1 lirc_dev
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
ip_tables              24576  3 iptable_mangle,iptable_filter,iptable_nat
x_tables               32768  3 iptable_mangle,ip_tables,iptable_filter
ipv6                  487424  42
`
output of lsusb:
its the one with no name 
` lsusb
Bus 001 Device 004: ID 534d:0021
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
lsusb -v :
`lsusb -v

Bus 001 Device 004: ID 534d:0021
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
[... 1149 lines cut ...]
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
`
output of dmesg | grep 534d:0021:
` dmesg | grep 534d:0021
[    5.239996] uvcvideo: Found UVC 1.00 device AV TO USB2.0 (534d:0021)
`

im going to unload uvcvideo and try usbtv now that youve mentioned it.

6by9, 2018-08-16:
I'd only raised usbtv as you'd stated that the device was using a utv007. Where did you get the information from that it was utv007?

I'm not going to dig back in the kernel history to see what the bars parameter did, but I think it is irrelevant anyway as it was for a different chipset.

Can you test on a different Linux system? I'm suspecting this is a generic problem on that device, not Pi specific.

https://hyperion-project.org/threads/is-my-easycap-the-problem.641/#post-5004
> just use the stk1160, all other chipset cause errors

Ae3NerdGod, 2018-08-16:
I bought it on Amazon and it was marketed as utv007, and uses a utv007 driver in Windows. Also I could find multiple posts from that same site of people giving up on their st1160 and switching to utv007. Nonetheless, the problem is not that it is giving errors, or working inconsistently, it actually works great, except that the borders are cropped and there doesn't seem to be any paramaters for the driver to change the crop. Again, it works as expected in Windows (vlc) but I can try it on another (non arm)Linux system and report back what driver and if the issue is still present as it is in the two pi's I've tried

Ae3NerdGod, 2018-08-16:
I can also provide pictures of the board and the chip on it,  I didn't see "utv007" anywhere on the chip but I also don't know if that's what is supposed to be on the chip

closed 2019-09-18 by JamesH65