raspberrypi/linux #2654: WARNING: inconsistent lock state

lucastanure, 2018-08-23:
When enable CONFIG_PROVE_LOCKING we have 

[    0.000000] Linux version 4.14.61-v7+ (tanure@Debian9) (gcc version 4.9.4 (Linaro GCC 4.9-2017.01)) #2 SMP Wed Aug 22 17:25:31 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80d95940, node_mem_map ba39f000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @ba341000 s40064 r8192 d21376 u69632
[    0.000000] pcpu-alloc: s40064 r8192 d21376 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  loglevel=8 dwct
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 931740K/970752K available (8192K kernel code, 645K rwdata, 2256K rodata, 1024K init, 8065K bss, 30820K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80900000   (9184 kB)
[    0.000000]       .init : 0x80c00000 - 0x80d00000   (1024 kB)
[    0.000000]       .data : 0x80d00000 - 0x80da1434   ( 646 kB)
[    0.000000]        .bss : 0x80da9120 - 0x8158959c   (8066 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25414 entries in 75 pages
[    0.000000] Running RCU self tests
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU lockdep checking is enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000035] Switching to timer-based delay loop, resolution 52ns
[    0.000720] Console: colour dummy device 80x30
[    0.005164] console [tty1] enabled
[    0.005244] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.005382] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.005464] ... MAX_LOCK_DEPTH:          48
[    0.005547] ... MAX_LOCKDEP_KEYS:        8191
[    0.005632] ... CLASSHASH_SIZE:          4096
[    0.005717] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.005804] ... MAX_LOCKDEP_CHAINS:      65536
[    0.005890] ... CHAINHASH_SIZE:          32768
[    0.005977]  memory used by lock dependency info: 4655 kB
[    0.006077]  per task-struct memory footprint: 1536 bytes
[    0.006244] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.006439] pid_max: default: 32768 minimum: 301
[    0.007153] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.007284] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.012510] Disabling memory control group subsystem
[    0.012964] CPU: Testing write buffer coherency: ok
[    0.015112] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.017024] Setting up static identity map for 0x100000 - 0x10003c
[    0.017776] Hierarchical SRCU implementation.
[    0.020483] smp: Bringing up secondary CPUs ...
[    0.023522] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.026580] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.029040] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.029428] smp: Brought up 1 node, 4 CPUs
[    0.029852] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[... 146 lines cut ...]
[    1.395393] [<801c0d64>] (generic_smp_call_function_single_interrupt) from [<8010ecf8>] (handle_IPI+0x1ac/0x3fc)
[    1.402620] [<8010ecf8>] (handle_IPI) from [<80101528>] (bcm2836_arm_irqchip_handle_irq+0x80/0xb0)
[    1.409818] [<80101528>] (bcm2836_arm_irqchip_handle_irq) from [<80860afc>] (__irq_svc+0x5c/0x7c)
[    1.417006] Exception stack(0xb9d33f38 to 0xb9d33f80)
[    1.420577] 3f20:                                                       00000001 00000001
[    1.427427] 3f40: 00000000 b9d38000 b9d32000 80d075cc 80d07568 80d98db5 00000001 410fd034
[    1.434155] 3f60: 00000000 b9d33f94 b9d33f58 b9d33f88 80181644 80108d0c 20000013 ffffffff
[    1.440870] [<80860afc>] (__irq_svc) from [<80108d0c>] (arch_cpu_idle+0x34/0x4c)
[    1.447551] [<80108d0c>] (arch_cpu_idle) from [<8085ff0c>] (default_idle_call+0x34/0x48)
[    1.454229] [<8085ff0c>] (default_idle_call) from [<8016ff0c>] (do_idle+0xd8/0x150)
[    1.460890] [<8016ff0c>] (do_idle) from [<80170334>] (cpu_startup_entry+0x28/0x2c)
[    1.467542] [<80170334>] (cpu_startup_entry) from [<8010e900>] (secondary_start_kernel+0x130/0x13c)
[    1.474224] [<8010e900>] (secondary_start_kernel) from [<00101bac>] (0x101bac)
[    1.477738] dwc_otg 3f980000.usb: DWC OTG Controller
[    1.481169] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
[    1.484588] dwc_otg 3f980000.usb: irq 62, io mem 0x00000000
[    1.487975] Init: Port Power? op_state=1
[    1.491214] Init: Power Port (0)
[    1.494787] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.498125] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.501373] usb usb1: Product: DWC OTG Controller
[    1.504506] usb usb1: Manufacturer: Linux 4.14.61-v7+ dwc_otg_hcd
[    1.507650] usb usb1: SerialNumber: 3f980000.usb
[    1.511706] hub 1-0:1.0: USB hub found
[    1.514658] hub 1-0:1.0: 1 port detected

pelwell, 2018-08-23:
This is a close relative of #2612, both of which are caused by the fact that the DWC_OTG FIQ support relies on the SMP call mechanism to run a function on a specific core, and that function runs with interrupts disabled.

Any thoughts, @P33M?

JamesH65, 2019-07-31:
@p33m

pelwell, 2019-07-31:
Probably fixed by https://github.com/raspberrypi/linux/commit/d903e18191d372cf917b0db788b5cd43bcd16025.