raspberrypi/linux #2655: *** DEADLOCK ***

lucastanure, 2018-08-23:
'''
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.61-v7+ (tanure@Debian9) (gcc version 4.9.4 (Linaro GCC 4.9-2017.01)) #2 SMP Wed Aug 22 17:25:31 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80d95940, node_mem_map ba39f000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @ba341000 s40064 r8192 d21376 u69632
[    0.000000] pcpu-alloc: s40064 r8192 d21376 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  loglevel=8 dwct
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 931740K/970752K available (8192K kernel code, 645K rwdata, 2256K rodata, 1024K init, 8065K bss, 30820K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80900000   (9184 kB)
[    0.000000]       .init : 0x80c00000 - 0x80d00000   (1024 kB)
[    0.000000]       .data : 0x80d00000 - 0x80da1434   ( 646 kB)
[    0.000000]        .bss : 0x80da9120 - 0x8158959c   (8066 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25414 entries in 75 pages
[    0.000000] Running RCU self tests
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU lockdep checking is enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000035] Switching to timer-based delay loop, resolution 52ns
[    0.000721] Console: colour dummy device 80x30
[    0.005165] console [tty1] enabled
[    0.005245] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.005383] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.005465] ... MAX_LOCK_DEPTH:          48
[    0.005548] ... MAX_LOCKDEP_KEYS:        8191
[    0.005633] ... CLASSHASH_SIZE:          4096
[    0.005718] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.005804] ... MAX_LOCKDEP_CHAINS:      65536
[    0.005891] ... CHAINHASH_SIZE:          32768
[    0.005978]  memory used by lock dependency info: 4655 kB
[    0.006078]  per task-struct memory footprint: 1536 bytes
[    0.006246] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.006440] pid_max: default: 32768 minimum: 301
[    0.007154] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.007284] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.012518] Disabling memory control group subsystem
[    0.012972] CPU: Testing write buffer coherency: ok
[    0.015123] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.017034] Setting up static identity map for 0x100000 - 0x10003c
[    0.017785] Hierarchical SRCU implementation.
[    0.020496] smp: Bringing up secondary CPUs ...
[    0.023534] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.026591] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.029053] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.029441] smp: Brought up 1 node, 4 CPUs
[    0.029864] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[... 263 lines cut ...]
[   19.657032]  #2:  (&ei->i_mmap_sem){++++}, at: [<80389a44>] ext4_setattr+0x7c8/0xaa8
[   19.666821] 
[   19.666821] stack backtrace:
[   19.673115] CPU: 0 PID: 326 Comm: dhcpcd Not tainted 4.14.61-v7+ #2
[   19.680409] Hardware name: BCM2835
[   19.684835] [<80110a40>] (unwind_backtrace) from [<8010cbb4>] (show_stack+0x20/0x24)
[   19.694821] [<8010cbb4>] (show_stack) from [<80842fc8>] (dump_stack+0xd4/0x118)
[   19.703380] [<80842fc8>] (dump_stack) from [<8017fbb4>] (print_circular_bug+0x218/0x338)
[   19.713947] [<8017fbb4>] (print_circular_bug) from [<80182de0>] (__lock_acquire+0x1320/0x1c04)
[   19.725228] [<80182de0>] (__lock_acquire) from [<80184020>] (lock_acquire+0xd0/0x248)
[   19.735984] [<80184020>] (lock_acquire) from [<803c3ff8>] (start_this_handle+0x32c/0x76c)
[   19.747271] [<803c3ff8>] (start_this_handle) from [<803c4b24>] (jbd2__journal_start+0x104/0x2f8)
[   19.759192] [<803c4b24>] (jbd2__journal_start) from [<803613f8>] (__ext4_journal_start_sb+0xac/0x1e8)
[   19.771569] [<803613f8>] (__ext4_journal_start_sb) from [<8038620c>] (ext4_truncate+0x224/0x610)
[   19.783585] [<8038620c>] (ext4_truncate) from [<80389874>] (ext4_setattr+0x5f8/0xaa8)
[   19.794853] [<80389874>] (ext4_setattr) from [<802e3bf0>] (notify_change+0x1cc/0x408)
[   19.806187] [<802e3bf0>] (notify_change) from [<802bda14>] (do_truncate+0x90/0xb4)
[   19.817278] [<802bda14>] (do_truncate) from [<802d137c>] (path_openat+0x3f0/0x100c)
[   19.828526] [<802d137c>] (path_openat) from [<802d327c>] (do_filp_open+0x70/0xd4)
[   19.839582] [<802d327c>] (do_filp_open) from [<802bebfc>] (do_sys_open+0x120/0x1d0)
[   19.850819] [<802bebfc>] (do_sys_open) from [<802becd4>] (SyS_open+0x28/0x2c)
[   19.859793] [<802becd4>] (SyS_open) from [<80108320>] (ret_fast_syscall+0x0/0x28)
[   24.943130] random: crng init done
[   24.950009] random: 7 urandom warning(s) missed due to ratelimiting
'''

pelwell, 2018-08-23:
Please give as much information about the kernel build options as possible,

lucastanure, 2018-08-23:
Arch Arm
CC      arm-linux-gnueabihf-gcc (Linaro GCC 4.9-2017.01) 4.9.4
#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 4.14.61 Kernel Configuration
#
CONFIG_ARM=y
CONFIG_ARM_HAS_SG_CHAIN=y
CONFIG_MIGHT_HAVE_PCI=y
CONFIG_SYS_SUPPORTS_APM_EMULATION=y
CONFIG_HAVE_PROC_CPU=y
CONFIG_STACKTRACE_SUPPORT=y
CONFIG_LOCKDEP_SUPPORT=y
CONFIG_TRACE_IRQFLAGS_SUPPORT=y
CONFIG_RWSEM_XCHGADD_ALGORITHM=y
CONFIG_FIX_EARLYCON_MEM=y
CONFIG_GENERIC_HWEIGHT=y
CONFIG_GENERIC_CALIBRATE_DELAY=y
CONFIG_NEED_DMA_MAP_STATE=y
CONFIG_ARCH_SUPPORTS_UPROBES=y
CONFIG_FIQ=y
CONFIG_VECTORS_BASE=0xffff0000
CONFIG_ARM_PATCH_PHYS_VIRT=y
CONFIG_GENERIC_BUG=y
CONFIG_PGTABLE_LEVELS=2
CONFIG_DEFCONFIG_LIST="/lib/modules/$UNAME_RELEASE/.config"
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
CONFIG_CROSS_COMPILE=""
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION="-v7"
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
CONFIG_KERNEL_GZIP=y
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
CONFIG_CROSS_MEMORY_ATTACH=y
CONFIG_FHANDLE=y
# CONFIG_USELIB is not set
# CONFIG_AUDIT is not set
CONFIG_HAVE_ARCH_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_SIM=y
CONFIG_HANDLE_DOMAIN_IRQ=y
# CONFIG_IRQ_DOMAIN_DEBUG is not set
[... 6215 lines cut ...]
CONFIG_HAS_IOPORT_MAP=y
CONFIG_HAS_DMA=y
# CONFIG_DMA_NOOP_OPS is not set
# CONFIG_DMA_VIRT_OPS is not set
CONFIG_CPU_RMAP=y
CONFIG_DQL=y
CONFIG_GLOB=y
# CONFIG_GLOB_SELFTEST is not set
CONFIG_NLATTR=y
CONFIG_LRU_CACHE=m
# CONFIG_CORDIC is not set
# CONFIG_DDR is not set
# CONFIG_IRQ_POLL is not set
CONFIG_LIBFDT=y
CONFIG_OID_REGISTRY=y
CONFIG_FONT_SUPPORT=y
# CONFIG_FONTS is not set
CONFIG_FONT_8x8=y
CONFIG_FONT_8x16=y
# CONFIG_SG_SPLIT is not set
CONFIG_SG_POOL=y
CONFIG_ARCH_HAS_SG_CHAIN=y
CONFIG_SBITMAP=y
# CONFIG_STRING_SELFTEST is not set
# CONFIG_VIRTUALIZATION is not set

pelwell, 2018-08-23:
Which is a very verbose way of saying `bcm2709_defconfig` + `CONFIG_DEBUG_INFO=y` + `CONFIG_PROVE_LOCKING=y`.

lucastanure, 2018-08-23:
dtoverlay=pi3-disable-wifi
dtoverlay=pi3-disable-bt
# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on
enable_uart=1

lategoodbye, 2018-08-23:
This issue isn't specific to the Raspberry Pi. Please retest with Linux 4.18 and report it the official maintainers.

closed 2018-08-24 by lucastanure