raspberrypi/linux #2660: Jumbo ping kills ethertnet interface in 4.14 and 4.18 - RPI3

mistepien, 2018-08-25:
Pi 3 Model B Rev 1.2

Connected to 1000M switch.
If you execute 
ping -s 4000 some_ip_eg_raspberry_pi
to or from raspberry pi then Ethernet interface is dead.

lategoodbye, 2018-08-25:
Could you please explain what do you mean with kills ethernet (driver crash)?
Can you provide a dmesg from the Raspberry pi via console after the Pi is unreachable?
Did you change the MTU on the Raspberry Pi 3?

mistepien, 2018-08-28:
No, it is unresponsive if ping -s 4000 is used (or if machine with 9000 mtu contacts). Can you reproduce the problem?

lategoodbye, 2018-08-28:
No, currently i don't have much time. But i will try to reproduce it.

lategoodbye, 2018-08-30:
First of all, according to the LAN95xx databook the maximum allowed packet length (without VLAN tag) is 1518 bytes, so there is no official support for Jumbo frames.

I also tried "ping -s 4000" in my non-jumbo LAN and it works as expected. So the only issue i can see is that the driver accept unsupported MTU values.

So my recommendation is to decrease the MTU to a value which all your devices support.

mistepien, 2018-08-30:
But the problem is not my mtu configuration. Problem is that raspberry pi eth interface goes south. I do think that databook says something about interface freeze.

Tomorrow I will test it with two computers set, without switch.

lategoodbye, 2018-08-30:
Okay, ssh worked before ping -s 4000 and after that it runs into a timeout?

mistepien, 2018-08-30:
Just like someone took off the cable.

I do not have access now to my hardware. Tomorrow I install fresh wheezy (I will write which version). I have netbook with 100Mbit interface. I will try with with crosslink, ip will be set static, without interference with mtu (thus 1500). And I will paste eth statistics.

maxnet, 2018-08-30:
>This is problem because huge part of my LAN is 9000 mtu (eg. very fast NAS). There is S912 media 
>player (libreelec) which is 100M/1500 mtu capable and it works fine.

Strange.
That is not supposed to work fine.

Everything inside same VLAN must have same MTU, as it has no way of detecting what the others support.
Only when communicating outside the subnet, you can get notification from your router, that the packet is being dropped because it is too large, and system can then retry sending smaller packets in response to that.

>But the problem is not my mtu configuration. Problem is that raspberry pi eth interface goes south. I do 
>think that databook says something about interface freeze.

Simply drops too large packets.

`ethtool -S` is not supported on lan95xx, but you can still see that happening by looking at the stats in /sys
This is what it looks likes after sending 25 jumbo packets to it:

```
$ grep "" /sys/class/net/eth0/statistics/*error*
/sys/class/net/eth0/statistics/rx_crc_errors:0
/sys/class/net/eth0/statistics/rx_errors:50
/sys/class/net/eth0/statistics/rx_fifo_errors:0
/sys/class/net/eth0/statistics/rx_frame_errors:25
/sys/class/net/eth0/statistics/rx_length_errors:25
/sys/class/net/eth0/statistics/rx_missed_errors:0
/sys/class/net/eth0/statistics/rx_over_errors:0
/sys/class/net/eth0/statistics/tx_aborted_errors:0
/sys/class/net/eth0/statistics/tx_carrier_errors:0
/sys/class/net/eth0/statistics/tx_errors:0
/sys/class/net/eth0/statistics/tx_fifo_errors:0
/sys/class/net/eth0/statistics/tx_heartbeat_errors:0
/sys/class/net/eth0/statistics/tx_window_errors:0
```

Can still communicate with other devices (that have 1500 mtu like the Pi) after that.

mistepien, 2018-09-01:
And the puzzle is solved. The problem is dhclient interfering with mtu.
In my lan dhpc server in dnsmasq on openwrt with options
`list dhcp_option 'option:mtu, 9000'`
It informs dhpc clients to set mtu to 9000. As I said I have few devices capable 100Mbit/1500mtu and they simply ignore this request. But in raspbian something goes wrong.

But if I comment line about mtu in /etc/dhcpcd.conf, like this
`#option interface_mtu`
then everything is just fine. Raspbian stretch with this "amendment" works in my lan.

maxnet, 2018-09-01:
>But if I comment line about mtu in /etc/dhcpcd.conf, like this
>#option interface_mtu
>then everything is just fine. Raspbian stretch with this "amendment" works in my lan.

Have you also tested letting the Pi download something large from your NAS that has mtu 9000?

mistepien, 2018-09-01:
Via rsync (from rsyncd running on NAS with 9000 mtu)
received 7,038,955,954 bytes  8,915,713.75 bytes/sec

lategoodbye, 2018-09-01:
Here is my proposal to fix the kernel issue:
https://gist.github.com/lategoodbye/bb93f1efceb87b91cebd80f3ce34cba3

lategoodbye, 2018-09-04:
This isn't necessary for testing:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

lategoodbye, 2018-11-09:
A fix has been accepted mainline and while be queued for stable:
https://marc.info/?l=linux-netdev&m=154173574216784&w=2

mistepien, 2018-12-28:
so we can close it :D

closed 2018-12-28 by mistepien

nanjirparmar, 2019-09-23:
Hi All,
Is jumbo MTU supported in RPI image ?
I have tried latest raspberry pi image having Linux version 4.19.73-v7l+. But still I am not able to set MTU > 1500 !

Any help please ?

guysoft, 2022-09-29:
Cool, thanks for reporting. 
@nanjirparmar I have no hardware to test and maintain something like this. Technically we use what Rpi provide. so if it works on RpiOS it would work on OctoPi