raspberrypi/linux #2661: VLANs over USB Ethernet (g_cdc/cdc_ether) - truncated IP packets

pakarw, 2018-08-25:
I have two Pi0's connected over USB for Ethernet networking, one of them set up in gadget mode with g_cdc. On both ends, I configured a single VLAN subinterface. The packets at MTU size sent over that subinterface are truncated by 4 bytes. This happens only in one direction: from g_cdc (gadget) to USB host (cdc_ether). The reverse traffic does not seem to be affected.

I got consistent results with kernels 4.9.59+, 4.14.50+ and 4.14.62+

```
# local (192.168.100.1) is in gadget mode, receiving large packets seems fine
pi@pi0:~$ scp pi@192.168.100.2:large_test_file .
large_test_file           17%   60MB   2.4MB/s   02:03 ETA

# from tcpdump on usb0.10, running on remote (192.168.100.2):
2018-08-25 13:39:10.669170 5e:3e:d1:c7:94:cd > 5e:d5:c7:bf:4d:a8, ethertype IPv4 (0x0800), length 1514: (tos 0x8, ttl 64, id 19105, offset 0, flags [DF], proto TCP (6), length 1500)
    192.168.100.2.22 > 192.168.100.1.56060: Flags [.], cksum 0x0e14 (correct), seq 4057188:4058636, ack 1261, win 543, options [nop,nop,TS val 265204 ecr 193432], length 1448
2018-08-25 13:39:10.671989 5e:3e:d1:c7:94:cd > 5e:d5:c7:bf:4d:a8, ethertype IPv4 (0x0800), length 1514: (tos 0x8, ttl 64, id 19106, offset 0, flags [DF], proto TCP (6), length 1500)
    192.168.100.2.22 > 192.168.100.1.56060: Flags [.], cksum 0xe022 (correct), seq 4058636:4060084, ack 1261, win 543, options [nop,nop,TS val 265205 ecr 193432], length 1448

# on sending, the connection got stalls
pi@pi0:~$ scp large_test_file pi@192.168.100.2:tmp/
large_test_file            3% 2112KB   0.0KB/s - stalled -

# from tcpdump on usb0.10, running on remote (192.168.100.2):
2018-08-25 13:40:33.761394 5e:d5:c7:bf:4d:a8 > 5e:3e:d1:c7:94:cd, ethertype IPv4 (0x0800), length 1510: truncated-ip - 4 bytes missing! (tos 0x8, ttl 64, id 49622, offset 0, flags [DF], proto TCP (6), length 1500)
    192.168.100.1.56062 > 192.168.100.2.22: Flags [.], seq 2059:3507, ack 2354, win 558, options [nop,nop,TS val 201744 ecr 273182], length 1448
2018-08-25 13:40:37.201441 5e:d5:c7:bf:4d:a8 > 5e:3e:d1:c7:94:cd, ethertype IPv4 (0x0800), length 1510: truncated-ip - 4 bytes missing! (tos 0x8, ttl 64, id 49623, offset 0, flags [DF], proto TCP (6), length 1500)
    192.168.100.1.56062 > 192.168.100.2.22: Flags [.], seq 2059:3507, ack 2354, win 558, options [nop,nop,TS val 202088 ecr 273182], length 1448
```

pelwell, 2018-08-25:
The VLAN header takes up 4 bytes, so you can't expect a 1514 byte payload to make it through intact. What would you expect to happen with an invalid MTU size? (This is a genuine question, not snark)

6by9, 2018-08-25:
Things have moved around a bit since 2012 and 23834e533184bd2185bce500c789f86b3668739b, but it looks like g_ether should support an MTU up to ~15kB now. https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/usb/gadget/function/u_ether.c#L51

I would be a little concerned that there are a couple of uses of ETH_FRAME_LEN (not VLAN_ETH_FRAME_LEN) in drivers/usb/gadget/function to set up wMaxSegmentSize (f_ecm, f_ncm, and f_subset), but I haven't looked into the detail of those usages. (They do appear related to cdc_ether).

closed 2019-09-18 by JamesH65