raspberrypi/linux #2667: Raspberry Pi Undervoltage

captainbob79, 2018-08-29:
I have a Raspberry Pi 3 running Kodi. Connected to it is an external HDD filled with video files. I use Kodi as a home media library to play the files. Recently I replaced my dying external drive with a new one and since then the Pi regularly gives me the lightning bolt warning and unmounts the external drive. A few moments later the drive is automatically remounted and is good to go for another few minutes before being unmounted again.

popcornmix, 2018-08-29:
Your power supply doens't provide enough power for the hard drive.
Technically it drops below 4.6V (causing the under-voltage warning) when the drive is taking significant current.
Does the drive have a separate power supply or just the USB connection?
What power supply are you using for the Pi3?

JamesH65, 2018-08-31:
Might be worth a look https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=183495

JamesH65, 2018-09-03:
This isn't really a kernel issue, you just need a less power hungry device, or power it separately.

closed 2018-09-03 by JamesH65