raspberrypi/linux #2669: GPIO setup (direction configuration) fails silently on occasion

GM-Script-Writer-62850, 2018-09-02:
So far i have only been able to experience this issue on my model 3 B+ board
i have not had this issue happen on any of my other boards (all single core models)
* My B+ (2014 model) will not do this even with 3 test scripts running

With the latest updates to the kernel i believe the probability of this has decreased, but it can still happen most notably when there are 2 test cases running on different GPIO pins (7, 8, and/or 9 for example)
I have managed to confirm this happens with both GPIO and wiringpi python libraries

[test.py.zip](https://github.com/raspberrypi/linux/files/2343747/test.py.zip)
This script does nothing but setup the GPIO pins over and over and verify that they are indeed setup properly and report any failures that are found and recursively try again till it passes reporting every failure along the way
When running 2 scripts the issue usually shows up for me in a few minutes, my guess is 3 will make it happen even faster (this appeared very consistent until i stared typed this paragraph)
In the event it seems to be taking too long change the 0.25 to a 0.01 in the sleep line near the end, this gave me about 8 failures in under a minute

My guess is something at the driver level for the GPIO pins can not handle 2 actions at the same time as you can only do this if you have multiple cores for processing

```
pi@raspberrypi:~ $ uname -r
4.14.62-v7+
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

Here is the thread i started when I first encountered this issue:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=212022

lategoodbye, 2018-09-03:
Can you reproduce this issue without using these "fancy" python libraries and only by accessing sysfs interface?

pelwell, 2018-09-03:
The RPi GPIO library bypasses the kernel driver and uses direct register accesses to control the GPIOs without any attempt at synchronisation. This is not a kernel bug.

closed 2018-09-03 by pelwell

GM-Script-Writer-62850, 2018-09-03:
@lategoodbye 
[test.sh.zip](https://github.com/raspberrypi/linux/files/2345594/test.sh.zip)
while using the sysfs on 2 pins as fast as the cpu can run it i am unable to get any failures
However if i slam the system one python script one shell script the shell script does fail at times
[test.py.zip](https://github.com/raspberrypi/linux/files/2345622/test.py.zip)