raspberrypi/linux #2670: sound/soc/codecs : cs4265.c backports

flatmax, 2018-09-03:
What is the best way to get these kernel patches backported for this codec ?

https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-4.20&id=f853d6b3ba345297974d877d8ed0f4a91eaca739
https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-4.20&id=be47e75eb1419ffc1d9c26230963fd5fa3055097
https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-4.20&id=6f18bcdaa24bae39c746b57b95af19ff3c41b17f

Codec file effected :
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/sound/soc/codecs/cs4265.c

pelwell, 2018-09-03:
The easiest way (from our point of view) is to create a Pull Request containing the commits. It serves as a place for discussions, at the end of which we can merge/rebase it.

We like backports to have "commit <hash> upstream." (followed by a blank line) as the first non-title line of the commit, but apart from that it is a straightforward process. See https://github.com/raspberrypi/linux/pull/2671 for what it looks like.

flatmax, 2018-09-03:
Thanks Phill, your energy is endless :100:

closed 2018-09-03 by flatmax