raspberrypi/linux #2673: Kernal Panic because of low voltage warning after updating Rasbian

kajackdfw, 2018-09-05:
My 3 newest RPi are dying when I update Rasbian to the newest version. They can not be rebooted because of kernel panic about the low voltage warnings. I have 4 different brands of USB power supplies, and none will supply enough power to avoid the voltage warning. Is the disabling the warning the only solution now?

popcornmix, 2018-09-05:
The right solution is to use a power supply that provides 5V to the Pi when it's under load.
Note: the voltage loss can be due to the power supply or the power supply cable to the Pi.

Syonyk, 2018-09-06:
The 3B+ is a power hungry beastie.  I regularly see >1A draw under heavy "normal" CPU load, and can see north of 2A under "cpuburn-a53."  They're touchy about the power supply, and they're also very touch about the MicroUSB cable used - longer cables, or thinner-than-usual wiring, combined with the drop from the MicroUSB port, means it browns out under load.  I have several USB power supplies that flat out won't boot the 3B+, even though they work fine on the 3.

I'd suggest getting the official Raspberry Pi power supply and trying again.  "Disabling the warning" is unlikely to resolve the issue, as the issue is almost certainly the low voltage causing the panic, not the warning text.  I've had a 3B+ on 4.14 and 4.18 into undervoltage warnings many times (deliberately and unintentionally) without seeing a panic from it, but I only pushed to the edge of the warning - I didn't try to seriously brown them out.

pelwell, 2018-09-06:
Yes - the messages are just warnings, the fault will be due to the processor not having sufficient voltage to function correctly.

kajackdfw, 2018-09-06:
I have tried 4 different power supplies, but I did use the same skinny usb cable. I will try a different cable. That is the first constructive advice so far. The opinion that my power supply is faulty is bs. I have been running 6 RPi with an assortment of power supplies. Only 2 rpi with apt-get updates are having power issues. Suppressing the warning or lowering the voltage threshold would be a good solution. With the new school year starting here in the USA , the Rasbian team may be swamped with this low voltage reports soon.

Thanks.

pelwell, 2018-09-06:
> That is the first constructive advice so far.

That's nonsense. The first response, 15 hours ago, said (my emphasis):

> Note: the voltage loss can be due to the power supply _or the power supply cable to the Pi_.

As for:

> Suppressing the warning or lowering the voltage threshold would be a good solution.

You don't understand - it isn't the warnings that are causing the crashes, it's the undervoltage. Lowering the threshold (which isn't an option because it's implemented in discrete logic) would remove the (apparently slim) chance that one gets to see the warning before the whole system dies.

JamesH65, 2018-09-06:
The power SUPPLY is faulty.  That's to say, by the time the power is supplied to the Pi, it is too low. So that means either the wall wart is providing too little power, or the cable from the wallwart to the Pi is so so thin the power is being lost.

Imagine the fuel low light comes on in your car. Does sticking some tape over the indicator or bending the sensor arm mean that you are not running out of fuel?

kajackdfw, 2018-09-06:
Please keep it civil. Sorry I over-looked the power cord part, but rants about unofficial power supplies is just going to get your blood pressure up. There are 10 of thousands of RPi with common phone usb power supplies in use in schools just waiting for the apt-get to trigger the kernel panic attack issue. I have rolled back my Rpi version to avoid the issue.

closed 2018-09-06 by kajackdfw

pelwell, 2018-09-06:
There's only one person ranting here. The official power supply is always mentioned as a known-good item, but there are plenty of alternatives out there, most of which (with good cables) will protect you against the SD-card-corrupting crash I feel lies in your future. You have been warned.

JamesH65, 2018-09-07:
@kajackdfw Just a heads up, you have had THREE Raspberry Pi employed engineers in this thread stating what the problem is - your power supply is inadequate. Moving back to earlier software will not help with that, all it does it make the logging, that is telling you about the problem, go away. It's NOT the logging causing the problem, it's the lack of power. Once you fix that the logging message will go away, as will any power related kernel issues. If you simply revert to an earlier version to get rid of the warning and don't fix the power issue, then you will be much more susceptible to problems.