raspberrypi/linux #2676: sc16is752   infinite interrupt loop in kernel module

idsoftgit, 2018-09-10:
I made a custom board with sc16is752. Everything work fine until kernel stall at 100% CPU usage and IRQ line hangs at low state and meander on the CS line. Looks like it run in circle in the sc16is7xx_port_irq handler. It is only possible if SC16IS7XX_IIR_NO_INT_BIT in IIR is active.
Any activity on RX line make it working back again.
This situation repeats at least 2-10 hr.

idsoftgit, 2018-09-24:
Looks like I found a solution. Errors from LSR reads only if something placed in receiver FIFO, but what happens if there is no data in FIFO and any error occurred? It's just return true from sc16is7xx_port_irq and keep poling further in the endless loop. It seems like a wrong behavior. I added "return true" if rxlen is zero and now it works without hanging, But I think the right way is call sc16is7xx_handle_rx(port, rxlen, iir) without any check of buffer length at all.
Is there somebody use this chip?

pelwell, 2018-09-24:
There are a number of users of this device. Please post your patch so I can better understand what you mean.

idsoftgit, 2018-09-26:
[patch.txt](https://github.com/raspberrypi/linux/files/2418694/patch.txt)
Ok, after a while of test I guess it works. Ofcourse it should be tested strongly enough.

idsoftgit, 2018-09-29:
It still hangs. But now it takes about 5 days for.

idsoftgit, 2018-11-06:
[patch.txt](https://github.com/raspberrypi/linux/files/2552786/patch.txt)
So, the reason of hangs was when interrupt occurs with zero buffer length and it leads to the infinite loop. Please add patch to mainline.

pelwell, 2018-11-06:
Thanks - that makes much more sense. It's now in rpi-4.14.y and rpi-4.19.y.

JamesH65, 2019-07-31:
Closing this issue as questions answered/issue resolved.

closed 2019-07-31 by JamesH65