raspberrypi/linux #2679: omxplayer doesn't display video on 4.19-rc3 but works on 4.18.7.

jeffersonchua, 2018-09-12:
omxplayer doesn't display video on 4.19-rc3 but works on 4.18.7.

Same environment. Latest firmware. Just kernel, overlays, modules updated.

No errors in dmesg. Just no output to HDMI.

.config attached for both. I took the working .config from 4.18.7 and run 'make config' to generate for 4.19.

[config-4.18.7.txt](https://github.com/raspberrypi/linux/files/2374278/config-4.18.7.txt)
[config-4.19.0-rc3.txt](https://github.com/raspberrypi/linux/files/2374280/config-4.19.0-rc3.txt)

lategoodbye, 2018-09-12:
Looks like 9b53aaf866589b3c30148968fd5e48380fd67f9e is missing.

popcornmix, 2018-09-12:
@lategoodbye I believe that commit is present (but squashed into an earlier commit).

lategoodbye, 2018-09-12:
@jeffersonchua 

Does `vchiq_test -f 1` works in both cases?

popcornmix, 2018-09-12:
I've just built the kernel and tested
```Linux domnfs 4.19.0-rc3-v7+ #256 SMP Wed Sep 12 13:33:13 BST 2018 armv7l GNU/Linux```
omxplayer is playing fine.

@jeff-dagenais can you try building kernel with default config? ``make ARCH=arm bcm2709_defconfig``

jeffersonchua, 2018-09-12:
> @jeff-dagenais can you try building kernel with default config? make ARCH=arm bcm2709_defconfig>

I'm pulling the whole 4.19 git again and will compile with the bcm2709
default setting. Will report back

jeffersonchua, 2018-09-12:
On Wed, Sep 12, 2018 at 8:44 PM Stefan Wahren <notifications@github.com>
wrote:

> @jeffersonchua <https://github.com/jeffersonchua>
>
> Does vchiq_test -f 1 works in both cases?
>
# vchiq_test -f 1
Functional test - iters:1
======== iteration 1 ========
Testing bulk transfer for alignment.
Testing bulk transfer at PAGE_SIZE.

... Same results above on all 3 cases:
       4.18.7 (omxplayer working)
       4.19.0-rc3 (omxplayer not working). Customized .config
       4.19.0-rc3 (omxplayer not working) bcm2709_defconfig

Again, omxplayer not running on 4.19.0-rc3 even with bcm2709_defconfig.

Console is working. HDMI audio ok with omxplayer. Just blank video.

With 4.18.7, omxplayer works fully.

Jeff.

Thanks,
Jeff

popcornmix, 2018-09-12:
Any change if you run omxplayer with ``--no-osd``?
What is gpu_mem set to? Does it help to increase it?
If you run omxplayer with -g is there any difference in ``omxplayer.log`` with working and non-working?

jeff-dagenais, 2018-09-12:
@popcornmix perhaps edit your comment mentioning me... I assume this was a typo. Good luck!

pelwell, 2018-09-12:
Edited - GitHub auto-completion can be really dumb.

jeffersonchua, 2018-09-13:
Found the issue. It was due to this line in config.txt ... removing it
makes omxplayer works again.
   dispmanx_offline=1

It was mentioned here (https://github.com/raspberrypi/firmware/issues/407)
to add that, but it doesn't work anymore for 4.19.0.

Also, do we still need this for pi2/3  ... and still need to remove that
for pi0?
   disable_l2cache=1

Thanks,
Jeff

jeffersonchua, 2018-09-13:
On Thu, Sep 13, 2018 at 12:52 PM Jeff Chua <jeff.chua.linux@gmail.com>
wrote:

> Found the issue. It was due to this line in config.txt ... removing it
> makes omxplayer works again.
>    dispmanx_offline=1
>
> It was mentioned here (https://github.com/raspberrypi/firmware/issues/407) [...]

Also, on a separate topic related to omxplayer, the latest ffmpeg will
cause some video to hang at the end of the video. Fast-forward works, but
if you let it play till the end, it'll hangs. And that only happens to a
few videos, not all.

Using ffmpeg git version earlier than Oct 07 2017 doesn't have this problem.

My Best,
Jeff

pelwell, 2018-09-13:
Please open a new Issue for a separate problem.

jeffersonchua, 2018-09-13:
On Thu, Sep 13, 2018 at 1:37 PM Phil Elwell <notifications@github.com>
wrote:

> Please open a new Issue for a separate problem.
>

Ok. Will do.

Jeff

popcornmix, 2018-09-13:
You should never set ``disable_l2cache`` unless you have a very good reason. 
Setting it to a non-default value will break without having a custom kernel that matches.

jeffersonchua, 2018-09-14:
On Thu, Sep 13, 2018 at 9:39 PM popcornmix <notifications@github.com> wrote:

> You should never set disable_l2cache unless you have a very good reason.
> Setting it to a non-default value will break without having a custom
> kernel that matches.
>

Good to know.

Jeff.

closed 2018-09-17 by jeffersonchua