raspberrypi/linux #2682: Error : Rapberry Pi 3 : unrecognized command line option ‘-mgeneral-regs-only’

kevinjjames, 2018-09-13:
Hi,

I am trying to compile the latest rpi-4.14.y branch using defconfig file : arch/arm64/configs/bcmrpi3_defconfig.

The issue is same as #1974. Compilation is failing with the error in subject line.

Using the gcc compiler : tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/

The console output is as follows:

kevinjjames@iris ~/raspberrypi/raspian-linux $ make ARCH=arm64 CROSS_COMPILE=${CCPREFIX} bcmrpi3_defconfig
#
# configuration written to .config
#
kevinjjames@iris ~/raspberrypi/raspian-linux $ make ARCH=arm64 CROSS_COMPILE=${CCPREFIX} -j4
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
scripts/kconfig/conf  --silentoldconfig Kconfig
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CC      scripts/mod/empty.o
arm-linux-gnueabihf-gcc: error: unrecognized command line option '-mgeneral-regs-only'
scripts/Makefile.build:328: recipe for target 'scripts/mod/empty.o' failed
make[2]: *** [scripts/mod/empty.o] Error 1
scripts/Makefile.build:587: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
Makefile:568: recipe for target 'scripts' failed
make: *** [scripts] Error 2
make: *** Waiting for unfinished jobs....

Any pointers?

Thanks,
Kevin

popcornmix, 2018-09-13:
Are you using the newer compiler suggested in linked issue?

kevinjjames, 2018-09-13:
Yes

popcornmix, 2018-09-13:
I think CROSS_COMPILE needs to point to a 64 bit cross compiler.

kevinjjames, 2018-09-13:
Actually CROSS_COMPILE is pointing to the 4.6.3 compiler version which was
pointed out in the original issue. i.e.
tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/bin/

Is this a 32 bit cross compiler? As per the original issue, even using the
64 bit cross compiler available in the tools repo, the issue was still seen.

Thanks,
Kevin

popcornmix, 2018-09-13:
No, you need a cross-compiler for 64-bit arm. e.g. 
``CROSS_COMPILE=aarch64-linux-gnu``
If you are running on Ubuntu,
``sudo apt-get install gcc-aarch64-linux-gnu``
will likely get it.

kevinjjames, 2018-09-13:
Sure, will try that.

But is it OK to use the The GNU C compiler for arm64 architecture for
building raspbian? I was assuming the compilers in tools repo were
specifically for Raspberry Pi and the aarch64-linux-gnu won't have the Pi
specific changes?

popcornmix, 2018-09-13:
The gcc we have in our repo is just for convenience. You can use any arm32 or arm64 cross compiler from wherever you want.

kevinjjames, 2018-09-14:
Yup.  Using `CROSS_COMPILE=aarch64-linux-gnu` worked.
Also found a page which contains instructions on how to boot the 64bit Image created.
http://www.tal.org/tutorials/booting-64-bit-kernel-raspberry-pi-3

Also read that arm64 doesn't have decompressing capacity; hence "make zImage" won't work. Had to do "make Image".

closed 2018-09-14 by kevinjjames

Nodi-Rubrum, 2020-09-14:
Any solution?  I am having the same issue.

popcornmix, 2020-09-15:
It might help if you explained what your specific problem is.
Assuming you are trying to cross-compile an arm64 kernel then the documentation is being updated in [pull request here](https://github.com/raspberrypi/documentation/pull/1653)

You can see the current state [here](https://github.com/raspberrypi/documentation/blob/f32ba8399504c1e8c40eb7b7a98de34555735d5e/linux/kernel/building.md)

Nodi-Rubrum, 2020-09-16:
I was following the directions posted.  But even trying the additional steps for buster 10, I still got the given error.  The additional steps were supposed to avoid the error, per the documentation.  Not sure if the documentation is wrong or maybe a missing step?  or something has changed as buster 10 has matured?

lurch, 2020-09-16:
@Nodi-Rubrum Can you provide more details please? "having the same issue" is a bit vague, especially as the original problem was reported 2 years ago.

Which OS are you running? On which platform? Which kernel are you trying to compile? Which steps / instructions have you followed? Which step gives you the error? Please copy'n'paste the error message if possible.

kevinjjames, 2020-09-16:
Hi,

I quite don't the remember the steps which solved the issue but seems like
I was trying to compile the raspian linux with FPU enabled and got the
error. Probably using an arm compiler with FPU support might solve the
issue.

Thanks,
Kevin