raspberrypi/linux #2684: dwc2 device mode causes enumeration errors if the udc is not bound to a gadget

ali1234, 2018-09-14:
To reproduce, enable the dwc2 overlay but don't configure any modules to use in device mode. Boot the Pi and plug it into a Linux PC and you get this:

```
[168362.857069] usb 2-2: new high-speed USB device number 6 using xhci_hcd
[168368.153068] usb 2-2: device descriptor read/64, error -110
[168383.769030] usb 2-2: device descriptor read/64, error -110
[168384.005014] usb 2-2: new high-speed USB device number 7 using xhci_hcd
[168389.145021] usb 2-2: device descriptor read/64, error -110
[168404.760992] usb 2-2: device descriptor read/64, error -110
[168404.869037] usb usb2-port2: attempt power cycle
[168405.520966] usb 2-2: new high-speed USB device number 8 using xhci_hcd
[168410.776991] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[168416.152996] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[168416.360932] usb 2-2: device not accepting address 8, error -62
[168416.488943] usb 2-2: new high-speed USB device number 9 using xhci_hcd
[168421.528973] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[168426.904948] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[168427.112924] usb 2-2: device not accepting address 9, error -62
[168427.113041] usb usb2-port2: unable to enumerate USB device
```

...and if you are very unlucky, all the USB ports on your PC will get disabled and you will have to power cycle it to make them work again.

Using gadgetfs it is possible to bind a gadget to the UDC from userspace, at run time. It takes a while to reach this state, during which time the above errors will be seen on the host.

If a driver is bound to the UDC and then unbound, then Pi correctly signals a disconnect to the host like this:
```
[169017.304085] usb 2-2: USB disconnect, device number 12
```

Once in this state the host will not attempt to enumerate it and it will not cause errors.

If you then bind the UDC again, it will correctly enumerate on the host.

So it seems to me that on first boot the dwc2 hardware is in an incorrect state where it signals to the host that a device is connected, but then it does not respond to enumeration because there is nothing bound to it. It should instead be put into the disconnected state until such time as it gets bound to a driver.

I have two theories as to why this could be the case: the firmware has a USB device mode boot method and may be leaving the dwc2 controller in the wrong state, or the Linux kernel driver may simply be incorrectly initializing the hardware state (or not initializing it at all).

lategoodbye, 2018-09-15:
Do you use the Raspbian Kernel 4.14?
Are you able to reproduce this issue with 4.18?

ali1234, 2018-09-15:
Tested with Raspbian kernel (4.14.50+) rpi-update kernel (4.14.69+), and 'BRANCH=next rpi-update' (4.14.17+). They all behave the same.

I am building a 4.18 kernel to test now.

ali1234, 2018-09-15:
I have now tested 4.18.7+ and it is the same.

lategoodbye, 2018-09-16:
Please report this issue to the following:
Minas Harutyunyan <hminas@synopsys.com>
Felipe Balbi <balbi@kernel.org>
Greg Kroah-Hartman <gregkh@linuxfoundation.org>
linux-usb@vger.kernel.org

ali1234, 2018-12-05:
I didn't get around to reporting this upstream yet - I will need to test upstream kernel before doing that.

However someone reported the same issue on IRC today when using the gpioexpander firmware.

It occured to me that this may be a firmware bug, if the firmware is leaving the USB hardware in an odd state. Although technically Linux should handle that, it may not be quick enough if it needs to unpack a large initrd or something.

JamesH65, 2019-07-31:
Please try the latest release kernel and report back results.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

ali1234, 2019-07-31:
Please name a commit that you want testing, and include the reason why you think it fixes the issue.

JamesH65, 2019-07-31:
The very latest release kernel (from apt), just to ensure the issue is either still there, nor not.

ali1234, 2019-07-31:
It still happens on `Linux buster 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l`

JamesH65, 2019-07-31:
OK, thanks. will leave open, although I doubt we will have much time to look at this one.