raspberrypi/linux #2687: dwc_otg rpi3b+ 4.14.52-v7 usb/serial dropping data

edwar64896, 2018-09-19:
I am interfacing with a device which exposes a serial interface over USB. I am regularly seeing data being "lost" or "dropped" in the kernel/driver layer. Attempts to use some of the dwc_otg configuration parameters are yet to be fruitful.

When I get a chance I'll post lsusb and dmesg dumps from the box in question. the serial device is a cp201x chip. The code is attempting to read a sequence of 17 bytes, however intermittently only 16 bytes are being received for the message string in question. This causes the high-level code to fail a checksum check. The same codebase has been used on a x86 machine with no issue. The USB layer in the RPI just seems to be randomly discarding bytes.

malacalypse, 2018-09-25:
I'm having a potentially related issue with USB MIDI. I get dropped events resulting in stuck notes or missed control messages. This is a low speed device. It does not seem to matter which port it is plugged into. 

```
[Tue 25 Sep 2018 20:38:05 GMT]$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 8, If 0, Class=Vendor Specific Class, Driver=lan78xx, 480M
        |__ Port 2: Dev 17, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 2: Dev 17, If 1, Class=Audio, Driver=snd-usb-audio, 12M
```

```
[Tue 25 Sep 2018 20:40:05 GMT]$ lsusb -v

Bus 001 Device 017: ID 1c75:0288
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1c75
  idProduct          0x0288
  bcdDevice            2.00
  iManufacturer           1
  iProduct                2
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          101
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0
      iInterface              0
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength            9
        bInCollection           1
        baInterfaceNr( 0)       1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         1 Audio
      bInterfaceSubClass      3 MIDI Streaming
      bInterfaceProtocol      0
      iInterface              0
[... 318 lines cut ...]
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```

Dmesg contains nothing interesting when packets are dropped, but the device info on connect is:
```
[69248.133919] usb 1-1.2: new full-speed USB device number 17 using dwc_otg
[69248.267433] usb 1-1.2: New USB device found, idVendor=1c75, idProduct=0288
[69248.267440] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[69248.267443] usb 1-1.2: Product: Arturia KeyStep 32
[69248.267447] usb 1-1.2: Manufacturer: Arturia
[69248.267451] usb 1-1.2: SerialNumber: 00000000001A
```

edwar64896, 2018-09-28:
If any admins have suggestions as to how I can further debug this, I would be more than happy to assist.

P33M, 2018-09-28:
Can you add the parameter `dwc_otg.nak_holdoff=1` in /boot/config.txt and retry? If there is negligible improvement, then try with nak_holdoff=0.

malacalypse, 2018-09-28:
I tried it with both configurations. Stuck events persisted in both cases, did not observe any difference in frequency or severity versus no nak_holdoff value set.

malacalypse, 2018-09-28:
But, did you mean `/boot/config.txt` or `/boot/cmdline.txt` ? I put this line in `/boot/config.txt` as requested. But the other threads for USB diagnostics are referring to `/boot/cmdline.txt` for other dwc_otg variables.

6by9, 2018-09-29:
Typo. It needs to go in /boot/cmdline.txt.

malacalypse, 2018-09-29:
Ok I’ll rerun these tests with that in cmdline.txt shortly.

malacalypse, 2018-09-29:
Good news: nak_holdoff=1 improved it significantly, but still a very few dropped events. holdoff=0 seems to have fixed it entirely, although I only gave it about 5 min of good data. That said, it's a clear and marked improvement. 

Does this help you figure out where the problem is, and is it possible to fix it so we can effectively guarantee that we'll never lose an event using these low speed devices? 

Is there any negative side-effect of keeping nak_holdoff=0 in the meantime?

Thank you!

edwar64896, 2018-09-30:
Unfortunately neither setting of nak_holdoff has made a noticeable difference to my situation. Still seeing sequences of 16 byte messages returning.

edwar64896, 2018-10-01:
#https://github.com/raspberrypi/linux/issues/2197

malacalypse, 2018-10-01:
@edwar64896 just checking you put that in `/boot/cmdline.txt` not `/boot/config.txt` right? I had no differences until I put it in the former, not the latter.

malacalypse, 2018-10-02:
@P33M I'm not using a serial device - just class-compliant MIDI and audio stuff.

pelwell, 2018-10-02:
MIDI is 31250 baud.

malacalypse, 2018-10-02:
Not over USB, there are no limits. That'd only for DIN serial MIDI.

P33M, 2018-10-05:
Ah it's an Arturia Keystep device. Previous testing with these devices has shown that they do not have any internal buffering past the 64-byte maximum packet size - so any data generated by key events that exceeds 64 bytes in size since the last time the endpoint was polled results in data loss.

Setting dwc_otg.nak_holdoff=0 in cmdline.txt is recommended if you're using devices from this manufacturer.

Still waiting for confirmation that this has changed the behaviour on the other device...

edwar64896, 2018-10-07:
@malacalypse yes the kernel configuration parameters have always gone into /boot/cmdline.txt

edwar64896, 2018-10-07:
I'm pretty sure that the device I am using has an internal hub as the usb gadget function it exposes is switchable in the config. You can either have it running as a serial device or as a mass storage device. Will this affect the way that the RPi sees it?

malacalypse, 2018-10-09:
@P33M Ok, good to know, and thank you for the explanation. With this configuration my setup appears stable, so I'll continue with that!

edwar64896, 2018-10-18:
Update:

Ok so to keep things moving, I built an RS232 to TTL converter, enabled the onboard UART and connected it all up via the onboard UART. 

Situation is much worse than with the USB! literally every single packet is losing data.

Does this give anyone any more of an idea as to what is happening? I am completely stumped.

malacalypse, 2018-10-22:
I spoke a bit too soon. I still have significant drops when I have the MIDI data go in the Keystep, out to another device (like a sequencer), back in from that device, and out again to the ultimate synth endpoint. So it seems like the timing on the USB interface has a strong effect on the reliability of the bulk data transfer here. If I get time I'll drop back a few kernel versions to see if that solves the issue.

JamesH65, 2019-07-31:
@p33m Any recent changes that may have helped here?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-09-18 by JamesH65