raspberrypi/linux #2690: Please include hid-bigbenff.c driver

hzulla, 2018-09-24:
Hi there,

my driver submission has just been accepted for the upcoming 4.20 mainline kernel by the input maintainers:
https://git.kernel.org/pub/scm/linux/kernel/git/jikos/hid.git/commit/?h=for-4.20/bigbenff&id=256a90ed9e46b270bbc4e15ef05216ff049c3721

The BigBen Interactive PS3OFMINIPAD kids' gamepad is a Playstation3 accessory and was designed for smaller hands. Linux didn't support its rumble effects, so device's HID protocol needed to be reverse-engineered to write this driver.

As Retropie is one of the projects my kids' use with the Raspberry Pi, please include this driver for the Pi's kernel. The patch applies cleanly on 4.14 with no changes.

Thank you.

pelwell, 2018-09-24:
The corresponding config change (`CONFIG_HID_BIGBEN_FF=m`) just builds the 9KB module.

Any objection to me merging the driver and config patch, @popcornmix?

pelwell, 2018-09-25:
Cool - no objection. Merged.

closed 2018-09-25 by hzulla