raspberrypi/linux #2691: Is it possible to backport "media: uvcvideo: Fix driver reference counting" to 4.14 from 4.18.9 kernel

akarapatis, 2018-09-24:
**Issue**
Currently, when removing USB cameras the media device controller node is not removed. As a result, when this is done multiple times (media devices number get close to 255) no new USB camera can be added.

For our case we need to do this as the USB cameras get hanged, so we power reset them but after some time they are not recognized by the camera (dmesg "media: could not get a free minor")

**Work-around**
Reboot PI

**Fix**
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v4.18.9&id=f9ffcb0a21e1fa8e64d09ed613d884e054ae8191

**References**
https://www.spinics.net/lists/linux-media/msg135855.html

pelwell, 2018-09-24:
No problem - done.

akarapatis, 2018-09-24:
@pelwell Thx!!

closed 2018-09-24 by akarapatis