raspberrypi/linux #2697: Double USB arcade joystick only detected as one

tnonnenm, 2018-09-30:
On every update to the kernel the USB-Joystick of my arcade cabinet is not detected as double. The vendor and device IDs have to be added to the black list setting the quirks mode.
Please find attached the patch that solves the problem:
[arcade.patch.txt](https://github.com/raspberrypi/linux/files/2431370/arcade.patch.txt)

popcornmix, 2018-09-30:
This is an upstream linux file that is untouched by this repo. As such the patch should be submitted upstream, and when accepted will appear here. This will then benefit the wider linux community.

Instructions are here: https://www.kernel.org/doc/html/v4.17/process/submitting-patches.html

tnonnenm, 2018-09-30:
Thanks, trying it there...

closed 2018-09-30 by tnonnenm