raspberrypi/linux #2698: trying to make a kernel for raspberry pi 3 b+ (bcm2837-rpi-3-b-plus)

mohmedelwany, 2018-10-01:
writing this command 
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-  bcmrpi3_defconfig`

return
` HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
arch/arm64/configs/bcmrpi3_defconfig:472:warning: override: reassigning to symbol USB_LAN78XX
arch/arm64/configs/bcmrpi3_defconfig:643:warning: symbol value 'm' invalid for LIRC
arch/arm64/configs/bcmrpi3_defconfig:1288:warning: override: reassigning to symbol MMC_BCM2835_MMC
arch/arm64/configs/bcmrpi3_defconfig:1289:warning: override: reassigning to symbol MMC_SDHCI_IPROC
#
# configuration written to .config
#
`

and 
`make -j4 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-
`

return this with fatal error:
` WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/mman.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  UPD     include/generated/uapi/linux/version.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  UPD     include/config/kernel.release
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/qspinlock.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/set_memory.h
  WRAP    arch/arm64/include/generated/asm/sizes.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  WRAP    arch/arm64/include/generated/asm/xor.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/dtc/flattree.o
  YACC    scripts/genksyms/parse.tab.c
  LEX     scripts/genksyms/lex.lex.c
  YACC    scripts/genksyms/parse.tab.h
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/dtc/data.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/dtc/livetree.o
  CC      scripts/mod/devicetable-offsets.s
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  UPD     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/pnmtologo
  HOSTCC  scripts/mod/file2alias.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  CC      kernel/bounds.s
  HOSTCC  scripts/conmakehash
  UPD     include/generated/timeconst.h
  UPD     include/generated/bounds.h
  CC      arch/arm64/kernel/asm-offsets.s
  YACC    scripts/dtc/dtc-parser.tab.h
  YACC    scripts/dtc/dtc-parser.tab.c
  HOSTCC  scripts/recordmcount
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTLD  scripts/mod/modpost
  UPD     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTCC  scripts/sortextable
  HOSTCC  scripts/asn1_compiler
  LDS     arch/arm64/kernel/vdso/vdso.lds
  VDSOA   arch/arm64/kernel/vdso/gettimeofday.o
  VDSOA   arch/arm64/kernel/vdso/note.o
  HOSTLD  scripts/dtc/dtc
  HOSTCC  scripts/extract-cert
  VDSOA   arch/arm64/kernel/vdso/sigreturn.o
scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: No such file or directory
 #include <openssl/bio.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.host:90: recipe for target 'scripts/extract-cert' failed
make[1]: *** [scripts/extract-cert] Error 1
make[1]: *** Waiting for unfinished jobs....
  VDSOL   arch/arm64/kernel/vdso/vdso.so.dbg
  VDSOSYM include/generated/vdso-offsets.h
Makefile:1065: recipe for target 'scripts' failed
make: *** [scripts] Error 2
`

pelwell, 2018-10-03:
Have you tried `sudo apt-get install libssl-dev`?

mohmedelwany, 2018-10-05:
thank you for your reply
The first command return the same warning. Is this normal ?!!

but the fatal error is gone
Thank you very much

stefson, 2018-10-07:
4.19rc is the mainline kernel, if you're going to use it make sure to tighten your seatbelts.

mohmedelwany, 2018-10-10:
thank you very Much

closed 2018-10-10 by mohmedelwany