raspberrypi/linux #2705: rpi-4.19: rtl8192cu: error: initialization from incompatible pointer type

lategoodbye, 2018-10-06:
Hi,
i tried to compile the rpi-4.19 branch with bcm2709_defconfig, but i'm getting the following compile issue:

```
   CC [M]  drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.o
 drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.c:1009:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
   .ndo_select_queue = rtw_select_queue,
                       ^~~~~~~~~~~~~~~~
 drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.c:1009:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
   CC [M]  net/wireless/lib80211_crypt_tkip.o
   CC [M]  drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/usb_intf.o
 cc1: some warnings being treated as errors
 scripts/Makefile.build:305: the rule for target „drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.o“ failed
 make[5]: *** [drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.o] Error 1
```

popcornmix, 2018-10-06:
You must have stricter compile options than me:
```
  CC [M]  drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.o
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.c:1009:2: warning: initialization from incompatible pointer type
  .ndo_select_queue = rtw_select_queue,
  ^
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.c:1009:2: warning: (near initialization for 'rtw_netdev_ops.ndo_select_queue')
```
Let me have a look.

popcornmix, 2018-10-06:
Looks to be superficial (an unused pointer has changed type in a callback). I've pushed a fixup.

lategoodbye, 2018-10-06:
I'm was using the Linaro toolchain: gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf

lategoodbye, 2018-10-06:
Thanks, compile issue is fixed

closed 2018-10-06 by lategoodbye