raspberrypi/linux #2707: Unable to download the Full source/tree

CrazyArjun, 2018-10-07:
Hi there, I am pentest user who has recently purchased the TP-W722N

pelwell, 2018-10-07:
```
/bin/sh: 1: bc: not found
```
Run `sudo apt-get install bc`, which is one of the first instructions in https://www.raspberrypi.org/documentation/linux/kernel/building.md.

CrazyArjun, 2018-10-08:
Hi Pelwell,

Thank you for letting me know the mistake. :-) I see that after installing that bc command, I am able to execute the command " make -j4 zImage modules dtbs ".

Now its been more than 5 minutes that it is still processing the command. I will post back if I get any errors. Thank you for your help.

pelwell, 2018-10-08:
Closing, as I think this is likely to have resolved your problem. Let us know if that isn't the case.

closed 2018-10-08 by pelwell

CrazyArjun, 2018-10-12:
HI Pelwell,

Thanks for the response. My issue partially resolved and the processing took close to 4 hours in my pi. On the next day when I try to execute the next set of commands which was given the manual for beta drivers I couldn't progress further as it says zImage folder is missing and unable to copy the files. Hence I tried redoing the entire project for second time and got stuck at the same place. Hence frustrated and returned the device and purchased a new one where my Kali Pi is detecting it. I will do the testing with the new device. If I need help, hope you don't mind me asking for help one more time. :-)

Thank you for your help. Have a wonderful weekend.