raspberrypi/linux #2708: latest kernel crashes every day

zoff99, 2018-10-09:
rpi3b+ and rpi3b
it freezes every day with kernel crashing.

how can i see what it causing it? please help.

JamesH65, 2018-10-09:
It would be better to post your problem on the forum, its will be seen by more people who may be able to help.

We would need much more information. Which kernel (our release one, or the latest on in github)? What power supply? What kernel message do you see when it crashes?

pelwell, 2018-10-09:
The usual questions are:
1. What distribution (which version, and any updates) are you running?
2. What other hardware is attached?
3. Random crashes and freezes are often caused by inadequate power supply. What does `vcgencmd get_throttled` report after booting?
4. What was the Last Known Good version of the software?
5. Have you tried downgrading to the LKG version?

If the Pi just freezes then it could be very hard to diagnose, but kernel crashes usually leave clues. The best way to capture a crash dump is using a serial cable attached to GPIOs 14 and 15 (header pins 8 and 10). Put `enable_uart=1` in config.txt, run a terminal application on a PC (or the other Pi) set to 115200 baud, and boot the Pi - you should see a login prompt appear in the terminal window. Now use the Pi as you would normally, wait for it to crash, and copy the crash dump from the terminal window.

zoff99, 2018-10-09:
1. raspian lite, latest version
2. usb devices
3. original raspi powersupply
3. What does vcgencmd get_throttled report after booting? --> do i need to run this after the crash, or anytime?
4. a few weeks ago (crashes after kernel was updated with apt-get dist-upgrade)
5. Whats LKG?

i cant attache stuff this is a production device running 24/7
but i could see kernel crash messages on the console when monitor is attached

zoff99, 2018-10-09:
can i downgrade kernel and modules on rapian lite? i mean sort of easy

pelwell, 2018-10-09:
> 2. usb devices

You're going to have to be more specific.

> 4. do i need to run this after the crash, or anytime?

It's hard to run it after a crash, so I meant any time. It would be more useful if the Pi had been running its regular workload for a while.

> 6. Whats LKG?

Sorry - Last Known Good.

> i cant attache stuff this is a production device running 24/7

...except when it crashes. What is your application? It may be relevant to the crash.

> but i could see kernel crash messages on the console when monitor is attached, it's definitly a kernel crash every time

Good. If you add `framebuffer_height=2160` to config.txt you will get more text on screen when it crashes, giving you more chance of capturing the most important details of the crash. Do this, then post a photo of the screen after a crash, taking care to make it as legible as possible.

pelwell, 2018-10-09:
You can downgrade the kernel and modules using rpi-update. Go to [this page](https://github.com/Hexxeh/rpi-firmware/commits/master), pick a release - I'd start with the August 10th version - and make a note of the hexadecimal number for that release, which in this case is c806ec4. Now run `sudo rpi-update c806ec4` to switch to that specific version, reboot, then see if that is more reliable.

pelwell, 2018-10-09:
If that doesn't work, another useful datapoint would be the March 23rd release (653446e), which is the last Raspbian firmware using the 4.4 kernel.

pelwell, 2018-10-09:
[ Corrected release hash to 6534d6e ]

zoff99, 2018-10-12:
the above is now from "4.14.73-v7+ #1148 SMP" (i updated to the latests with rpi-update in the hopes i could be fixed)

pelwell, 2018-10-12:
OK, now try going backwards: `sudo rpi-update c806ec4`

If the system is too unstable to allow an rpi-update, try with `arm_freq=900` in config.txt.

Also, you never properly answered my question about `vcgencmd get_throttled`.

zoff99, 2018-10-12:
always 0:
```
throttled=0x0
```

zoff99, 2018-10-12:
what does this mean:
```
kernel:[193361.345618] Internal error: Oops: 17 [#1] SMP ARM
```

sounds very unspecific

pelwell, 2018-10-12:
I think it's a Domain Fault - some kind of memory access permission failure. Try with reduced clocks:
```
arm_freq=1200
sdram_freq=450
```

zoff99, 2018-10-12:
i am running with 
arm_freq=1200
sdram_freq=300

no change. crashes still.

whats a domain fault?
also for info: i changed the Pi and the sdcard, still crash.

pelwell, 2018-10-12:
> whats a domain fault?

Like I said, some kind of memory access permission failure. Google it if you need more gruesome detail.

Remove all non-essential hardware - you still haven't said what you have attached.

JamesH65, 2019-07-31:
@zoff99 Was there any more information on this? Have you tried updated to the very latest kernel (apt update/apt upgrade) as we know that works fine.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

zoff99, 2019-07-31:
@JamesH65 sadly no change. now i have an external watchdog that reboots the pi when its not responding. its a sad workaround.

it sure can be the USB device i have plugged in. can think of anything else, since i have upgraded kernel and hardware and still the same issue.

zoff99, 2019-07-31:
```
Bus 001 Device 004: ID 04fa:2490 Dallas Semiconductor DS1490F 2-in-1 Fob, 1-Wire adapter
```

this thing has worked before without any issues, and those same adapters work on 2 other pi's without any issue.
i don't know what to do to get more debug information on the cause

zoff99, 2019-07-31:
also using pi original powersupply

JamesH65, 2019-07-31:
Weird then, as we have Pi's that run for years with no problems.  If you remove the USB device does the problem go away?

zoff99, 2019-07-31:
@JamesH65 i'll try this the next days ...

closed 2019-09-18 by zoff99