raspberrypi/linux #2722: I2S sync and DMA transfer error with HiFiBerry Digi using jackd

mr-berndt, 2018-10-17:
Hi, I experience problems with mentioned cards when trying to use jackd. 

The Problem shows as:
[ 23.966746] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[ 102.845095] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated

It happens exclusively when trying to use jackd together with a HiFiBerry Digi+ I/O or a HiFiBerry Digi+ pro. It doesn’t happen with regular HiFiBerry, or IQAudioDAC.
Using aplay works without issue.

I checked with two OS, Raspbian with kernel 4.14.71 and my own build root build with kernels 4.14.34 and 4.14.74.

Jack startup looks ok but jackd fails shortly after:
pi@raspberrypi:~ $ jackd -d alsa
jackdmp 1.9.12
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2016 Grame.
Copyright 2016-2018 Filipe Coelho.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
#JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
creating alsa driver ... hw:0|hw:0|1024|2|48000|0|0|nomon|swmeter|-|32bit
configuring for 48000Hz, period = 1024 frames (21.3 ms), buffer = 2 periods
ALSA: final selected sample format for capture: 24bit little-endian
ALSA: use 2 periods for capture
ALSA: final selected sample format for playback: 24bit little-endian
ALSA: use 2 periods for playback

...

JackAudioDriver::ProcessAsync: read error, stopping...

When that comes, dmesg shows:
[ 23.966746] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[ 102.845095] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated

The problem is confirmed with two different cards respectively and two different pis.

Best regards
mr-berndt

HiassofT, 2018-10-20:
It looks like you configured jack for both capture and playback. In capture mode the WM8804 will sync it's clocks to the incoming SPDIF stream so you have to make sure a valid signal is present on the input.

You'll probably get the same error messages if you try to run "arecord" with no SPDIF input signal.

I'm not familiar with jack but you either need to disable capture at all or configure it so that it opens the capture device only when you actually want to record something.

JamesH65, 2019-07-31:
Please update to the latest kernel which may contain a fix for this issue.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2019-09-18:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2019-09-18 by JamesH65

benstadin, 2019-09-18:
Sorry, I forgot to give feedback on this. I can confirm the latest version fixed the problem.