raspberrypi/linux #2724: /etc/kernel/postinst.d/apt-auto-removal not working well

basti12203, 2018-10-17:
Hello I run the raspberry as a router and need some self compiled kernel modules managed by dkms. So far so good. 
Now I see that when an kernel update is installed all modules of the current running kernel are removed, because the kernel package is removed.
apt-auto-removal create the list in /etc/apt/apt.conf.d/01autoremove-kernels but this names are different from raspbian name scheme. (for example Kernel 4.14.70 is in package raspberrypi-kernel 1.20180919-1)

Is it generally possible to install multiple kernel-packages on same time in raspbian?
So the simplest way would be to update apt-auto-removal for the right package names.

Best Regards

pelwell, 2018-10-29:
Closing, as the question has been answered.

closed 2018-10-29 by pelwell

basti12203, 2018-11-01:
Here are 2 simple scripts. (Not tested very well).
[files.zip](https://github.com/raspberrypi/linux/files/2538577/files.zip)