raspberrypi/linux #2725: Jumbo Packets on Ethernet Gadget

ejolson2005, 2018-10-19:
Between the 4.9.x series and the 4.14.x series there was a regression that prevented the USB Ethernet Gadget from using jumbo packets.  Apparently the initial regression occurred in 4.12.x.  More information at

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931#p1382020

6by9, 2018-10-20:
Isn't this fixed by 8024dcb1164fd8266f3815aea1ebf7232365de61 ? #2556 Upstream commit date 6th May 2018, merged to rpi-4.14.y on 17 Jun 2018.

scramble45, 2019-07-29:
can confirm using a Pi 4 : `BCM54213` it should support up to 10KB but cannot set it.

JamesH65, 2019-07-31:
Can you give more detail on what you are trying to do and how to test it please.

ejolson2005, 2020-08-06:
This is now fixed in all recent kernels so I'm closing.

closed 2020-08-06 by ejolson2005