raspberrypi/linux #2726: netlink error for cangw on Raspberry Pi 3 PiCAN2 Duo

rg4444, 2018-10-23:
Running cangw from `https://github.com/linux-can/can-utils` commands (cangw -F) on raspberry Pi 3 return error:
netlink error -1 (Operation not permitted)
Linux kernel: Linux/arm 4.14.71, mod-gw enabled
modprobe can-gw returns no error.
CAN board - PICAN2 DUO, CAN0 and CAN1 interfaces are up and running.

6by9, 2018-10-23:
I see you have asked the same question on the linux-can project - https://github.com/linux-can/can-utils/issues/111
I'd wait for an answer there before expecting us to search for a kernel issue.

6by9, 2018-10-24:
I see from the linked issue you just hadn't run as root. This issue should be closed.

JamesH65, 2018-10-24:
Closing this issue as questions answered/issue resolved.

closed 2018-10-24 by JamesH65