raspberrypi/linux #2728: sdhost: command MMC_SEND_EXT_CSD can unmount rootfs

lategoodbye, 2018-10-25:
During my investigation about this [issue](https://bugzilla.redhat.com/show_bug.cgi?id=1607872) with the upstream version of sdhost, i've found a problem which affects downstream (bcm2835-sdhost.c) and upstream (bcm2835.c).

Preparation:
- Install mmc-utils on Raspbian SD card ( mmc-utils_0+git20180327.b4fe0c8c-1_armhf.deb )

Steps to reproduce on Raspberry Pi 3B:
- run the following command a few times `sudo mmc extcsd read /dev/mmcblock0` (since this is timing critical it need some attempts)

Expected results:
- timeout of command MMC_SEND_EXT_CSD since this is a SD card not eMMC

Experienced results:
- at least after 10 attempts the root partition is unmounted which is really bad
 
Affected kernel versions:
- rpi-4.14.78
- rpi-4.19
- 4.19 (upstream)

Current investigation results:
Debugging this issue was a real pain because most of the commands depends on the rootfs. But i think i narrowed it down to that point that the command MMC_SEND_EXT_CSD confuses the driver / controller. The command MMC_SEND_EXT_CSD fails with an timeout BEFORE the controller FSM has finished. So the next valid command to the SD card could fail with 

`previous command (8) not complete (EDM 00010004)`

after this strange IO errors occur, the ext4 journal is aborted and the rootfs isn't available. The sdhost isn't able to break free from this situation.

As a band-aid for the upstream driver i made the following [patch](https://github.com/lategoodbye/rpi-zero/commit/20dc36bcc4e548339c6290fcca2c7cfd54781d1c). This avoids the issue but is probably not the best solution. I assume you have (more) documentation about the sdhost controller to identify what's wrong here.

lategoodbye, 2018-10-25:
Additionally here is a screenshot after the issue occured on a Raspberry Pi 3B:
[image]

pelwell, 2018-10-26:
Thanks for the debug effort on this issue. Using an NFS root partition takes most of the pain out of SD driver errors, and network booting further streamlines the process.

This is not an error I've encountered before since the normal MMC subsystem will avoid sending MMC_SEND_EXT_CSD to an SD card unless forced to. You are correct that we do have at least some documentation for the SDHOST interface, but it wasn't sufficient to stop the writing of the original downstream driver being an exercise in trial and error, and this fix is more of the same.

The active ingredient in your workaround isn't the waiting - without prodding, the controller will remain in the READWAIT state indefinitely - it's setting the FORCE_DATA_MODE bit, which is enough to get the FSM out of the rut.

See https://github.com/raspberrypi/linux/pull/2729 for two possible fixes.

pelwell, 2018-10-26:
With those patches applied I can run the offending command in a loop with no apparent ill effect.

lategoodbye, 2018-10-26:
Your patch looks nice.

Btw i tried my patched upstream version against the original issue from Ognian, but this one isn't fixed. The system seems to hang (heartbeat LEDs blinks fast).

Here is the source code to provoke the initial issue:
https://github.com/Ognian/sdmon.git

lategoodbye, 2018-10-27:
Okay, after applying https://github.com/raspberrypi/linux/commit/ed6ceb039aa24f472c70f941cfa12f0ff7a17c1f and https://gist.github.com/pelwell/265baccf92ea4f2d8e47c4f9e1655714 on top of upstream 4.19, the MMC_SEND_EXT_CSD doesn't cause no harm anymore.

In case of sdmon i'm getting `timeout waiting for hardware interrupt` for a SD card which probably doesn't support this command.

Here is the relevant dmesg output in case DMA is enabled:
[dmesg-4.19-CMD56-dma.txt](https://github.com/raspberrypi/linux/files/2521788/dmesg-4.19-CMD56-dma.txt)
Here is the relevant dmesg output in case DMA is disabled:
[dmesg-4.19-CMD56-nodma.txt](https://github.com/raspberrypi/linux/files/2521789/dmesg-4.19-CMD56-nodma.txt)

lategoodbye, 2018-11-03:
I also tested sdmon once with current 4.14 723af8f6d90e65cb10d84279ff4b60fe93cba5d8

```
[  267.988862] mmc0: timeout waiting for hardware interrupt.
...
[  278.228798] mmc0: timeout waiting for hardware interrupt.
...
[  278.236320] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[  278.236522] mmcblk0: timed out sending r/w cmd command, card status 0x400b00
[  278.236526] mmcblk0: command error, retrying timeout
```

[dmesg-4.14.78-CMD56-dma.txt](https://github.com/raspberrypi/linux/files/2545372/dmesg-4.14.78-CMD56-dma.txt)

Ognian, 2018-11-12:
Hello,
I tested today 4.14.79 and sdmon with Apacer AP-MSD32GCA-1HTM, one of the cards capable of this commands.
The output is
```
sudo ./sdmon /dev/mmcblk0
{
"version": "0.01",
"date": "2018-11-12T12:06:03.000Z",
"device":"/dev/mmcblk0",
"idata.response[]":"0x00 0x00 0x00 0x00",
"error":"1st CMD56 CALL FAILED: Connection timed out"
"flashId": ["0x98","0xde","0x94","0x93","0x76","0x51","0x08","0x04","0x04"],
"icVersion": ["0x20","0x12"],
"fwVersion": [82,105],
"ceNumber": "0x04",
"spareBlockCount": 45,
"initialBadBlockCount": 77,
"goodBlockRatePercent": 98.20,
"totalEreaseCount": 297559,
"enduranceRemainLifePercent": 90.27,
"avgEreaseCount": 292,
"minEreaseCount": 3,
"maxEreaseCount": 307,
"powerUpCount": 157,
"abnormalPowerOffCount": 0,
"totalRefreshCount": 0,
"productMarker": ["0x00","0x00","0x00","0x00","0x00","0x00","0x00","0x00"],
"laterBadBlockCount": 0,
"success":true
}
```
dmesg:
```
[  106.071818] mmc0: timeout waiting for hardware interrupt.
[  106.071831] [05f35d3f] PRD1 b9e0ce10 0
[  106.071839] [05f35d4d] PRD2 2 0
[  106.071847] [05f35d53] PRD3 b9dfb604 0
[  106.071855] [05f35d53] PDM> b9db31b0 0
[  106.071863] [05f35d54] REQ< b9db30d0 10801
[  106.071871] [05f35d54] CMD< 12 16df98
[  106.071878] [05f35d57] CMDD b8 200
[  106.071886] [05f35d58] SDMA b9db31b0 b9e0ce10
[  106.071893] [05f35d59] FCM< b9db30d0 b9db3148
[  106.071901] [05f35d5a] RSP  900 0
[  106.071908] [05f35d5a] FCM> b9db30d0 0
[  106.071916] [05f35d5b] CMD  12 200
[  106.071923] [05f35d5b] REQ> b9db30d0 0
[  106.071931] [05f36f4a] DMA< b9db31b0 1
[  106.071938] [05f36f4a] DMA  52 10902
[  106.071946] [05f36f5b] FDA< b9db30d0 0
[  106.071953] [05f36f5b] TCM< b9db31b0 0
[  106.071960] [05f36f5c] CMD< c 0
[  106.071968] [05f36f5d] FCM< b9db30d0 b9db317c
[  106.071975] [05f36f60] RSP  b00 0
[  106.071983] [05f36f60] FCM> b9db30d0 b9db317c
[  106.071991] [05f36f61] TCM> b9db31b0 0
[  106.071999] [05f36f61] FDA> b9db30d0 b9db317c
[  106.072006] [05f36f62] DMA> 0 0
[  106.072013] [05f36f63] TSK< b9db30d0 0
[  106.072021] [05f36f6b] TSK> b9db30d0 0
[  106.072028] [05f374f4] PRD< b9db09b0 0
[  106.072036] [05f374f5] PRD1 b9e0ce10 0
[  106.072043] [05f374f8] PRD2 1 0
[  106.072051] [05f374fc] PRD3 b9dfb604 0
[  106.072058] [05f374fd] PDM> b9db09b0 0
[  106.072066] [05f374fd] REQ< b9db08d0 10801
[  106.072073] [05f374fe] CMD< 12 5bc58
[  106.072081] [05f37503] CMDD 8 200
[  106.072088] [05f37504] SDMA b9db09b0 b9e0ce10
[  106.072096] [05f37505] FCM< b9db08d0 b9db0948
[... 219 lines cut ...]
[  106.073754] [068ca2b8] TIM< 0 0
[  106.073767] mmc0: cmd op 56 arg 0x10 flags 0x35 - resp 00000900 00000000 00000000 00000000, err 0
[  106.073775] mmc0: data blocks 1 blksz 200 - err 0
[  106.073780] mmc0: =========== REGISTER DUMP ===========
[  106.073785] mmc0: SDCMD  0x000040b8
[  106.073790] mmc0: SDARG  0x00000010
[  106.073796] mmc0: SDTOUT 0x017d7840
[  106.073801] mmc0: SDCDIV 0x00000003
[  106.073806] mmc0: SDRSP0 0x00000900
[  106.073811] mmc0: SDRSP1 0x0000380b
[  106.073816] mmc0: SDRSP2 0xffffffff
[  106.073822] mmc0: SDRSP3 0x0002c01f
[  106.073827] mmc0: SDHSTS 0x00000081
[  106.073832] mmc0: SDVDD  0x00000001
[  106.073837] mmc0: SDEDM  0x00010801
[  106.073843] mmc0: SDHCFG 0x0000050e
[  106.073848] mmc0: SDHBCT 0x00000200
[  106.073853] mmc0: SDHBLC 0x00000001
[  106.073858] mmc0: ===========================================
[  106.081896] sdhost-bcm2835 3f202000.mmc: __mmc_blk_ioctl_cmd: data error -110
pi@raspberrypi:~/sdmon_src $ 
```

Maybe this helps
Ognian

Ognian, 2018-12-04:
@lategoodbye one more question: reading once again thru the few  bits of information I have about the 2 consecutive commands my software has to execute, I noticed that the first command ( CMD56 in write mode ) notes that "no data is expected from the card", so I wonder if I have to set some special bit to indicate to the driver that actually no data is expected from the sd card?

lategoodbye, 2018-12-04:
Sorry, i don't have any documentation about the SDHOST controller.

pelwell, 2018-12-04:
The SDHOST interface relies on SDCMD_WRITE_CMD and SDCMD_READ_CMD being set appropriately. The sdhost driver derives their values from the MMC_DATA_WRITE and MMC_DATA_READ flags:
```
		if (cmd->data->flags & MMC_DATA_WRITE)
			sdcmd |= SDCMD_WRITE_CMD;
		if (cmd->data->flags & MMC_DATA_READ)
			sdcmd |= SDCMD_READ_CMD;
```

The register dump on failure includes:
```
SDCMD  0x000040b8
```
This indicates that SDCMD_WRITE_CMD is set and SDCMD_READ_CMD is clear, and the command is 56 (as expected). The rest of the dump shows that the driver is trying to write a 512 byte block, and that no data has been written.

lategoodbye, 2018-12-06:
Phil's patch is now in linux-next.

Ognian, 2018-12-07:
@lategoodbye tried to find a way to use the sdhci-iproc without to have to build a custom kernel, but didn't find one (i.e. no dtoverlay)... any ideas?

JamesH65, 2019-07-31:
Cannot see what is required here.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-07-31 by lategoodbye