raspberrypi/linux #2730: vchiq: Top-down description of driver

lategoodbye, 2018-10-26:
I wasn't sure were to raise this issue (here or in the documentation repo), but i think this fits better. There is [one point](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/staging/vc04_services/interface/vchi/TODO#L48) on the TODO list for the vchiq driver before it can leave staging: documentation

This driver is too complex to get the idea behind it by looking at the code. So it would be nice if someone at RPF can provide so kind of overview / top-down description of the driver and its interfaces.

lategoodbye, 2018-11-28:
Would be an issue in documentation better?

andrum99, 2018-11-28:
The linked document says: 

> "VCHI is just a tool to let drivers talk to the firmware."

A slight expansion on that would be:

"The Raspberry Pi runs two separate operating systems: Linux on the ARM, and a threedx RTOS called vcos running on the Videocore 4 VPU: vcos is commonly called "the firmware". Vcos is tasked with performing certain system services, for example running the display and audio outputs. The VCHI driver provides an interface to allow Linux drivers to talk to vcos. Raspberry Pi develop vcos and the VCHI interface based on demand for certain features and bug fixes, in concert with Broadcom".

The documentation issue is part of the process of getting VCHI included in the upstream Linux kernel. As such, the issue the linked document raises belongs in this repository. If you fee the need to explain the function of VCHI beyond the Linux kernel community, then you should file a separate issue in the documentation repository. A good starting point for any addition to the documentation repository might be my short explanation above.

lategoodbye, 2018-11-28:
@andrum99 Thanks for your feedback. We need a better explanation of the software architecture (e.g. what is the function of those kernel threads: keepalive, recycle, slot handler, sync) and the userspace interfaces of VCHI ( /dev/vchiq , ioctl calls, debugfs ). So it should be focused on the kernel driver.

andrum99, 2018-11-29:
Thanks @jamesh65. I will add a PR to the documentation repository with the information we have collected here about this part of the Pi architecture.

pelwell, 2021-10-14:
Overdue for closing.

closed 2021-10-14 by pelwell