raspberrypi/linux #2731: bcm2835_smi_nand not initializing (nand: chip.cmd_ctrl() callback is not provided)

No0ne, 2018-10-27:
There was an update to all NAND drivers (https://github.com/raspberrypi/linux/commit/4bd4ebcc540c35d4477b098cf26394f976551464) which didn't get applied to bcm2835_smi_nand since it is not in the mainline kernel.
```
[    0.000000] Linux version 4.9.11+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #971 Mon Feb 20 20:40:26 GMT 2017
...
[    9.765751] smi-bcm2835 20600000.smi: initialised
[   10.157228] nand: chip.cmd_ctrl() callback is not provided
```

It did work on older kernel versions:
```
[    0.000000] Linux version 4.4.50+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #970 Mon Feb 20 19:12:50 GMT 2017
...
[   11.357678] nand: device found, Manufacturer ID: 0xec, Chip ID: 0xf1
[   11.357716] nand: Samsung NAND 128MiB 3,3V 8-bit
[   11.357732] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
```

I've written a small patch but I don't if its correct and not missing something so I didn't create a pull request. Please review:
```
diff --git a/drivers/mtd/nand/raw/bcm2835_smi_nand.c b/drivers/mtd/nand/raw/bcm2835_smi_nand.c
index 78dc4fcc7..5a2374dc3 100644
--- a/drivers/mtd/nand/raw/bcm2835_smi_nand.c
+++ b/drivers/mtd/nand/raw/bcm2835_smi_nand.c
@@ -50,10 +50,18 @@
 struct bcm2835_smi_nand_host {
 	struct bcm2835_smi_instance *smi_inst;
 	struct nand_chip nand_chip;
-	struct mtd_info mtd;
 	struct device *dev;
 };
 
+static const struct mtd_partition partition_info[] = {
+  [0] = {
+		.name	= "bcm2835-smi-nand-0",
+		.offset	= 0,
+		.size	= MTDPART_SIZ_FULL
+  }
+};
+#define NUM_PARTITIONS (ARRAY_SIZE(partition_info))
+
 /****************************************************************************
 *
 *   NAND functionality implementation
@@ -135,10 +143,8 @@ static int bcm2835_smi_nand_probe(struct platform_device *pdev)
 	struct mtd_info *mtd;
 	struct device *dev = &pdev->dev;
 	struct device_node *node = dev->of_node, *smi_node;
-	struct mtd_part_parser_data ppdata;
 	struct smi_settings *smi_settings;
 	struct bcm2835_smi_instance *smi_inst;
-	int ret = -ENXIO;
 
 	if (!node) {
 		dev_err(dev, "No device tree node supplied!");
@@ -185,8 +191,7 @@ static int bcm2835_smi_nand_probe(struct platform_device *pdev)
 	/* Link the structures together */
 
 	this = &host->nand_chip;
-	mtd = &host->mtd;
-	mtd->priv = this;
+	mtd = nand_to_mtd(this);
 	mtd->owner = THIS_MODULE;
 	mtd->dev.parent = dev;
 	mtd->name = DRIVER_NAME;
@@ -194,7 +199,6 @@ static int bcm2835_smi_nand_probe(struct platform_device *pdev)
 	/* 20 us command delay time... */
 	this->chip_delay = 20;
 
-	this->priv = host;
 	this->cmd_ctrl = bcm2835_smi_nand_cmd_ctrl;
 	this->read_byte = bcm2835_smi_nand_read_byte;
 	this->write_byte = bcm2835_smi_nand_write_byte;
@@ -202,6 +206,7 @@ static int bcm2835_smi_nand_probe(struct platform_device *pdev)
 	this->read_buf = bcm2835_smi_nand_read_buf;
 
 	this->ecc.mode = NAND_ECC_SOFT;
+	this->ecc.algo = NAND_ECC_HAMMING;
 
 	/* Should never be accessed directly: */
 
@@ -213,15 +218,15 @@ static int bcm2835_smi_nand_probe(struct platform_device *pdev)
 	if (nand_scan(mtd, 1))
 		return -ENXIO;
 
-	nand_release(mtd);
-	return -EINVAL;
+	mtd_device_register(mtd, partition_info, NUM_PARTITIONS);
+	return 0;
 }
 
 static int bcm2835_smi_nand_remove(struct platform_device *pdev)
 {
 	struct bcm2835_smi_nand_host *host = platform_get_drvdata(pdev);
 
-	nand_release(&host->mtd);
+	nand_release(nand_to_mtd(&host->nand_chip));
 
 	return 0;
 }
```

lategoodbye, 2018-10-27:
@No0ne 

Please let us know against which branch did you create this patch?

Some comments about your patch:
- add some error handling in case mtd_device_register fails
- recent NAND drivers seems to call nand_set_flash_node (but i'm not aware of a possible resulting binding change) maybe this would be nice to have

No0ne, 2018-10-29:
I'm working with the rpi-4.19.y branch, but I have another problem which prevents me from verifying everything is working correctly: https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=217978

lategoodbye, 2018-10-29:
You mean the problem, that all blocks are marked as bad? But this problem already exists with 4.4?

No0ne, 2018-10-29:
Yes thats right, but I think the "bad blocks" issue is a different problem specific to my nand chip.
This issue is simply about bcm2835_smi_nand.ko not loading at all since 
https://github.com/raspberrypi/linux/commit/4bd4ebcc540c35d4477b098cf26394f976551464 .

lategoodbye, 2018-10-29:
Sorry for the confusion the second question refers to the block issue.

I know that the whole issue refers about driver loading.

No0ne, 2018-11-05:
I've tried everything on a 3B and I got strange results:
```
Nov  2 11:57:41 nand3pi kernel: [ 2395.558265] nand: second ID read did not match ec,f1 against ec,00
Nov  2 11:57:41 nand3pi kernel: [ 2395.558282] nand: No NAND device found

Nov  2 11:58:35 nand3pi kernel: [ 2449.563278] nand: second ID read did not match ec,f1 against ec,ec
Nov  2 11:58:35 nand3pi kernel: [ 2449.563294] nand: No NAND device found
```
I'll hook up a logic analyzer to see whats going on!

I've also tried reading the Waveshare board using NANDway from https://github.com/hjudges/NORway which works fine btw.

lategoodbye, 2018-11-05:
Please always describe what you did (especially NANDway).
Does this mean your 4.19 branch isn't able to detect the NAND instead of blocks marked as bad?

No0ne, 2018-11-05:
I will post all details if I have definitive proof about whats not working here (hardware or software) after checking with a logic analyzer.
(I only did check with NANDway to confirm my nand chip isn't bad)

dgiese, 2019-03-24:
any update for this issue?

JamesH65, 2019-12-20:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2019-12-20 by JamesH65

TheCrazyT, 2021-04-09:
@No0ne also interested in any updates, made a workaround mentioned on https://github.com/raspberrypi/linux/commit/72ce5a4330588174a0e138c1a87626bf16217020 but its not very satisfying.