raspberrypi/linux #2734: [USB] Broken pipe while enumerating Hub+LAN dongle

FD-, 2018-10-29:
I'm having troubles getting a Hub+Ethernet dongle to work with the Raspberry Pi. Whenever I attach the dongle to the Pi, the integrated ethernet chip doesn't show in lsusb. The kernel log contains the information that it could not be enumerated:

```
[   68.131540] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
[   68.262202] usb 1-1.3: New USB device found, idVendor=1a40, idProduct=0101
[   68.262217] usb 1-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   68.262225] usb 1-1.3: Product: USB2.0 HUB
[   68.265403] hub 1-1.3:1.0: USB hub found
[   68.266375] hub 1-1.3:1.0: 4 ports detected
[   68.581541] usb 1-1.3.2: new full-speed USB device number 9 using dwc_otg
[   68.681553] usb 1-1.3.2: device descriptor read/64, error -32
[   68.901535] usb 1-1.3.2: device descriptor read/64, error -32
[   69.121531] usb 1-1.3.2: new full-speed USB device number 10 using dwc_otg
[   69.221528] usb 1-1.3.2: device descriptor read/64, error -32
[   69.441534] usb 1-1.3.2: device descriptor read/64, error -32
[   69.561630] usb 1-1.3-port2: attempt power cycle
[   70.221589] usb 1-1.3.2: new full-speed USB device number 11 using dwc_otg
[   70.661591] usb 1-1.3.2: device not accepting address 11, error -32
[   70.761596] usb 1-1.3.2: new full-speed USB device number 12 using dwc_otg
[   71.201611] usb 1-1.3.2: device not accepting address 12, error -32
[   71.201718] usb 1-1.3-port2: unable to enumerate USB device
```

<details><summary>Full dmesg (click to expand)</summary>
<p>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.78-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1156 SMP Tue Oct 23 14:34:39 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c85280, node_mem_map ba39f000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @ba348000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=0bd3cdbf-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940232K/970752K available (7168K kernel code, 576K rwdata, 2076K rodata, 1024K init, 698K bss, 22328K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c9017c   ( 577 kB)
                      .bss : 0x80c97f04 - 0x80d468b0   ( 699 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25284 entries in 75 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000019] Switching to timer-based delay loop, resolution 52ns
[    0.000267] Console: colour dummy device 80x30
[    0.000284] console [tty1] enabled
[    0.000309] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[... 381 lines cut ...]
- Same dongle unit works with Windows
- Also works when starting Pi with dwc_otg.speed=1 (though that breaks keyboard and mouse)
- Works when using the dwc2 module instead of dwc_otg (though that uses 100% CPU while idle)
- Interestingly, some people are successfully using the visually same dongle model with their Pis. However, I have also found reports of users seeing the same issues as described here.

Please let me know if the hardware could be faulty in a way that makes it still work in the described scenarios. I am wondering if a faulty crystal could cause this.

The dongle is one of the dirt-cheap (3$) generic products available through AliExpress and Ebay.
It contains a Terminus Technology SL2.2s USB 2.0 HUB TT0509A18 chip and a CoreChip SR9700 ethernet chip. I have posted pictures of the internals on the [Raspberry Pi forums](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=225509). 

### Setup:
**Used kernel**: Newest from rpi-update (4.14.78-v7+)
**Tested Pis**: Pi0, Pi 3B+
**Power supplies**: Tested 3 different supplies + Powering the dongle itself
**Connection**: Ethernet+Hub dongle connected directly into the Pis

### What I have tried:
- Setting dwc_otg.fiq_enable to 0 and 1: No change
- Setting dwc_otg.fiq_fsm_enable to 0 and 1: No change
- Setting dwc_otg.fiq_fsm_mask to any value between 0x0 and 0x10: No change
- Setting dwc_otg.speed to 1: The chip is recognised, but no keyboard nor mouse
- Captured USB communication (Wireshark): Could see URB status -32 from hub to kernel -> Must be a lower-level problem
- Compiled kernel with verbose logging in dwc_otg module ([logged tons of information](https://gist.github.com/FD-/bb4ceb8be3912d7d84d54bad17e5ca08)): No big red warnings

Please let me know if you need any additional information on the issue!

P33M, 2018-10-30:
Where can I buy one of these devices? A link to a reputable e-shop selling these as a line item (not an auction for a limited quantity) would be preferred.

FD-, 2018-10-30:
I'm afraid I don't know of any reputable shop that sells them (I guess that's why they're so cheap). If you consider AliExpress a reputable shop, I found this product has multiple reviews mentioning the same problems with the Raspberry Pi:

https://www.aliexpress.com/item/Hot-Brand-Micro-USB-to-Network-LAN-Ethernet-RJ45-Adapter-with-3-Port-USB-2-0/32691036545.html

ickromwerk, 2018-11-03:
Have the same problem with BURO BU-HUB4-U2.0-Slim [link](https://www.citilink.ru/catalog/computers_and_notebooks/hubs/389734/)

It has **Terminus Technology SL2.2s USB 2.0 HUB TT0509A18** chip inside

Hi speed devices work ok with it(wifi/bluetooth dongle). Any 1.1 device gives the same dmesg errors(tried ftdi, two keyboards)

Works great with PC.

FD-, 2018-11-04:
That's very interesting indeed! The ethernet chip (CoreChip SR9700) is USB 1.1, so the problem actually seems to be with talking to a USB 1.1 device over the Terminus SL2.2s TT0509A18 chip. Maybe this helps to narrow down the issue, @P33M ?

ickromwerk, 2018-11-04:
@FD-, how did you set up DWC_OTG debug level?

SET_DEBUG_LEVEL in dwc_otg_dbg.h does not called from anywhere. Is it possible to set debug level at runtime?

FD-, 2018-11-04:
I also couldn't really make sense of SET_DEBUG_LEVEL, so I chose a somewhat hacky approach:
I changed DWC_DEBUGPL to remove the level check:

    #define DWC_DEBUGPL(lvl, x...) do{ __DWC_DEBUG(USB_DWC x ); }while(0) 

and changed CHK_DEBUG_LEVEL to always return true:
 
    # define CHK_DEBUG_LEVEL(level) (1)

Unfortunately, this lead to a huge amount of debug messages being logged. I was lucky to be able to extract some of the messages before the system locked up due to no more disk space. Had to reflash the sd card in the end.

I also had to enable debugging in the dwc_common makefile.

ickromwerk, 2018-11-04:
Ok. I thought you find a better way :) I rewrote this macro as:
`# define DWC_DEBUGPL(lvl, x...) do{ if ((lvl)&DWC_OTG_DEBUGLEV)__DWC_DEBUG(USB_DWC x ); }while(0)`
and use Makefile to set debug level:
`ccflags-y	+= -DDWC_OTG_DEBUGLEV=DBG_HCD`

I think you may miss broken pipe moment in your log capture. Because it must be shown up at line like:
`...
[ 115.087836] _complete: urb b70e0c80, device 6, ep 1 IN, status=0
...`
where status will be -32

FD-, 2018-11-04:
Yeah, as mentioned, I could only save a small portion of the log (I think I only have the messages of one second's time). Please post your log if it captures details about the issue!

FD-, 2018-11-04:
I'm afraid I don't know anything about these interrupts, but the stalling thing is certainly interesting!

ickromwerk, 2018-11-04:
Here is [verbose](https://gist.github.com/ickromwerk/d8e55ccf2e550db3ea2beccd2d9069a7) part till status=-32

ickromwerk, 2018-11-04:
And here goes strange thing. DBG_HCDI logs interupts. Each record looks like:
```
[   59.783905] DWC_otg: DWC OTG HCD Interrupt Detected gintsts&gintmsk=0x02000008 core_if=da425800
[   59.783916] DWC_otg: --Host Channel 5 Interrupt: Channel Halted--
[   59.783924] DWC_otg: --Host Channel 5 Interrupt: Transfer Complete--
[   59.783943] DWC_otg: DWC OTG HCD Finished Servicing Interrupts
```

Here entry and exit messages and in the middle is a description. They all looks the same from the system start.

After detecting usb1.1 device they look like [this](https://gist.github.com/ickromwerk/4b16d03c2d57975f2fa9e7d52f44dc17)

mypiandrew, 2018-11-05:
Hi

We found similar problems with Genesys GL850 Hubs

Kernel version 4.9.80 works for us :

`rpi-update 5c80565c5c0c7f820258c792a98b56f22db2dd03   `

4.14.x doesn't appear to work no matter what we try

bumping up to the latest 4.19 kernel however appears to fix the issue... :

`BRANCH=next rpi-update`

uname -a
Linux raspberrypi 4.19.0-v7+ #1157 SMP Tue Oct 23 18:10:04 BST 2018 armv7l GNU/Linux

Would be interested to see if you find the same results.

P33M, 2018-11-05:
A "stall" is a genuine USB device condition (i.e. returned status from the hub). What's odd is that going from 4.14 -> 4.19 fixes the issue, as there have been no changes to the dwc_otg driver in that timeframe. The usual suspect for symptoms like these is timing-related breakage provoked by non-standard device implementations - it looks like there is (another) class of hub that doesn't play well with the Pi.

FD-, 2018-11-05:
It seems @mypiandrew's issue is not the same as the one with the Terminus Technology SL2.2s USB 2.0 HUB TT0509A18 described here. I have now tested kernel versions 4.1, 4.9, 4.14, 4.19 on a Pi0 and all show the exact same behaviour (error -32 and unable to enumerate).

FD-, 2018-11-05:
@P33M I'm afraid I just have one unit that I still need for its hub functionality. Also, it seems like shipping would cost about 2x the price of the dongle itself (including shipping from China). 

I'd suggest to contact the AliExpress seller I linked above about the possibility of getting a demo unit. I'd think they're willing to help if you prove them your affiliation with Raspberry Pi. After all, it seems many customers buy the dongle for their Pi0s and write bad reviews when they find it incompatible. 
If you don't have an AliExpress account, I could act as an intermediary, though I'd imagine that would make things a bit complicated.

As mentioned, this product on AliExpress has multiple reviews mentioning the same issue:
https://www.aliexpress.com/item/Hot-Brand-Micro-USB-to-Network-LAN-Ethernet-RJ45-Adapter-with-3-Port-USB-2-0/32691036545.html
Alternatively, a forum user [recently reported the same issue](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=226020&p=1387889) with this AliExpress product:
https://www.aliexpress.com/item/Super-Speed-2017-High-Quality-USB-to-RJ45-Ethernet-Network-Adapter-Card-With-3-Ports-USB/32831851624.html

6by9, 2018-11-05:
I happen to have one of these devices that reports as ```1a40:0101 Terminus Technology Inc. Hub``` and ```0fe6:9700 Kontron (Industrial Computer Source / ICS Advent) DM9601 Fast Ethernet Adapter```.
@P33M You're welcome to borrow it. It's the variant with a micro-USB plug on it, so plugs straight into a Pi0.

The USB hub side has come up immediately for me with a keyboard and mouse plugged in, and the network side seems to be happy too.
These are the devices where they have left the MAC address EEPROM off, so they all end up with the same MAC address. They also claim to be USB2.0 when they're actually USB1.1 or something similar. They're known to be a touch ropey!

P33M, 2018-11-05:
My suspicion is that the hub chip(s) in question are knock-offs. I can find no reference to a "SL 2.2s" hub on anything that isn't a Raspberry Pi related post. Other Terminus hubs (including the reduced pin count SOIC package FE 1.1 device) have worked successfully when connected to a Pi before. In fact, opening up the model that @6by9 has confirms that it's a FE1.1s inside.

The "genesys GL850" is apparently a real product so I'd be keen to see if this is a special case.

FD-, 2018-11-05:
Will try to find out more about that! Would it be possible to post pictures of @6by9's dongle's internals for comparison? I've posted pictures of mine [in the forum](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=225509).

6by9, 2018-11-05:
https://drive.google.com/file/d/17xdr7gKpDpfnMxb73-hEzw4LKDWTSJvN/view?usp=sharing
https://drive.google.com/file/d/1fFDP6_KxB_87rLcTJKa2lAScTJLU2JcN/view?usp=sharing

FD-, 2018-11-05:
Thanks so much! Am I allowed to post them on the forum?

6by9, 2018-11-05:
The abscence of C28 on your board would imply that they really are scrimping on the power. 100uF 16V on my board.
P33M also commented that you're missing the magnetics on the ethernet (the black block behind the ethernet socket on mine), so you have no isolation and it may well misbehave. C12, C13, C15, and C18 are apparently 0 ohm links instead.
The two crystals seem to be haphazardly mounted.

> Thanks so much! Am I allowed to post them on the forum?

If you wish, yes.

FD-, 2018-11-05:
Thanks for the explanation and pictures! From the outside, all these adapters looked like they were the exactly the same.

Also, it seems like @P33M is right that the hub chip on mine is likely a knock-off. It doesn't have that Terminus T logo that yours has. Instead, it has some other logo (possibly "CC"; It's so tiny that it's nearly impossible to read.)

May I ask where you got your dongle from and how much it cost? 

I'll post these findings on the forum so that people know which of these adapters work with the Pi and which to avoid.

ickromwerk, 2018-11-05:
No way you can avoid them. The only way is to fix broken Pi code. This hub works pretty well on other systems.

6by9, 2018-11-05:
> May I ask where you got your dongle from and how much it cost?

Ebay in December 2015. About £5 IIRC - I haven't got the history back that far immediately to hand.

P33M, 2018-11-05:
> No way you can avoid them. The only way is to fix broken Pi code. This hub works pretty well on other systems.

It remains to be seen whether this can be fixed "in code". We already have documented examples of non-USB2.0 compliant hub chips (particular VIA USB3.0 hubs) misbehaving when connected to the Pi and there is nothing we can do in software to accommodate the hardware incompatibilities. Without plugging a protocol analyser in the loop between the hub and the Pi, the source of the incompatibility is undetermined.

FD-, 2018-11-05:
> Ebay in December 2015. About £5 IIRC - I haven't got the history back that far immediately to hand.

Thanks! Mine was a bit cheaper (~3 dollars) on ebay, so it seems they tried to make extra savings in the construction.

> It remains to be seen whether this can be fixed "in code". We already have documented examples of non-USB2.0 compliant hub chips (particular VIA USB3.0 hubs) misbehaving when connected to the Pi and [...]

I just found the datasheet for the SL2.2s: https://wenku.baidu.com/view/5bfdec3c182e453610661ed9ad51f01dc28157a9
It's in Chinese, which strengthens the theory it's a knockoff. I agree with @ickromwerk that nonetheless, it would be great to have it supported by the Raspberry Pi!

FD-, 2018-11-05:
The logo is indeed CC, seemingly for CoreChips. The logo looks identical to that on [this chip](https://lcsc.com/product-detail/USB_CoreChips-SL2-1A_C192893.html), though that's not the exact same chip.

ickromwerk, 2018-11-05:
[これはですね？](https://wenku.baidu.com/view/5bfdec3c182e453610661ed9ad51f01dc28157a9.html)

FD-, 2018-11-05:
I finally got hold of the full datasheet, a schematic drawing and a PCB design file in PDF form:
https://archive.org/details/SL2.2sDatasheetV12
Seems like the manufacturer is actually a company called SL Chip ShenZhen Co., Ltd. I don't know why the chip carries a CoreChips logo when it's actually designed by another company. Maybe SL just designed these, but doesn't produce them.

hunnimonstr, 2018-11-08:
@pm33, i have 3 x usb hub/network port devices that match photographs of  'failing' versions of these hubs that i bought for use with my Pi-0 dev boards (micro/otg usb plug).
@fd over on the Pi user forum pointed me here..
been a while since i plugged in to test but iirc lsusb returned an entry for the additional usb hub and nothing else..  this led me on a late night google scavenge that proved fruitless (probably because its a long time since i reuilt a system kernel)  

I am in the UK, So if you want a useless hub to play with let me know your postal addy and i will drop one in the post.
the accompanying driver cd proved useless/unreadable on 2 x of the 3 hub-cd bundles i took a punt on.. 
 i guess i can zip and attache the cd contents here?

FD-, 2018-11-12:
Just marking @P33M here so he's notified about @hunnimonstr's comment.

P33M, 2018-11-12:
@hunnimonstr  As per previous comment - I would be interested in getting hold of these.

> It may be faster if I just get my hands on the device(s) in question:  if you can spare/donate an example device that exhibits the issue can you email [info@raspberrypi.org](mailto:info@raspberrypi.org) with the subject "Github issue 2734 - Failing USB hubs for analysis" to get a dialogue started?

P33M, 2018-11-15:
@mypiandrew I have responded to your email. info@ remains devoid of other correspondence, though...

P33M, 2018-12-10:
I am now in receipt of both the cheap hub and the CM-based hardware. However most of our test equipment is in crates awaiting our office move...

FD-, 2018-12-10:
@P33M Just to clarify: By cheap hub, do you mean a hub+ethernet combo or a GL850 Hub?

P33M, 2018-12-10:
The issue appears to be linked to full-speed devices behind GL850/or unknown equivalents. I have 2 devices that exhibit the issue which may or may not have identical hub chips.

P33M, 2019-01-15:
The failing hub appears to ignore the very first start-split transaction that I send it. It responds with "ACK" but subsequent CSPLIT packets either respond STALL or a sequence of NYET-STALL. Nothing comes out of the other end, i.e. no transactions appear at the device.

Looking at analyzer logs, it looks like we're setting port 1 in the ssplit packet regardless of what the actual port is. That's likely to cause no end of trouble.

P33M, 2019-01-15:
If I plug a FS device into the combo hub in port 1, it works. If I plug a FS device into port 1 and another FS device into ports 3 or 4 (port 2 is the internal ethernet adapter which I have disabled) then the device fails to probe in the same fashion. This one-liner fixes the problem on the combo ethernet hub, I will shortly prove the fix works on @mypiandrew's test hardware.
```
diff --git a/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c b/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c
index ed96940..99a58ad 100644
--- a/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c
+++ b/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c
@@ -232,7 +232,7 @@ static int _hub_info(dwc_otg_hcd_t * hcd, void *urb_handle, uint32_t * hub_addr,
                else
                        *hub_addr = urb->dev->tt->hub->devnum;
        }
-       *port_addr = urb->dev->tt->multi ? urb->dev->ttport : 1;
+       *port_addr = urb->dev->ttport;
    } else {
         *hub_addr = 0;
        *port_addr = urb->dev->ttport;
```
This bug has been latent for ages - years, in fact. It's baffling why there aren't more people periodically tripping over this, because if you ever plug a full speed device into a single-TT hub on anything other than port 1, you're going to have a bad time.

FD-, 2019-01-15:
Wow, this is great news!
I can confirm the fix works for my HUB+Lan dongle. I wonder how @6by9's dongle was working when this issue likely affected it as well - does it probably have the ethernet adapter attached to port 1?

You did a fantastic job @P33M ! Thanks for your efforts!

FD-, 2019-01-15:
@P33M AFAICT, this particular problem (with the HUB+Lan dongle) did occur in 4.19 and 4.14 alike. I'm afraid that likely means @mypiandrew's issue isn't the same then - fingers crossed your fix works for both problems nevertheless!

P33M, 2019-01-15:
If a single full-speed device is connected to port 1 then the issue doesn't appear. @mypiandrew's hardware has a full-speed device connected to port 2, as does your adapter - which is what makes the discrepancy between 4.19 (working) and 4.14 on the myPi board odd.

@mypiandrew can you post dmesg logs in the failing case with 4.14?

FD-, 2019-01-15:
@P33M I'm afraid I don't know enough about USB to understand the first paragraph, but:

I did research on the SL2.2s and am pretty sure it's not a Terminus Tech product. In fact, I found the datasheet (in Chinese) and a Google translation of its first page (I think) said something like "The first hub IC developed in China". I believe they just chose the name SL2.2s and fake the VID/PID so customers (and existing drivers?) were fooled into believing it was a product of an established brand.

Interestingly, there also are FE1.1s clones that seem to more closely mimic the original. These are labelled "FE1.1 USB 2.0 Hub" (without a "T" Terminus Tech logo). I found one of these in another USB+Ethernet dongle (pictures in the [forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=225509&p=1415785#p1415785)).

Lastly, it also seems like counterfeit FE1.1s chips exist that carry the original silk screen but are in fact clones.

Just to highlight behaviour could differ even for the seemingly same chip.

ickromwerk, 2019-01-15:
@P33L, does it mean that" Terminus Technology 1a40:0101" hub with the suspicious marking of SL2.2s on the chip requires additional driver to work correctly?

mypiandrew, 2019-01-16:
Hi @P33M
Interesting find regarding the one line bug, I did some googling on this and found this related thread which may be of interest.

https://www.spinics.net/lists/linux-usb/msg135414.html

I have found in the past that the “see a problem, fix a problem” fault-finding mantra is especially important with software as all too often a small change/fix in one place has a knock-on effect elsewhere.

Talking around the problem specifically relating to our hardware for a moment, I would be interested to see the result of this change on 4.14 and if it provides a consistent fix across multiple (re)boots. I say this because we found that the error can present intermittently on 4.14, whereas 4.9.80 and 4.19 kernels either side are consistently good regardless of how or how often the unit is (re)booted.  To me this implies some kind of race-condition or some small delays added elsewhere are perhaps impacting the issue (I believe we’ve speculated on this before on other threads)

For interest if I look back I can see we ran into a very similar looking USB problems when going from kernel 4.1 to 4.4 (see github issue #1633) and again looking at some internal email history with 4.9.19 which was “fixed” somewhere around 4.9.36.

If you still need them I’ll upload some logs again for a little later today for 4.14, but it will be similar to those I’ve uploaded before. 

Also if it would help with productivity let me know if you need another Compute Module shipped over (preloaded with 4.14 OS)

P33M, 2019-01-16:
I would suspect that the broadcast behaviour of the hub chips is going to be in "undefined behaviour" territory - what if port 1 is enabled as a high-speed port, and we're trying to talk to port 2 as a full-speed port? Order of operations may be important here - there is no defined sequence in which the kernel will probe USB devices as they are all done in an asynchronous, multi-threaded manner. 

I wouldn't be surprised if 4.14 suddenly starts working (across multiple versions) with this fix backported to each one.

mypiandrew, 2019-01-16:
what kernel version has this fix in?

mypiandrew, 2019-01-16:
Linux raspberrypi 4.14.93-v7+ #1189 SMP Mon Jan 14 16:52:29 GMT 2019 armv7l GNU/Linux

popcornmix, 2019-01-16:
That kernel is older than the fix. Try updating now - latest rpi-update kernel includes the fix.

mypiandrew, 2019-01-16:
Result :
https://gist.github.com/mypiandrew/dccb6043e7036d28b81c1cb4a1986d1d

ickromwerk, 2019-01-16:
Confirm to work at latest rpi-update kernel with sl2.2 hub

This step brings pi to the next level. All thanks to the mighty P33L)

mypiandrew, 2019-01-16:
That's interesting, well spotted.

Let me do 10 or so reboots in a row and let's see what the results give us

mypiandrew, 2019-01-16:
Here you go

https://gist.github.com/mypiandrew/fe4484686c3952b29d8da9f177407b76

mypiandrew, 2019-01-16:
This might be better as a wider number of pass/fail

https://gist.github.com/mypiandrew/ad9792635d8c7b370578185ef23eb4ae

Does indeed look like time index is a key factor

P33M, 2019-01-16:
In that case, can you make it always succeed by fiddling with the cmdline.txt parameter boot_delay? It's measured in milliseconds and should slow down the kernel's early booting. I'd suggest starting at 100 and seeing what difference that makes. Note that this slows down every single printk() call up until the kernel switches away from the boot console, so 100ms will add ~20s to the boot time.

mypiandrew, 2019-01-16:
@P33M 

In addition to the above 

In case it's useful we've done some testing using cards we have with the GL850G (STT) and the GL852GT parts on (same card, different hub chip variants, the GL852GT is extended temp)

GL850G (STT) -- 4.14.93-v7+ -- boot_delay=8 idVendor=05e3, idProduct=0608
https://gist.github.com/mypiandrew/e2bdd32038c7497cc4eb62eb26efcd3c

GL852GT (MTT) -- 4.14.93-v7+   idVendor=05e3, idProduct=0610
https://gist.github.com/mypiandrew/d05bcbba1361afece865a2f6b23c078d

GL852GT (MTT) -- 4.14.93-v7+ -- boot_delay=8    idVendor=05e3, idProduct=0610
https://gist.github.com/mypiandrew/fd17862d534e0da3fd3ebb9682a630d4

So it appears both Genesys Hub types have the same issue and bumping the detect time stamp on a bit is critical for some reason. Where exactly the delay is needed I can't say for sure.

JamesH65, 2019-07-31:
@P33M Are we happy that the fixes mentioned above are all in place and have fixed this issue?

@jkuek I doubt we will have the time to do any back porting to such an old kernel version.

P33M, 2019-07-31:
I fixed the OP's issue. If any other respondents have specific bugs still present, then they should open new issues.

closed 2019-07-31 by P33M