raspberrypi/linux #2735: raspberrypi_axi_mon (probably) not working on 4.19

6by9, 2018-10-30:
@JamesH65 
I happened to have ```dtparam=axiperf=on``` in config.txt and booted a freshly compiled 4.19 kernel.
The kernel log is getting spammed with 
```
[ 1359.204120] WARNING: CPU: 0 PID: 172 at /home/dave/Pi/linux/drivers/firmware/raspberrypi.c:161 rpi_firmware_property+0xf4/0x10c
[ 1359.204122] Modules linked in: binfmt_misc brcmfmac brcmutil evdev vc4 drm_kms_helper drm sha256_generic drm_panel_orientation_quirks snd_soc_core snd_compress cfg80211 snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt rfkill fb_sys_fops raspberrypi_hwmon hwmon bcm2835_v4l2(C) bcm2835_codec(C) videobuf2_dma_contig v4l2_common v4l2_mem2mem videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) videodev snd_pcm media snd_timer i2c_bcm2835 snd vc_sm_cma(C) raspberrypi_axi_monitor uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
[ 1359.204222] CPU: 0 PID: 172 Comm: rpi-axiperfmon Tainted: G        WC        4.19.0-v7+ #16
[ 1359.204226] Hardware name: BCM2835
[ 1359.204234] [<801110b0>] (unwind_backtrace) from [<8010ca7c>] (show_stack+0x20/0x24)
[ 1359.204241] [<8010ca7c>] (show_stack) from [<807bd850>] (dump_stack+0xc8/0x114)
[ 1359.204247] [<807bd850>] (dump_stack) from [<8011f4e0>] (__warn+0xf4/0x120)
[ 1359.204255] [<8011f4e0>] (__warn) from [<8011f5f8>] (warn_slowpath_null+0x4c/0x58)
[ 1359.204262] [<8011f5f8>] (warn_slowpath_null) from [<8065ef34>] (rpi_firmware_property+0xf4/0x10c)
[ 1359.204273] [<8065ef34>] (rpi_firmware_property) from [<7f112450>] (monitor_thread+0x23c/0x304 [raspberrypi_axi_monitor])
[ 1359.204282] [<7f112450>] (monitor_thread [raspberrypi_axi_monitor]) from [<8013f0ec>] (kthread+0x12c/0x168)
[ 1359.204290] [<8013f0ec>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[ 1359.204293] Exception stack(0x9ccddfb0 to 0x9ccddff8)
[ 1359.204296] dfa0:                                     00000000 00000000 00000000 00000000
[ 1359.204301] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1359.204306] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 1359.204315] ---[ end trace 1fd6b4a792199b5b ]---
[ 1359.204318] ------------[ cut here ]------------
```
Line 161 reads
```
	if (WARN_ON(buf_size > sizeof(data) - sizeof(*header)))
		return -EINVAL;
```
so something seems to be off on buffer sizing. I've not investigated any further.

lategoodbye, 2018-10-30:
Just a guess, maybe this a1547e0bca519b92b09ac5530c6accb7c3f69d1a causing this regression.

6by9, 2018-10-30:
Thanks @lategoodbye, that would be a valid candidate as this is asking for 11 u32's.
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/perf/raspberrypi_axi_monitor.c#L239

JamesH65, 2018-10-31:
Looks like it to me. A simple change to 48 rather thn 32 on the max lengths i think would suffice but I need to test it, and I don't have a 19 tree up yet. Will set it up today.

JamesH65, 2018-10-31:
Bumping to 48 from 32 and I do not see any exception stack traces in the log, so looks like @lategoodbye was absolutely spot on. I'll do a PR.