raspberrypi/linux #2741: HifiBerry DAC+ Pro pops on clock source change

dpapavas, 2018-11-04:
The HifiBerry DAC+ Pro board generates a full-scale pop (i.e. the intesity of the pop doesn't depend on the DAC's volume setting) when the clock source is changed.  The pop doesn't depend on the amplifier connected to the DAC, as I get it with both my amplifiers and it even shows up on a DMM connected to the output.  It should be reproducible with the following script:

```python
import alsaaudio, time, random

d = alsaaudio.PCM()
p = True

while True:
    time.sleep(1)

    f = p and 48000 or 44100
    n = p and 2 or 1
    m = p and alsaaudio.PCM_FORMAT_S16_LE or alsaaudio.PCM_FORMAT_S24_LE

    d.setrate(f)
    d.setchannels(n)
    d.setformat(m)

    if random.random() > 0.25:
        print "pop!"
        p = not p
```

(Note that the auto-mute feature should be disabled in the mixer, as it masks the problem, as will be explained below and ,again, be warned that the pop loudness doesn't depend on the volume level setting of the DAC, and it will be *very* loud unless you have some sort of external volume control after the DAC that you can turn down.)

The pop seems to originate in the clock source change in function `snd_rpi_hifiberry_dacplus_select_clk` of `sound/soc/bcm/hifiberry_dacplus.c` and only happens when the clock source actually changes, in which case the current clock is powered down and another powered up, introducing a discontinuity in the external clock while the DAC is running.  There's no pop when the call to this function results in the same clock remaining active, or when the DAC is used in slave mode (where the external clocks aren't used).  The pop can also be avoided by either commenting out the clock switching code or the code that sets the clock dividers in the `pcm512x` codec's `hw_params` callback, indicating that the clock error is somehow related to either the clock error caused by the switch, or the required reconfiguration of the DAC.

In any case, the pop can be avoided in two ways:
1. It doesn't happen when the DAC is put in standby mode (via setting the `RQST` bit of the `POWER` register) before updating the clock source, followed by re-setting it to normal mode afterwards.
2. Alternatively the DAC can be muted (via setting the `RQML | RQMR` bits of the `MUTE` register), before updating the clock source and unmuted afterwards.  (Since the default setting, is to auto-mute the output after 21ms of inactivity, the pop is avoided most of the time in practice, but not always.)

This would have been straightforward enough, and I would have created a pull request but the issue is compounded by the following:

The `pcm512x` codec, which is used by the HifiBerry DAC+ Pro driver has a problem with popping and clicking in general, because it doesn't implement the `digital_mute` operation.  Most of the clicks and pops are relatively low volume, except for the special case described here, but are are annoying nonetheless.  I have a patch implementing the `digital_mute` operation, which I intend to submit upstream, but I have to wait for the 4.19 kernel to become available for the Pi, as I currently can't compile the latest version of the `pcm512x` codec with the current 4.14 kernel.

With the patched `pcm512x` codec the DAC seems to be muted when the `snd_rpi_hifiberry_dacplus_select_clk` function is called, which fixes the problem, so one could ask whether perhaps that would be the correct fix, making a fix inside `snd_rpi_hifiberry_dacplus_select_clk` unnecessary.  I'm not sure, as I can't really tell whether the mute introduced at the codec level is guaranteed to coincide with the DAC+'s clock source change, or whether it's a coincidence that may cease to be the case in the future, reintroducing the pops.

Since changing the clock source while the DAC is running creates a problem, I would think it reasonable to fix this problem at its source, regardless of whether it happens to be fixed elsewhere.  The fix doesn't seem to have any adverse effects in practice, but has the following potential problem:  Both the operation mode of the DAC (i.e. standby/normal mode) and, assuming the `digital_mute` patch is accepted, the mute state, are controlled by the `pcm512x` codec driver, in response to requests by ALSA to set the DAC's power/mute state.  Since the power down/mute, change clock source, power up/unmute sequence isn't atomic, as far as I can see, race conditions might potentially be introduced, howere unlikely in practice.

Perhaps there's a lock that can be acquired, to prevent ALSA from changing the power/mute state while the clock source is changed, but I haven't been able to find anything concrete.  Hopefully someone with more experience with this sort of thing will have a better idea on what the proper course of action is.  I'd be happy to create and submit a PR with the proper fix, if one is considered necessary.

pelwell, 2018-11-05:
The rpi-4.19.y branch already exists and is used for testing by early adopters. Feel free to submit a Pull Request as soon as you have a patch.

dpapavas, 2018-11-05:
I will.  I just considered waiting until the 4.19 branch is "released" (i.e. available via apt), as I only have one Pi, which I use daily as a media player and didn't feel like risking messing up the system using `BRANCH=next rpi-update`.

In any case, the patch considered here is orthogonal to the patch in the `pcm512x` codec, which requires 4.19.  I'm just not sure whether the patch can be implemented in a "proper" way, i.e. one that doesn't potentially race with the `pcm512x` driver for control of certain registers.

pelwell, 2018-11-05:
So you're waiting until everyone is using 4.19 before introducing potentially breaking changes? That's the wrong way round. We want the bugs ironed out (as much as that's ever possible) before it becomes used more widely. SD cards are so cheap now that its worth having a dedicated test card to avoid breaking things you rely on.

dpapavas, 2018-11-05:
Well, I assumed there'd be other kernel releases after the first 4.19 release, where the patch could be tested.  I also intended to test the patch for a period via my daily use of a Pi, which would make using a separate system a bit of a nuisance.  Nevertheless, I'll admit to being lazy and see about copying my system to a new card, so that I can submit the other patch.

So, am I to take it that the fix considered here needs to wait for the other patch (which it doesn't depend on) or is it considered entirely unnecessary?

popcornmix, 2019-02-05:
I tried cherry picking the upstream patch and it applied cleanly (I think thanks to git rerere and previously rebasing on 5.0 kernel) and builds so I've pushed it. Let me know if it looks ok.

dpapavas, 2019-02-05:
Thanks for your quick response.  It looks ok but you'll also have to cherry pick [this](https://patchwork.kernel.org/patch/10740163/) follow-up commit as well (sorry, I didn't expect you to apply the patch, and didn't think to mention it sooner).  I tried `git am`-ing it, to test the result, but it won't apply, so I'd best wait until you've picked it, to make sure I'll test the code that actually gets commited.

popcornmix, 2019-02-05:
Applied cleanly for me. I've pushed - can you confirm it works?

dpapavas, 2019-02-06:
Ok, I compiled and tried it out and it seems to work fine (both with respect to the intended fix and the sound card behavior in general).  Many thanks for reviewing and applying the patches.

Are we then going to see about the second set of patches ([[1]](http://mailman.alsa-project.org/pipermail/alsa-devel/2019-January/144795.html) and [[2]](http://mailman.alsa-project.org/pipermail/alsa-devel/2019-January/144794.html)) when they make it into the next kernel?  (Please don't merge them yet, as they require matching changes in other drivers).

popcornmix, 2019-02-07:
If patches are merged upstream that would be useful (and have been tested downstream), then let us know what to cherry-pick, or even better submit a pull request.

dpapavas, 2019-02-11:
Ok, I'll close this for now then.  Thanks again for your help.

closed 2019-02-11 by dpapavas