raspberrypi/linux #2743: rpi-4.19 fails to boot on arm64

eccgecko, 2018-11-09:
**Is this the right place for my bug report?**
I know arm64 nor arch is not officially supported, but I feel like this needs to be reported nonetheless and didn't know where else best to do that.

**Describe the bug**
rpi-4.19.y does not boot on arm64/aarch64.

**To reproduce**
Try building 4.19.y kernel for arm64 pi and booting.

**Expected behaviour**
Raspberry Pi 3B+ should boot into aarch64 arch linux os on 4.19.y kernel.

**Actual behaviour**
Tried building the 4.19 kernel for use on arm64 architecture. I am running arch on my pi 3b+. When making the baseline config, using the command `make arch=arm64 bcmrpi3_defconfig` I am receiving the following errors:

`arch/arm64/configs/bcmrpi3_defconfig:472:warning: override: reassigning to symbol USB_LAN78XX`
`arch/arm64/configs/bcmrpi3_defconfig:643:warning: symbol value 'm' invalid for LIRC`
`arch/arm64/configs/bcmrpi3_defconfig:1288:warning: override: reassigning to symbol MMC_BCM2835_MMC`
`arch/arm64/configs/bcmrpi3_defconfig:1289:warning: override: reassigning to symbol MMC_SDHCI_IPROC`

However, these errors are not fatal to the build, so am unsure if these are the reason kernel fails to boot. Can continue the build using the usual methods (methods that have worked fine on my rpi 3B+ running aarch64 and using exact same commands when compiling rpi-4.14.y kernels on the pi).

`make ARCH=arm64 Image dtbs modules -j 4` works fine, building initramfs works fine, but as soon as I move the newly compiled kernel over to /boot/ as I usually do and reboot, pi fails to boot. Moving back to old 4.14.79 kernel fixes it.

**System**
Raspberry Pi 3B+
Arch Linux Arm aarch64
4.14.79-v8+

eccgecko, 2018-11-09:
Having read through https://github.com/raspberrypi/linux/commit/d622e3810022817de8b2a7c50e623cfcff30e0b2
I am wondering if it has something do with 
> ARM64 now uses the bcm2835 platform just like ARM32.

Is that possibly the reason? Do I now need to copy over a dtb for bcm2835 3b+ ?

lategoodbye, 2018-11-10:
Doesn't boot is a interpretation of some observation which you didn't describe (LED behavior, output on HDMI or serial).
Do you use an additional bootloader?

pelwell, 2018-11-12:
I just tried a clean aarch64 compile of rpi-4.19.y, producing a 4.19.1-v8+ build. It boots to the desktop without errors (except for:
```
[    1.001156] Error: Driver 'sdhost-bcm2835' is already registered, aborting...
```
which is due to having two drivers for the SDHOST interface).

The complete dmesg output can be found [here](https://gist.github.com/pelwell/b6b97f7b016e67942f5cd3ad9078d1cb).

eccgecko, 2018-11-12:
@pelwell Interesting. So I must be doing something wrong, or have some unique quirk on my system that is causing this.

May I ask, using the bcmrpi3_defconfig did your `make` throw up any of the same warnings I referenced it my first post?

pelwell, 2018-11-12:
Yes, I get exactly the same warnings:
```
arch/arm64/configs/bcmrpi3_defconfig:472:warning: override: reassigning to symbol USB_LAN78XX
arch/arm64/configs/bcmrpi3_defconfig:643:warning: symbol value 'm' invalid for LIRC
arch/arm64/configs/bcmrpi3_defconfig:1288:warning: override: reassigning to symbol MMC_BCM2835_MMC
arch/arm64/configs/bcmrpi3_defconfig:1289:warning: override: reassigning to symbol MMC_SDHCI_IPROC
```
I've regenerated bcmrpi3_defconfig, also taking out CONFIG_MMC_BCM2835 to avoid the conflict with the `sdhost-bcm2835` driver. It now boots without errors.

There are quite a lot of compile-time warnings on aarch64 - if there's time I may tidy up some of them.

eccgecko, 2018-11-13:
Damn. Spoke too soon. Just rebooted using `sudo shutdown -r now` and the pi refused to boot again. At least some progress was made and I now know it is possible to boot into 4.19.1-v8 aarch64 kernel. Not sure what exactly the problem is though :(

pelwell, 2018-11-13:
I'm not using arm64 builds on a regular basis, but on the few occasions I've (re)booted into it in the last few days I've not had a problem.

pelwell, 2018-11-13:
Upstream will want to know that you are running a kernel built from a pure upstream tree, but there's no harm in asking if anybody else has seen the problem. You can also discuss the issue here, although official support is likely to be sporadic - close this issue and open a new one, so the title is more accurate.

lategoodbye, 2018-11-13:
Without a real indication (stacktrace, kernel msg) that the issue is kernel related, your chance is very little get any feedback. I usually test the mainline kernel (arm64/defconfig) on top of Arch Linux and didn't have any issues with 4.19. So i suggest to connect a HDMI monitor / serial adapter to your Pi to see what's the problem.

eccgecko, 2018-11-13:
@lategoodbye Yeah, I’ve realised that without connecting to an external monitor for feedback, the chances of getting to the bottom of this are next to zero. I will have to borrow a monitor, as right now I have nothing to connect it to at home. Thanks.

Thanks @pelwell I will do as @lategoodbye says first, and get hold of a monitor, and then report my findings either in another issue here, or upstream. I realise arm64 support is sporadic here, but I appreciate the efforts to help out. Certainly some progress was made, as I was able to actually boot, even if it isn’t clear what exactly the issue was. Thanks for all your help.

closed 2018-11-13 by eccgecko