raspberrypi/linux #2744: ethernet problem, smsc95xx module need reload

manio, 2018-11-11:
**Describe the bug**
Hi!. I am using my rpi 24h/day. From time to time I have unexpected behavior with the ethernet.
It stops working and I need to reload the USB to get it back online.
The problem are starting with the following kernel errors:

`[923287.854598] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -110`
`[923287.865405] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS`
`[923287.874604] smsc95xx 1-1.1:1.0 eth0: MII is busy in smsc95xx_mdio_read`
`[923287.884458] smsc95xx 1-1.1:1.0 eth0: Failed to read MII_BMSR`

The LAN connection is lost. No pings after this time.
18 minutes later the kernel is reporting this backtrace:
https://pastebin.com/b64mUQze

I have serial console access to the raspberry pi, so I did:
`echo 0 > /sys/devices/platform/soc/20980000.usb/buspower`
and
`echo 1 > /sys/devices/platform/soc/20980000.usb/buspower`

And this fixed the problem - after assigning the IP address the connectivity was restored.
dmesg from this "usb-restart":
https://pastebin.com/ACUNn4xz

**System**
Raspberry Pi Model B Rev 2
Raspberry Pi reference 2016-05-27

* Which firmware version (`vcgencmd version`)?
Mar 23 2018 16:31:10 
Copyright (c) 2012 Broadcom
version d92349b3d22c277e44d508a2aae6af2b1e5bb224 (clean) (release)

* Which kernel version (`uname -a`)?
Linux 10.1.0.4 4.14.31+ #1104 Thu Mar 29 16:29:19 BST 2018 armv6l GNU/Linux

lategoodbye, 2018-11-13:
Please provide `lsusb -t` before the issue happen.

manio, 2018-11-13:
@lategoodbye 
Here you are: https://pastebin.com/L5vBiTA7

manio, 2018-12-07:
I also discovered another class of similar ethernet problem:
`[2711566.193956] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -110`
`[2711566.203510] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS`
`[2711566.210835] smsc95xx 1-1.1:1.0 eth0: Timed out reading MII reg 01`
`[2711566.218563] smsc95xx 1-1.1:1.0 eth0: Failed to read MII_BMSR`

As you can see there is: '**Timed out reading MII reg 01**' instead of 'MII is busy in smsc95xx_mdio_read'.
This time the network was still working but it was a huge performance drop. The pings to the raspberry was really big compared to normal state - it was up to seconds(!!!) instead of about 1ms.
It was really hard to do something remotely over ssh. Nevertheless I was able to launch top command and I was seeing that the CPU is idle !!!

Surprisingly: I found a remedy for this situation: I had to flood the raspberry with network traffic of any kind - it doesn't matter if the linux itself is processing it or not. I mean a UDP flow is sufficient.
So I did:
`iperf -u -c rpi -t0 -b10m`
... and the pings went back to normal!

**Isn't it magic??**
What may be causing this? In other words: when the ethernet chip is busy because of input traffic, it is working ok.
Of course the above situation is going back to normal after the full raspberry pi reset.

JamesH65, 2019-07-31:
@manio Can you try the latest kernel please, IIRC there have been changes that may be relevant to you issue.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

manio, 2019-07-31:
Hi!
You're probably reading in my mind:) I just wanted to close this issue but I was waiting a little bit more to be sure if it occurs on newer kernel which I am currently testing. So far - so good. I didn't notice these two kind of problems, since I switched to kernel 4.19.56+.
I am closing it for now, I'll ofc reopen if the problem is still there...
Thanks!

closed 2019-07-31 by manio