raspberrypi/linux #2747: Can't build against kernel source of aarch64 kernel

Akito13, 2018-11-13:
Running version 4.14.80-v8+ aarch64 on Raspberry Pi 3B+. When I try to `./configure` any zfs version against the kernel source it produces the error
```
checking whether modules can be built... no
configure: error: *** Unable to build an empty module.

```

and when I run "make modules_prepare" in the kernel source then the error is as follows:
```

gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
Kbuild:21: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 1
Makefile:1084: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
```

I could omit this by cross-compiling from my computer but then I encounter a different error.

I'm out of ideas. The same ZFS versions were compiled on the current standard Raspi kernel and work.

pelwell, 2018-11-13:
aarch64 is not a supported option, but do you have an aarch64 toolchain installed on the Pi? You will probably still need to use the CROSS_COMPILE option to specify the location of the toolchain, even though it isn't really cross-compilation.

Akito13, 2018-11-13:
> 
> 
> aarch64 is not a supported option, but do you have an aarch64 toolchain installed on the Pi? You will probably still need to use the CROSS_COMPILE option to specify the location of the toolchain, even though it isn't really cross-compilation.

I tried to do this trick, but I didn't see any way of successfully installing the correct cross-compilation toolchain. Installing it on my computer was a breeze but on the Raspi it would've been an utter dependency hell.

pelwell, 2018-11-13:
I always use [crosstool-ng](https://crosstool-ng.github.io/) for building toolchains - it takes care of all the dependencies - but I haven't tried natively on the Pi.

Akito13, 2018-11-13:
> 
> 
> I always use [crosstool-ng](https://crosstool-ng.github.io/) for building toolchains - it takes care of all the dependencies - but I haven't tried natively on the Pi.

Thanks, will check that out later today.

Akito13, 2019-01-07:
> 
> 
> @Akito13 Any progress on this, can it be closed?

Thank you for kindly asking. I put the project on ice for now, because I couldn't have any success after further tries. Therefore I think it would be wise to close this issue for now and maybe reference or reopen it in the future once some progress is made. Thank you for asking though. I am used to developers having a super lose and funny _**close issue**_ extra finger that triggers every time the issue doesn't progress for a day or when the issue is "too hard" to solve.

closed 2019-01-07 by Akito13