raspberrypi/linux #2760: Missing GPIO line names for newer Raspberry Pi models

nikonthethird, 2018-11-19:
The device tree (DTS) files for newer Raspberry Pi models (e.g. Raspberry Pi 3 B) are missing the GPIO line name definitions.

For instance, [here](https://github.com/raspberrypi/linux/blob/e33ef2c4cd5dc96aa05a7d328eff61c183c94748/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts#L37) are the line names for the Raspberry Pi B.

But in [the corresponding file](https://github.com/raspberrypi/linux/blob/e33ef2c4cd5dc96aa05a7d328eff61c183c94748/arch/arm/boot/dts/bcm2710-rpi-3-b.dts) for the Raspberry Pi 3 B, there are no line names.

For reference, commit 731b26a6ac17f24057c559361c6d0cb7cb79baed introduced the line names.

The problem with the missing line names is that the `gpioinfo` utility prints this for a Raspberry Pi B:

```
gpiochip0 - 54 lines:
    line   0:       "SDA0"       unused   input  active-high
    line   1:       "SCL0"       unused   input  active-high
    line   2:       "SDA1"       unused   input  active-high
    line   3:       "SCL1"       unused   input  active-high
    line   4:  "GPIO_GCLK"       unused   input  active-high
    line   5:  "CAM_GPIO1"       unused   input  active-high
    line   6:    "LAN_RUN"       unused   input  active-high
    line   7:  "SPI_CE1_N"       unused   input  active-high
...
```

but this for a Raspberry Pi 3 B, which makes it hard to figure out which line is what:

```
gpiochip0 - 54 lines:
    line   0:      unnamed       unused   input  active-high 
    line   1:      unnamed       unused   input  active-high 
    line   2:      unnamed       unused   input  active-high 
    line   3:      unnamed       unused   input  active-high 
    line   4:      unnamed       unused   input  active-high 
    line   5:      unnamed       unused   input  active-high 
    line   6:      unnamed       unused   input  active-high 
    line   7:      unnamed       unused   input  active-high
...
```

pelwell, 2018-11-20:
The gpio-line-names feature is currently only present in the upstream DTS files (those named `bcm28*`), and even then only for a subset of models (A, A+, B, B+, Zero and ZeroW).

If it's any help, all Pis with the 40-pin header have exactly the same pin functions available on GPIOs 0-27. But beware - although pin functions are easily modified using DT overlays, the pinctrl mechanism for drivers and other utilities such as `raspi-gpio`, that list of GPIO names is completely static.

nikonthethird, 2018-11-20:
Thank you, that does help.

I was just wondering, since the sysfs interface for GPIOs will be removed in 2020, then the only available interface for GPIOs will be the character device one, if the pin numbers will be the same.

So `echo 26 > export` is basically the same as exporting line offset 26 of gpio chip 0? Looking [at the schematic](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_3b_1p2_reduced.pdf), that seems to be the case.

nikonthethird, 2018-11-20:
That solves everything for me then. 👍

Feel free to close the issue since it seems to me now that the actual line names are not that important anyways.

seamusdemora, 2022-02-18:
@pelwell ,

> The gpio-line-names feature is currently only present in the upstream DTS files (those named `bcm28*`), and even then only for a subset of models (A, A+, B, B+, Zero and ZeroW).
>
> If it's any help, [...]

I'm just beginning to explore the GPIO character device API. and discovered that `gpioinfo` provides only `unnamed` on my 3B+ (bullseye), but provides useful names on my 4B (buster). If I understand your input correctly, it seems that this is due to the fact that the pin names are absent from the dts file supplied for the 3B+... is that correct? 

If so, can you point me to any documentation that explains where to find the applicable dts file for 3B+, and how to make these changes? 

Another question wrt to the `pinctrl` mechanism: What are the implications of that if the names are added to the 3B+ dts? 

Thanks!

pelwell, 2022-02-18:
> If so, can you point me to any documentation that explains where to find the applicable dts file for 3B+, and how to make these changes?

The dts file for the 3B+ is called bcm2710-rpi-3-b-plus.dts (the current version is here: https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/bcm2710-rpi-3-b-plus.dts). How to build dtb files is covered in https://www.raspberrypi.com/documentation/computers/configuration.html#device-trees-overlays-and-parameters and https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building (`make ARCH=arm bcm2709_defconfig`, and `make ARCH=arm dtbs` should suffice), and the way to make the changes is to add the gpio-line-names property to the above .dts file and rebuild the .dtb file.

pelwell, 2022-02-18:
> Another question wrt to the pinctrl mechanism: What are the implications of that if the names are added to the 3B+ dts?

None. AFAIK.

pelwell, 2022-02-18:
Note that the line-names property can be copied from the upstream file bcm2837-rpi-3-b-plus.dts.

maxpolzin, 2022-02-18:
I am encountering the same problem. Names available on pi4, no names on pi3 b+. I am on kernel 5.10, Debian buster. Does that make a difference or can I simply download and replace my dts file with the upstream version to get the pin names.

pelwell, 2022-02-18:
> can I simply download and replace my dts file with the upstream version to get the pin names.

No - you will lose a number of features at least. The correct solution is to add the necessary GPIO names to the remaining models.

pelwell, 2022-02-20:
The line names are added by commit ac66b3f (until the next rebase, which the hash will change).

seamusdemora, 2022-02-21:
@pelwell 👍  Thanks for this! I guess the procedure is to d/l the appropriate file you've updated, and then follow the procedure you outline previously to build it?

> The dts file for the 3B+ is called bcm2710-rpi-3-b-plus.dts (the current version is here: https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/bcm2710-rpi-3-b-plus.dts). How to [...]

pelwell, 2022-02-21:
Or wait for the next firmware release.

seamusdemora, 2022-02-21:
> Or wait for the next firmware release

That sounds a wee bit simpler :)   AIUI, firmware (incl .dtb files) for all models except the 4B is updated via the apt update & upgrade process, no? 

I see the following in `/boot` on my 3B+:   
`-rwxr-xr-x 1 root root   31266 Jan 22 21:11 bcm2710-rpi-3-b-plus.dtb`

And status of firmware may be tracked from the [firmware/boot repo](https://github.com/raspberrypi/firmware/tree/master/boot)?

popcornmix, 2022-03-02:
The update is in latest rpi-update kernel.
It will appear in apt at a later date.

lategoodbye, 2022-04-11:
I try to upstream some of the fixes:
https://marc.info/?l=linux-arm-kernel&m=164970745528178&w=2