raspberrypi/linux #2766: dhcpcd.conf not working

aario, 2018-11-25:
Hi Folks,
I have edited my dhcpcd.conf like any raspberry pi tutorial on the net to have static IP on my wifi. It doens't work.

/etc/dhcpcd.conf:
```
interface wlan0
static ip_address=192.168.0.55/24
static routers=192.168.0.1
static domain_name_servers=192.168.0.1 8.8.8.8
```
```
# ifconfig wlan0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.47  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 ...
        inet6 ...
        ether ...
        RX packets 532  bytes 46101 (45.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 450  bytes 73107 (71.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
And yes. I already tried restart. I tried to reverse engineer how the configuration tool works but it's a huge shell script and very little comments within the code.

pelwell, 2018-11-25:
This isn't really the correct forum to discuss this because a) it isn't a kernel issue, and b) because it is probably a configuration error on your part. But let's run with it to shorten the process...

I think your dhcpcd.conf is being ignored. What does `systemctl status dhcpcd.service` report?

aario, 2018-11-25:
Thanks for quick respond.
```
● dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dhcpcd.service.d
           └─wait.conf
   Active: active (running) since Sun 2018-11-25 18:10:51 UTC; 2min 17s ago
  Process: 323 ExecStart=/usr/lib/dhcpcd5/dhcpcd -q -w (code=exited, status=0/SUCCESS)
 Main PID: 522 (dhcpcd)
   CGroup: /system.slice/dhcpcd.service
           ├─522 /sbin/dhcpcd -q -w
           └─895 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 -Dnl80211,wext

Nov 25 18:10:57 the-pi dhcpcd[522]: wlan0: probing address 192.168.0.55/24
Nov 25 18:10:58 the-pi dhcpcd[522]: wlan0: soliciting an IPv6 router
Nov 25 18:10:59 the-pi dhcpcd[522]: wlan0: Router Advertisement from fe80::...
Nov 25 18:10:59 the-pi dhcpcd[522]: wlan0: adding address 2a02:8109:88c0:1f38:fd86:8c37:54fa:8fcc/64
Nov 25 18:10:59 the-pi dhcpcd[522]: wlan0: adding route to ...::/64
Nov 25 18:10:59 the-pi dhcpcd[522]: wlan0: adding default route via fe80::...
Nov 25 18:10:59 the-pi dhcpcd[522]: wlan0: requesting DHCPv6 information
Nov 25 18:11:02 the-pi dhcpcd[522]: wlan0: using static address 192.168.0.55/24
Nov 25 18:11:02 the-pi dhcpcd[522]: wlan0: adding route to 192.168.0.0/24
Nov 25 18:11:02 the-pi dhcpcd[522]: wlan0: adding default route via 192.168.0.1
Warning: dhcpcd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
```

aario, 2018-11-25:
Ok. It was possibly me who overwrote the settings of dhcpcd.service. So that's not the case.
But the problem solved by disabling ipv6!

closed 2018-11-25 by aario