raspberrypi/linux #2775: /usr/bin/raspi-config will freese boot if rootfs is not last partition because waiting key press in non interactive mode

doublehp, 2018-12-02:
Not sure exactly how /usr/bin/raspi-config is run at first boot, but I am 99% certain it is.

In short, my pi froze on "is not the last partition. Don't know how to expand" at first boot. And I have been waiting for network to come up for 1h, untill I plugged a monitor to see this dialog bog.

I use headless pis; a lot of them, and I don't have time to plug a monitor on them each time I install them. I just burn the SD card, insert the card, plug network and USB, and a few minutes later, I try network connexion.

I don't burn the cards manually; I burn them with a home made script which brings many modifications into rootfs after dd. In particular, I alter etc/hostname, crontab and network/interfaces. Since yesterday, I made my script also add a third partition at the end of the card. This way, when the filesystem crashes, I want to see which file system crashes: rootfs, or my data. 

The message I got was 
`whiptail --msgbox "$ROOT_PART is not the last partition. Don't know how to expand" 20 60 2`

which belongs to function
`do_expand_rootfs`

which is called by
```
#
# Command line options for non-interactive use
#
for i in $*
do
  case $i in
[...]
  --expand-rootfs)
    INTERACTIVE=False
    do_expand_rootfs
    printf "Please reboot\n"
    exit 0
    ;;
[...]
```

=> if it's a non interactive section, it should not call a function that is waiting life time for pressing a key.

The proposed fix is ... to add this condition before all whiptail lines:
`
[ "INTERACTIVE" = True ] && whiptail ... || echo "Da error message" >&2`

This way, message will still be reported in syslogs (which I read).

This is a very quick list of lines that are probably concerned by this bug (whiptail lines not preceeded by INTERACTIVE test) :
```

# grep -n whiptail usr/bin/raspi-config -B1 | grep -v INTERACTIVE | grep -e "^--" -A1 | grep whiptail | grep -e "--msgbox"
189:    whiptail --msgbox "Root partition has been resized.\nThe filesystem will be enlarged upon the next reboot" 20 60 2
286:    whiptail --msgbox "Display overscan compensation is $STATUS" 20 60 1
333:          whiptail --msgbox "Could not read current screen dimensions - unable to enable pixel doubling" 20 60 1
372:    whiptail --msgbox "Pixel doubling is $STATUS" 20 60 1
434:      whiptail --msgbox "No wireless interface found" 20 60
441:      whiptail --msgbox "Could not communicate with wpa_supplicant" 20 60
465:        whiptail --msgbox "Wi-fi country set to $COUNTRY" 20 60 1
477:    whiptail --msgbox "\
528:          whiptail --msgbox "Device-specific memory settings were found. These have been cleared." 20 60 2
575:    whiptail --msgbox "\
636:    whiptail --msgbox "Set overclock to preset '$1'" 20 60 2
646:    whiptail --msgbox "Set overclock to preset 'None'" 20 60 2
685:    whiptail --msgbox "The SSH server is $STATUS" 20 60 1
723:    whiptail --msgbox "The VNC Server is $STATUS" 20 60 1
764:    whiptail --msgbox "The SPI interface is $STATUS" 20 60 1
810:    whiptail --msgbox "The ARM I2C interface is $STATUS" 20 60 1
897:      whiptail --msgbox "The serial login shell is $SSTATUS\nThe serial interface is $HSTATUS" 20 60 1
1025:    whiptail --msgbox "Waiting for network on boot is $STATUS" 20 60 1
1048:      whiptail --msgbox "The splash screen is not installed so cannot be activated" 20 60 2
1094:    whiptail --msgbox "Splash screen at boot is $STATUS" 20 60 1
1136:    whiptail --msgbox "Remote access to the GPIO server is $STATUS" 20 60 1
1187:    whiptail --msgbox "The camera interface is $STATUS" 20 60 1
1228:    whiptail --msgbox "The one-wire interface is $STATUS" 20 60 1
1332:    whiptail --msgbox "Predictable network interface names are $STATUS" 20 60 1
1443:      whiptail --msgbox "No wireless interface found" 20 60
1450:      whiptail --msgbox "Could not communicate with wpa_supplicant" 20 60
1520:      whiptail --msgbox "Failed to set SSID or passphrase" 20 60
```

**To reproduce**

allocate a third partition after dd, before first boot:

`echo -e "n\np\n3\n13000000\n\nw\n" | fdisk "/dev/sdb"`

**Expected behaviour**

first boot should complete without requirering keyboard interaction, whatever the issue.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

Not sure, Pi2 or Pi3

* Which OS and version (`cat /etc/rpi-issue`)?

```
2018-11-13-raspbian-stretch-lite.zip

cat etc/rpi-issue 
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2
```

pelwell, 2018-12-02:
If you are modifying rootfs, isn't it easier to just prevent the auto-resize from running? A very brief search found this page: https://raspberrypi.stackexchange.com/questions/47773/disable-auto-file-system-expansion-in-new-jessie-image-2016-05-10

doublehp, 2018-12-02:
Right, I was half wrong: the culprit is not raspi-config, but usr/lib/raspi-config/init_resize.sh 

The root issue remains the same: some init/boot script locks the whole boot of the machine because of some whiptail waiting for a keypress, and this is not acceptable on an rPi which must be presumed to not have any monitor.

Yes your link gives several solution; but still:
- init_resize.sh  should not lock the boot
- raspi-config expects keystrokes in non interactive mode

XECDesign, 2018-12-03:
The reason it is the way it is is to flag up that there's a problem and make sure that you're aware of it. If there's no error, it continues without requiring any input. It's working as intended.

pelwell, 2018-12-03:
The whole reason to have filesystem resizing is to allow a single image to be used on cards of different sizes. If you have another partition after the rootfs then why wouldn't you make the rootfs large enough to fill the gap? You must know how big the space is because your build system created the partition that marks the end of it.

XECDesign, 2018-12-03:
The script is only intended to run on stock Raspbian images. If it's running on an unsupported image, that's an issue that's worth flagging up to the user.

doublehp, 2018-12-03:
I disagree; but let's close because you are obviously not going to change your mind.

I am not arguing about partitions; I am arguing about producing a message that requires interaction. There is a very simple and smooth solution:

change /usr/lib/raspi-config/init_resize.sh

```
if main; then
  whiptail --infobox "Resized root filesystem. Rebooting in 5 seconds..." 20 60
  sleep 5
else
  sleep 5
  whiptail --msgbox "Could not expand filesystem, please try raspi-config or rc_gui.\n${FAIL_REASON}" 20 60
fi
```

into

```
if main; then
  whiptail --infobox "Resized root filesystem. Rebooting in 5 seconds..." 20 60
  sleep 5
else
  whiptail --infobox "Could not expand filesystem, please try raspi-config or rc_gui.\n${FAIL_REASON}" 20 60
  sleep 60
fi
```

If user has a monitor, he is informed; if he does not have a monitor, boot will continue.

closed 2018-12-03 by doublehp

pelwell, 2018-12-03:
> If user has a monitor, he is informed; if he does not have a monitor, boot will continue.

Leaving him totally uninformed about the fact that he may have a tiny rootfs. Our way is better.