raspberrypi/linux #2776: Missing i2c-0/1

ggardet, 2018-12-03:
**Describe the bug**
I am trying to use an upstream kernel with i2c-0/1, but I have only i2c-2, so I guess an i2c overlay is missing from ```arch/arm/boot/dts/overlays```?

**To reproduce**
Boot an upstream kernel with firmware provided DT and try to access i2c-0/1

**Expected behaviour**
i2c-0/1 should be accessible, at least with an overlay.

**Actual behaviour**
No i2c-0/1 available

**System**
openSUSE Tumbleweed

* Which model of Raspberry Pi? Pi1 - B
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)? firmware from 2018-11-04 16:31
* Which kernel version (`uname -a`)? 4.19.4-1-default

**Logs**
none

**Additional context**

pelwell, 2018-12-03:
On a Pi 1 Model B, I2C1 is intended for the use of the VPU, but if you aren't using a camera or external display it's probably safe to use both.

In config.txt add the following:
```
dtparam=i2c_arm=on
dtparam=i2c_vc=on
```

ggardet, 2018-12-03:
It did the trick! Thanks. The problem was on my side. Sorry for the noise.

ggardet, 2018-12-03:
@pelwell well, it shows up in Linux, but actually it detects no device whereas there is one connected. 
Would I need more options?
i2c-2 works fine and returns me the screen i2c infos.

pelwell, 2018-12-03:
What does `raspi-gpio | grep -i i2c` return?

ggardet, 2018-12-03:
Here is the output of ```cat /sys/kernel/debug/pinctrl/20200000.gpio-pinctrl-bcm2835/pinmux-pins```
```
Pinmux settings per pin
Format: pin (name): mux_owner gpio_owner hog?
pin 0 (gpio0): 20205000.i2c (GPIO UNCLAIMED) function alt0 group gpio0
pin 1 (gpio1): 20205000.i2c (GPIO UNCLAIMED) function alt0 group gpio1
pin 2 (gpio2): 20804000.i2c (GPIO UNCLAIMED) function alt0 group gpio2
pin 3 (gpio3): 20804000.i2c (GPIO UNCLAIMED) function alt0 group gpio3
pin 4 (gpio4): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 5 (gpio5): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 6 (gpio6): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 7 (gpio7): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 8 (gpio8): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 9 (gpio9): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 10 (gpio10): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 11 (gpio11): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 12 (gpio12): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 13 (gpio13): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 14 (gpio14): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 15 (gpio15): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 16 (gpio16): (MUX UNCLAIMED) pinctrl-bcm2835:314
pin 17 (gpio17): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 18 (gpio18): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 19 (gpio19): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 20 (gpio20): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 21 (gpio21): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 22 (gpio22): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 23 (gpio23): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 24 (gpio24): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 25 (gpio25): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 26 (gpio26): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 27 (gpio27): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 28 (gpio28): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 29 (gpio29): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 30 (gpio30): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 31 (gpio31): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 32 (gpio32): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 33 (gpio33): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 34 (gpio34): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 35 (gpio35): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 36 (gpio36): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 37 (gpio37): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 38 (gpio38): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 39 (gpio39): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 40 (gpio40): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 41 (gpio41): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 42 (gpio42): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 43 (gpio43): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 44 (gpio44): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 45 (gpio45): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 46 (gpio46): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 47 (gpio47): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 48 (gpio48): 20202000.mmc (GPIO UNCLAIMED) function alt0 group gpio48
pin 49 (gpio49): 20202000.mmc (GPIO UNCLAIMED) function alt0 group gpio49
pin 50 (gpio50): 20202000.mmc (GPIO UNCLAIMED) function alt0 group gpio50
pin 51 (gpio51): 20202000.mmc (GPIO UNCLAIMED) function alt0 group gpio51
pin 52 (gpio52): 20202000.mmc (GPIO UNCLAIMED) function alt0 group gpio52
pin 53 (gpio53): 20202000.mmc (GPIO UNCLAIMED) function alt0 group gpio53
```

pelwell, 2018-12-03:
I think your hardware or wiring is faulty. If you have a spare card,  download a Raspbian Lite image and, with the same config.txt settings, see if i2cdetect finds anything.

gerwin-git, 2018-12-03:
Nö

pelwell, 2018-12-03:
This is no longer a Linux kernel issue.

closed 2018-12-03 by pelwell