raspberrypi/linux #2778: add CONFIG_F2FS_FS_SECURITY=y

vch42, 2018-12-06:
I think the CONFIG_F2FS_FS_SECURITY=y option should be enabled in the kernel.

It enables support for the XATTR **security** namespace, which atm is not supported due to this option currently not being set to yes.
The xattr security namespace is used by SELinux and is also a requirement for Samba4 AD/DC full support.

We already have XATTR and POSIX_ACL support enabled for F2FS, with **user** namespace support, it only makes sense to fully support XATTR on F2FS by also enabling CONFIG_F2FS_FS_SECURITY.

Thank you.

JamesH65, 2019-01-07:
Do you have any numbers for the impact this has on memory requriements and overall CPU performance impact?

JamesH65, 2019-08-01:
Did a quick recompile with this enabled - adds 352 bytes on the Pi4 zImage. Looking at the code, cannot see if having much of a effect on the performance, and it's only used when f2fs is being use I believe. So  seems like an OK addition.

JamesH65, 2019-08-01:
@popcornmix @pelwell

popcornmix, 2019-08-01:
Okay.

JamesH65, 2019-09-19:
@dom Not yet enabled - do we intend to do so?

tc287, 2021-10-15:
If f2fs is intended to be usable for root filesystem, it makes sense for it to have feature parity with ext4 where possible. Currently, CONFIG_F2FS_FS_SECURITY appears to be the only config option missing:
```
# grep -C0 -e F2FS -e EXT4 /usr/src/linux-headers-5.10.17-v8+/.config 
CONFIG_EXT4_FS=y
CONFIG_EXT4_USE_FOR_EXT2=y
CONFIG_EXT4_FS_POSIX_ACL=y
CONFIG_EXT4_FS_SECURITY=y
# CONFIG_EXT4_DEBUG is not set
--
CONFIG_F2FS_FS=y
CONFIG_F2FS_STAT_FS=y
CONFIG_F2FS_FS_XATTR=y
CONFIG_F2FS_FS_POSIX_ACL=y
# CONFIG_F2FS_FS_SECURITY is not set
# CONFIG_F2FS_CHECK_FS is not set
# CONFIG_F2FS_IO_TRACE is not set
# CONFIG_F2FS_FAULT_INJECTION is not set
# CONFIG_F2FS_FS_COMPRESSION is not set
```

As a specific example, F2FS_FS_SECURITY enables the use of `setcap` to grant capabilities to binaries (such as CAP_NET_RAW) instead of requiring them to be suid-root. This is used by the default-installed `ping` (if you update/`--reinstall`; I guess the imaging process doesn't support capabilities?):
```
# getcap /bin/* /sbin/*
/bin/fping = cap_net_raw+ep
/bin/mtr-packet = cap_net_raw+ep
/bin/ping = cap_net_raw+ep
```

streaps, 2022-01-26:
Very good. I've just upgraded to the latest kernel (raspberrypi-kernel 1:1.20220120-1) and it is working as expected:

```
$ ./misdn_info
Cannot open mISDN due to Operation not permitted. (Does your Kernel support socket based mISDN?)

$ sudo setcap cap_net_raw+ep misdn_info 
$ getcap misdn_info 
misdn_info cap_net_raw=ep

$ ./misdn_info

Found 1 port
  Port  0 'HFC-S_USB.1':     TE/NT-mode BRI S/T (for phone lines & phones)
                              2 B-channels: 1-2
                                B-protocols: RAW HDLC X75slp
```

closed 2022-01-26 by pelwell