raspberrypi/linux #2787: Why I have kernel's messages spam on all my tty's?

zervankoru, 2018-12-22:
Hello. I am using Arch Linux ARM with this standard kernel (latest version)
If I, for example, insert usb device, kernel messages appear in my terminal. No matter what tty1-6 I am using. This problem apeared with one of my system upgdates using pacman -Syu. I haven/t got any other arm devices to check is it ALARM problem or this kernel. Can you give your comment? Is it normal?

initrider, 2018-12-22:
+1 It also affects me.

zervankoru, 2018-12-22:
For ones, what do not understand, what we are talking about, here is the example.
[image]
[image]
Here I just take out and then insert my wireless keyboard's usb key.
I know, I just cn use quiet, but I dont want to hide boot process, but I want to hide them from tty console. You can also see, while in cmdline.txt console=tty1, these messages appears on tty1, tty2 and others.

zervankoru, 2018-12-22:
And changing console to tty2 or others also doesn't change anything

zervankoru, 2018-12-22:
I will ask them, but I thought that the reason in the kernel itself. Does it affects Raspbian?

pelwell, 2018-12-22:
Not that I've noticed. If it did, I wouldn't have to use `dmesg -w` so often.

zervankoru, 2018-12-22:
I've asked on Arch Linux ARM IRC, but it seems, that no one don't know why it happens. Maybe developers of Raspbian know something? I cant find, ehere I can ask them.

pelwell, 2018-12-22:
What does `cat /proc/sys/kernel/printk` report on Arch?

zervankoru, 2018-12-23:
7417

pelwell, 2018-12-23:
The raspberrypi-sys-mods package includes [this file](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/etc.armhf/sysctl.d/98-rpi.conf), the content of which is:
```
kernel.printk = 3 4 1 3
vm.min_free_kbytes = 16384
```
Level 3 is Error level, while the 7 you see on Arch is Debug. This command should stop the console spam:
```
$ sudo echo 3 4 1 3 > /proc/sys/kernel/printk
```
On Raspbian the change is made at boot time through systemd, but Arch devs will know the best way for you.

closed 2018-12-23 by zervankoru

initrider, 2023-01-23:
It's me