raspberrypi/linux #2788: Lockup when copying files on NFS share on RBP3B+

graysky2, 2018-12-22:
**Describe the bug**
Errors and timeouts when using rsync to copy files from an NFS share to the same NFS share.
Perhaps this is related to https://github.com/raspberrypi/linux/issues/2482

**To reproduce**
* Mount the NFS share.
* Use rysnc as described below.
* Monitor with `dmesg -e -w` as the rsync job grinds on.

**Expected behaviour**
Rsync should complete without errors.
```
rsync -a --delete-after -W -x -q /scratch/armc8/root/ /scratch/armc8/facade
```
**Actual behaviour**
Rsync does not complete; dmesg reports a number of problems.

I attached the complete dmesg output in the logs section below but here is a partial output illustrating the error:
```
[  +0.000005] nfs: server ease.lan not responding, still trying
[  +1.061951] nfs: server ease.lan OK
[  +0.000007] nfs: server ease.lan OK
[  +0.000019] nfs: server ease.lan OK
[  +0.000667] nfs: server ease.lan OK
[  +0.000037] nfs: server ease.lan OK
[  +0.000036] nfs: server ease.lan OK
[  +0.000015] nfs: server ease.lan OK
[  +0.000012] nfs: server ease.lan OK
[  +0.000012] nfs: server ease.lan OK
[  +0.000012] nfs: server ease.lan OK
[Dec22 10:44] watchdog: BUG: soft lockup - CPU#2 stuck for 22s! [10.1.8.228-mana:1151]
[  +0.008170] Modules linked in: rpcsec_gss_krb5 btsdio brcmfmac rc_cec brcmutil vc4 cec rc_core nf_log_ipv4 nf_log_common ipt_REJECT drm_kms_helper xt_LOG drm microchip hci_uart cfg80211 xt_limit btqca drm_panel_orientation_quirks syscopyarea btbcm xt_addrtype btintel sysfillrect lan78xx sysimgblt fb_sys_fops bluetooth xt_conntrack raspberrypi_hwmon ecdh_generic bcm2835_thermal pwm_bcm2835 rfkill i2c_bcm2835 bcm2835_wdt ip6table_filter ip6_tables nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter bpfilter
[  +0.055229] CPU: 2 PID: 1151 Comm: 10.1.8.228-mana Not tainted 4.19.10-1-ARCH #1
[  +0.007777] Hardware name: Raspberry Pi 3 Model B+ (DT)
[  +0.005500] pstate: 20000005 (nzCv daif -PAN -UAO)
[  +0.005054] pc : nfs_sb_active+0x98/0xf0
[  +0.004133] lr : nfs_delegation_reap_unclaimed+0x3c/0x108
[  +0.005676] sp : ffff00000dea3c50
[  +0.003484] x29: ffff00000dea3c50 x28: 0000000000000000
[  +0.005592] x27: ffff800030d064c8 x26: ffff80002f5d0a40
[  +0.005592] x25: ffff0000097c96c8 x24: 0000000000000000
[  +0.005590] x23: 0000000000000000 x22: ffff800030d06400
[  +0.005591] x21: ffff800030d064c8 x20: ffff8000313aa400
[  +0.005591] x19: ffff80002f485800 x18: 0000000000000000
[  +0.005590] x17: ffff7e000082e880 x16: 00000000fffffff8
[  +0.005592] x15: 0000000000000003 x14: 1800000000000000
[  +0.005590] x13: 0000000000000000 x12: 0100000000000000
[  +0.015436] x11: 00000006756df2ec x10: 00000000000009a0
[  +0.015328] x9 : ffff00000dea3960 x8 : ffff800035e39780
[  +0.015345] x7 : 0000000000000000 x6 : 0000000000000000
[  +0.015122] x5 : ffff80002f48589c x4 : 000000000000000c
[  +0.014783] x3 : ffff8000313aa400 x2 : ffff80002f48589c
[  +0.014487] x1 : 0000000000000000 x0 : 0000000000000001
[  +0.014115] Call trace:
[  +0.010777]  nfs_sb_active+0x98/0xf0
[  +0.011860]  nfs_delegation_reap_unclaimed+0x3c/0x108
[  +0.013324]  nfs4_state_end_reclaim_reboot+0x174/0x240
[  +0.013310]  nfs4_recovery_handle_error+0x5c/0x1b0
[  +0.012956]  nfs4_do_reclaim+0x214/0x250
[  +0.012076]  nfs4_state_manager+0x438/0x960
[  +0.012373]  nfs4_run_state_manager+0x2c/0x40
[  +0.012579]  kthread+0x130/0x138
[  +0.011429]  ret_from_fork+0x10/0x1c
```
**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Model = Pi3B+
* OS and version = _Both_ Arch ARM aarch64 and armv7h
* Firmware version (`vcgencmd version`) = `Dec 17 2018 23:56:39 version da468960fe03ecbaa8e3f1ee01c7217c3bd01fa8 (clean) (release)`

Note that the firmware package is only available on the armv7h flavor; no firmware is available on aarch64.

* Kernel on armv7h = `Linux workbench 4.14.89-1-ARCH #1 SMP Tue Dec 18 14:02:18 EST 2018 armv7l GNU/Linux`
* Kernel on aarch64 = `Linux workbench 4.19.10-1-ARCH #1 SMP Tue Dec 18 19:48:51 MST 2018 aarch64 GNU/Linux`

**Logs**
[Link](https://gist.github.com/graysky2/41c66069bb847c1644d65129e82c2661) to complete dmesg.

graysky2, 2018-12-22:
I have been reviewing https://github.com/raspberrypi/linux/issues/2482 and came across [this](https://github.com/raspberrypi/linux/issues/2482#issuecomment-397492506) comment suggesting this setting: `ethtool -K eth0 tx-tcp-segmentation off` prior to the copy command.

I tested this with my rsync command above and still the bug is present.

pelwell, 2018-12-22:
If you could reproduce the lockup on a Raspbian image (or a stock Pi kernel from the firmware repo) it would help a lot.

graysky2, 2018-12-22:
@pelwell - I will try with  Raspbian Stretch Lite (November 2018) right now.

graysky2, 2018-12-22:
@pelwell - OK.  I update the system, rebooted, and successfully carried out the same test 9 times under it without errors.
```
$ uname -a
Linux workbench 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

Lots of uncontrolled variables with this comparison thought....  same hardware but different architecture and kernel versions.  I believe rsync and nfs-utils are also different versions.  In any case, where to go from here to troubleshoot?

pelwell, 2018-12-22:
The next step would be to take the kernel and modules from our firmware repo and drop them into your ARCH system. As long as the version names differ slightly the modules should coexist happily - you just need to overwrite kernel7.img to switch.

graysky2, 2018-12-22:
Is this the file you recommend I use: https://github.com/raspberrypi/firmware/archive/1.20181112.tar.gz

pelwell, 2018-12-22:
I suggest downloading the zip file from https://github.com/raspberrypi/firmware?files=1 (see Clone or Download), and using just the modules, and the kernel from /boot.

graysky2, 2018-12-22:
OK.  I downloaded and extracted the zip file:
```
/scratch/firmware-master % ls -lh
drwxr-xr-x 3 facade users  480 Dec 21 09:10 boot/
drwxr-xr-x 3 facade users   60 Dec 21 09:10 documentation/
drwxr-xr-x 2 facade users  200 Dec 21 09:10 extra/
drwxr-xr-x 3 facade users   60 Dec 21 09:10 hardfp/
drwxr-xr-x 4 facade users   80 Dec 21 09:10 modules/
drwxr-xr-x 3 facade users   60 Dec 21 09:10 opt/
```

To be clear, on the RPi3B+, you want to:
```
cp /scratch/linux-firmware-master/boot/* /boot/
```
?

graysky2, 2018-12-22:
> You only really needed to copy the kernel IMG from the boot directory. Did you copy the modules as well?

I did not copy the modules... I read your previous line too literally (ie only the things under /boot).  In any case, the test is on its 8th iteration now without an error.  

I will replace the Arch-provided /boot contents and then copy only `/scratch/firmware-master/boot/kernel7.img` over and repeat the tests.  Let me know if you disagree with this testing plan.

pelwell, 2018-12-22:
And the modules - you do need the modules.

graysky2, 2018-12-22:
> And the modules - you do need the modules.

..are you sure?  Perhaps they are the reason why my root filesystem was readonly, but even without them, I can do the rsync task 8 times in a row without an error.

pelwell, 2018-12-22:
You are clearly managing to do basic things without them, but lots of tasks may not be running which would make the test less representative.

graysky2, 2018-12-22:
> You are clearly managing to do basic things without them, but lots of tasks may not be running which would make the test less representative.

OK... where shall I copy them?  The distro default is `/usr/lib/modules/4.14.89-1-ARCH` so `/usr/lib/modules/` ?

Like:
```
cp -r /scratch/firmware-master/modules /usr/lib/modules/
```
?

pelwell, 2018-12-22:
Yes, that looks right to me.

graysky2, 2018-12-22:
@pelwell - OK!

In summary:
* Replaced the distro provided `/boot/kernel7.img` with the one from your zip file.
* Copied over `modules` to `/usr/lib/modules/` from your zip file.

Rebooting the system gave zero systemd errors.  Conducting the test now.

graysky2, 2018-12-22:
AH!  Errors are occurring:
```
[Dec22 16:03] INFO: rcu_sched self-detected stall on CPU
[  +0.000192] 	2-...: (2099 ticks this GP) idle=6de/140000000000001/0 softirq=6375/6375 fqs=1050 
[  +0.000236] 	 (t=2100 jiffies g=3111 c=3110 q=4417)
[  +0.000148] NMI backtrace for cpu 2
[  +0.000110] CPU: 2 PID: 507 Comm: 10.9.8.228-mana Not tainted 4.14.90-v7+ #1183
[  +0.000198] Hardware name: BCM2835
[  +0.000130] [<8010ff30>] (unwind_backtrace) from [<8010c174>] (show_stack+0x20/0x24)
[  +0.000224] [<8010c174>] (show_stack) from [<8078b864>] (dump_stack+0xd4/0x118)
[  +0.000213] [<8078b864>] (dump_stack) from [<807916c8>] (nmi_cpu_backtrace+0xc8/0xcc)
[  +0.000227] [<807916c8>] (nmi_cpu_backtrace) from [<807917b4>] (nmi_trigger_cpumask_backtrace+0xe8/0x124)
[  +0.000271] [<807917b4>] (nmi_trigger_cpumask_backtrace) from [<8010e578>] (arch_trigger_cpumask_backtrace+0x20/0x24)
[  +0.000296] [<8010e578>] (arch_trigger_cpumask_backtrace) from [<8018638c>] (rcu_dump_cpu_stacks+0xa8/0xd8)
[  +0.000274] [<8018638c>] (rcu_dump_cpu_stacks) from [<80185bf0>] (rcu_check_callbacks+0x878/0xa10)
[  +0.007806] [<80185bf0>] (rcu_check_callbacks) from [<8018bf28>] (update_process_times+0x40/0x6c)
[  +0.007962] [<8018bf28>] (update_process_times) from [<8019dbc4>] (tick_sched_handle+0x64/0x70)
[  +0.008338] [<8019dbc4>] (tick_sched_handle) from [<8019de04>] (tick_sched_timer+0x50/0xac)
[  +0.008305] [<8019de04>] (tick_sched_timer) from [<8018ce94>] (__hrtimer_run_queues+0x154/0x2f0)
[  +0.008609] [<8018ce94>] (__hrtimer_run_queues) from [<8018d26c>] (hrtimer_interrupt+0xcc/0x220)
[  +0.008594] [<8018d26c>] (hrtimer_interrupt) from [<8063fdfc>] (arch_timer_handler_phys+0x40/0x48)
[  +0.008704] [<8063fdfc>] (arch_timer_handler_phys) from [<8017ae4c>] (handle_percpu_devid_irq+0x88/0x23c)
[  +0.008694] [<8017ae4c>] (handle_percpu_devid_irq) from [<80175844>] (generic_handle_irq+0x34/0x44)
[  +0.008841] [<80175844>] (generic_handle_irq) from [<80175e84>] (__handle_domain_irq+0x6c/0xc4)
[  +0.008699] [<80175e84>] (__handle_domain_irq) from [<80101504>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[  +0.008782] [<80101504>] (bcm2836_arm_irqchip_handle_irq) from [<807a743c>] (__irq_svc+0x5c/0x7c)
[  +0.008698] Exception stack(0x87ee1e00 to 0x87ee1e48)
[  +0.004405] 1e00: 00000000 00000000 00005088 00005077 a1ed1280 ade8e998 ade8e800 ad6d1ea8
[  +0.008885] 1e20: ad6d1e00 ad6d1e00 ad2f982c 87ee1e64 87ee1e38 87ee1e50 802a8dd4 803d404c
[  +0.009495] 1e40: 20070013 ffffffff
[  +0.004711] [<807a743c>] (__irq_svc) from [<803d404c>] (nfs_delegation_grab_inode+0x48/0x4c)
[  +0.009530] [<803d404c>] (nfs_delegation_grab_inode) from [<803d50e8>] (nfs_delegation_reap_unclaimed+0x7c/0xd8)
[  +0.009466] [<803d50e8>] (nfs_delegation_reap_unclaimed) from [<803d0740>] (nfs4_state_end_reclaim_reboot+0x118/0x188)
[  +0.009346] [<803d0740>] (nfs4_state_end_reclaim_reboot) from [<803d0808>] (nfs4_recovery_handle_error+0x58/0x158)
[  +0.009095] [<803d0808>] (nfs4_recovery_handle_error) from [<803d1a6c>] (nfs4_do_reclaim+0x734/0x7a8)
[  +0.009083] [<803d1a6c>] (nfs4_do_reclaim) from [<803d1f10>] (nfs4_run_state_manager+0x430/0x774)
[  +0.008873] [<803d1f10>] (nfs4_run_state_manager) from [<8013ddd4>] (kthread+0x13c/0x16c)
[  +0.009005] [<8013ddd4>] (kthread) from [<801080ac>] (ret_from_fork+0x14/0x28)
```

pelwell, 2018-12-22:
Interesting.

We actually missed a step in the Raspbian test, which was to use rpi-update to get the latest kernel. If you still have the Raspbian Lite card around it would help to narrow down whether this is a straight kernel issue introduced with newer builds or whether Arch is using it in a different way.

graysky2, 2018-12-22:
> We actually missed a step in the Raspbian test, which was to use rpi-update to get the latest kernel. If you still have the Raspbian Lite card around it would help to narrow down whether this is a straight kernel issue introduced with newer builds or whether Arch is using it in a different way.

I actually ran the following on the Raspbian box:
```
# apt-get update
# apt-get upgrade
```

Is that the same?  I can boot into it and run `rpi-update` if you wish.

pelwell, 2018-12-22:
The Raspbian kernel and firmware packages are periodically updated from the firmware repo, but rpi-update always gets the most recent builds. Please run rpi-update, but first make a note of the kernel version before updating.

graysky2, 2018-12-22:
Before:
```
$ uname -a
Linux workbench 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

After:
```
$ uname -a
Linux workbench 4.14.90-v7+ #1183 SMP Fri Dec 21 14:03:50 GMT 2018 armv7l GNU/Linux
```

pelwell, 2018-12-22:
No problem. You should now have the identical kernels and modules in Raspbian and Arch.

graysky2, 2018-12-22:
On my  way out the door... 6 iterations completed so far without error on Raspbian (updated) whereas under Arch modified with these files, it could not complete a single one (see [above](https://github.com/raspberrypi/linux/issues/2788#issuecomment-449597490)....  let's assume all 21 of these are fine on Raspbian (updated).  What is the next thing to try to identify the cause under Arch?  Something in our kernel config or...?

graysky2, 2018-12-23:
@pelwell - OK.  I took the fresh Arch ARM armv7h image and replaced:
* the distro provided `/boot/kernel7.img` with the one from your zip file.
* Copied over modules/* from your zip files to /usr/lib/modules/ on the live filesystem.
* No other modifications.

It boots fine and completed 21 iterations of the rsync copy test without error.  I rebooted it and the nfs server and again, it completed 13 iterations of the rsync copy test without error (I only ran 13 this time).

Conclusion: booting with the zip file kernel and modules on an otherwise native Arch ARM armv7h image does not experience the bug.

pelwell, 2018-12-23:
That's good because the results are consistent, but it puts the ball in the Arch court - I don't think there is much I can do here.

pelwell, 2018-12-23:
Remember that, as well as a potentially student kernel commit, Arch may be using a different toolchain.

graysky2, 2018-12-23:
@pelwell - I'm sure we are building from the same commits:[for example](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-raspberrypi/PKGBUILD#L8) but you are correct about the tool chain.  Our current one is based on gcc v8.2.0.

... is there anything obvious to you in our kernel [config](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-raspberrypi/config) that might be driving this?

graysky2, 2018-12-24:
To try to get at the differences in the tool chain, I used the Arch packages from March 12th, 2017 (actually the entire system from that date) which provides gcc version 6.3.1.  I see that the Raspbian image I have is using gcc 6.3.0-18+rpi+deb9u1 and the Arch ARM version I cited is the nearest match although obviously not identical.

I built kernel version 4.14.90 using it and after I booted into it, the rsync test failed with similar errors (system was frozen so I could not ssh in to paste out the dmesg output/I didn't see anything related to the errors in the journalctl output.

lategoodbye, 2018-12-25:
@graysky2 Did you try to copy the ARCH build kernel on the Raspbian SD card?

graysky2, 2018-12-26:
@lategoodbye - Good idea.  I swapped into the Arch modules and `kernel7.img` and the test completed without error (21 iterations).  Puzzling indeed.

graysky2, 2018-12-26:
@lategoodbye - Thanks for your help.  Let me take these in order:

* I tried stripping it down to a minimal set with the same results (list shown at the bottom).
* I **cannot** reproduce this on a RPi3B on Arm ARM... it seems unique to the RPi3B+ only.
* I don't think Arch ARM offers an armv7h kernel that isn't in the 4.14.x series BUT I have confirmed the bug on Arch ARM aarch64 (4.19.10 at the time).
* To compare the config options differences, I built the Arch ARM kernel using the Raspbian config and got the same result (bug).  See [this](https://github.com/raspberrypi/linux/issues/2788#issuecomment-449663991) post for details.  

Arch ARM uses `systemd-networkd` to manage the onboard NIC whereas Raspbian uses a custom one `/lib/systemd/system/networking.service` it seems.  I need to explore the Raspbian option in more depth to see if I can recreate it on Arch ARM.

List of user-enabled systemd services:
```
% tree /etc/systemd/system
/etc/systemd/system
├── dbus-org.freedesktop.network1.service -> /usr/lib/systemd/system/systemd-networkd.service
├── dbus-org.freedesktop.resolve1.service -> /usr/lib/systemd/system/systemd-resolved.service
├── getty.target.wants
│   └── getty@tty1.service -> /usr/lib/systemd/system/getty@.service
├── multi-user.target.wants
│   ├── remote-fs.target -> /usr/lib/systemd/system/remote-fs.target
│   ├── sshd.service -> /usr/lib/systemd/system/sshd.service
│   ├── systemd-networkd.service -> /usr/lib/systemd/system/systemd-networkd.service
│   ├── systemd-resolved.service -> /usr/lib/systemd/system/systemd-resolved.service
├── network-online.target.wants
│   └── systemd-networkd-wait-online.service -> /usr/lib/systemd/system/systemd-networkd-wait-online.service
├── sockets.target.wants
│   └── systemd-networkd.socket -> /usr/lib/systemd/system/systemd-networkd.socket
├── sysinit.target.wants
│   ├── rngd.service -> /usr/lib/systemd/system/rngd.service
│   └── systemd-timesyncd.service -> /usr/lib/systemd/system/systemd-timesyncd.service
```

lategoodbye, 2018-12-26:
@graysky2 Can you point me to the Arch armv7h images ( https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3 )?

graysky2, 2018-12-26:
For armv7h: http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-2-latest.tar.gz
That will give you the 4.14.x series of kernels and the officially supported firmware.

graysky2, 2018-12-26:
If I use rysnc over ssh either either a read or write mode, I do not experience the bug.  I only get the bug if I am rsyncing from the mount to the mount.  Does that have any implications?

Below `/scratch` is the NFS mount (on the RPi3B+):
```
 if ! mountpoint -q /scratch; then
   mount 10.1.9.228:/scratch /scratch
 fi

 # read from NFS and write over ssh
 for i in {1..9}; do
   rsync -a --delete-after -W -x -q /scratch/armc8/root/ facade@10.1.9.228:/scratch/armc8/facade.$i
  done
```

```
 if ! mountpoint -q /scratch; then
   mount 10.1.9.228:/scratch /scratch
 fi

 # read from ssh and write over NFS
 for i in {1..9}; do
   rsync -a --delete-after -W -x -q facade@10.1.9.228:/scratch/armc8/root /scratch/armc8/facade.$i
 done
```

lategoodbye, 2018-12-28:
@graysky2 Are you able to make a wireshark trace (i think a minute before until 4 minutes after the lockup should be sufficient)?

graysky2, 2018-12-28:
I've never used wireshark before but I'd be happy to do it if you can send
the me stanza to provide the output you're seeking.

lategoodbye, 2018-12-30:
wireshark is a graphical to capture network traffic. There are two options where you can capture the traffic, either on the server side or on the client side. I assume you want to capture on the Raspberry Pi side because of privacy or permission. In this case i would recommend to use tcpdump as a commandline replacement for wireshark.

You can start tracing the network with following command (not recommend via SSH connection, make sure there is enough disc space)
`tcpdump -i eno1 -w trace.pcap`

With Strg+C you finish the trace. Drawback in this case is that you don't know when issue appears. So you will need to restart with every rsync attempt.

graysky2, 2018-12-31:
I will try `tcpdump` and find a way to share the output.

lategoodbye, 2019-01-01:
@graysky2 This size is expected. You only need to provide a download link and a timestamp when the issue occurred.

JamesH65, 2019-01-07:
Do you have this patch, along with turning TSO off via the module parameter? This was a fix required when transferring large files on the 3B+

https://github.com/raspberrypi/linux/commit/5762758699e1ddab22bf4c14eb225941761c52c8

JamesH65, 2019-01-07:
@graysky2 I don't know which kernel you are building, and the symptoms are similar to the ones that patch fixed, so thought I would suggest it. It may well be in your kernel already, in which case a red herring.

graysky2, 2019-01-07:
@JamesH65 - I get this bug on armv7h kernels as recent as 4.14.90.  I'm pretty sure the patch was applied months ago to the 4.14.x series.

graysky2, 2019-01-12:
For others experiencing this bug, refer to [this](https://github.com/raspberrypi/linux/issues/2788#issuecomment-450143395) comment.  For me, a work-around is to append the `--bwlimit=9000` switch to the rsync command.  For whatever reason, it _seems to_ prevent triggering this bug.

JamesH65, 2019-01-12:
That's pretty interesting, good find.

lategoodbye, 2019-01-12:
@graysky2 Still no chance to upload the tcpdump trace to a cloud service? Another possibility to reduce the filesize is to open the pcap in Wireshark and re-export only the relevant time.

phil0u, 2019-04-27:
Hi,
Let me know if you want me to open a dedicated thread, but I feel this is the same problem I'm facing:

Newly bought RBP3B+, with a AUKRU 5V3A power supply.
I'm trying to use this setup for a second Kodi media player (other one is a Pi 3, works 24/365 without a glitch)
I first tried LibreELEC distro LibreELEC-RPi2.arm-9.0.1.img then LibreELEC-RPi2.arm-8.2.5.img then I tried the latest official RASPBIAN distro: 2019-04-08-raspbian-stretch-full.img, which I'm using right now for tests.

Symptom is the same every time : any light load on the ethernet link will lockup the Pi3B+ within seconds. A simple "apt-get update" will lockup the Pi3B+

No problem if using the WLAN. So the problem is definitely linked to the lan78xx/Ethernet adapter

I've switched SD cards, same behavior.
I've read most threads I could find regarding this issue : I tried forcing the link to 10/100/1000. Lockups each time.
I disabled EEE (dtparam=eee=off), no change
I tried disabling tx-checksumming ( ethtool -K eth0 tx off), no change
I'm running latest firmware so TSO is already disabled :
rpi-update has the latest firmware.

$ uname -a
Linux raspberrypi 4.19.36-v7+ #1213 SMP Thu Apr 25 15:08:02 BST 2019 armv7l GNU/Linux

$ vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

$ cat /etc/debian_version 
9.8

I've connected the Pi3B+ to:
1/ a Power Line Communication ethernet adapter , 100Mbits -> apt-get update will lockup
2/ a ISP router -> apt-get update will lockup
3/ a managed Cisco switch with flow control enabled on the port -> apt-get update will lockup

I made a tcpdump capture using a network SPAN on the Cisco Switch, and i can see some packets being DUP ack'ed up to five times and multiple TCP window update packets.

The file being retrieved is http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages.xz and the capture stops after 5Mo (out of 11Mo)

Let me know anything I could try (new kernel, other firmware,...) in order to solve this bug, as for now, this makes my Pi3B+ totally unusable.

Thanks,

JamesH65, 2019-04-29:
@phil0u This sound like a HW fault - ethernet should simply work well out of the box with the standard Raspbian. I just tried gabbing that file on a Pi3B+ and all was fine. I'd suggest sending it back for replacement.

phil0u, 2019-05-01:
@JamesH65 Thanks for the advice. I spent some more time trying to install older LibreELEC version which included kernel fixes (after reading the following thread #2608 ), still same problem (4.14 kernel).
Tried with latest LibreELEC nightly build with 5.0.7 kernel, same problem.
Will send the unit back.
Thanks

closed 2022-02-18 by graysky2