raspberrypi/linux #2794: raspberrypi 3b kernel Internal error: Oops: 17 [#1] SMP ARM for BCM2835 #195

cxueqin, 2019-01-05:
Hi friend:
   I use raspberrypi 3B board as a  wifi station mode to connect to an AP. The wifi connection was made by wpa_supplicant with 802.1X authtication params.  The 802.1x params list below:
   * key_mgmt : WPA-EAP
   * eap: PEAP
   * proto: WPA RSN
   * pairwise: CCMP
   * group: CCMP
   * NetworkIdentity
   * password

 The wifi connection was fine at first. I used wpa_cli to reassociate the AP with the same params every two minutes. In other words, the rp 3b board test 802.1x access authentication every two minutes. After hundreds of successful reassociations, the kernel went wrong. The syslog shown

> Jan  4 07:14:40 NetFrog kernel: [21393.314945] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
> Jan  4 07:14:40 NetFrog kernel: [21393.315365] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
> Jan  4 07:14:40 NetFrog kernel: [21393.315375] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
> Jan  4 07:14:45 NetFrog kernel: [21398.610986] ------------[ cut here ]------------
> Jan  4 07:14:45 NetFrog kernel: [21398.611241] WARNING: CPU: 0 PID: 42 at net/wireless/sme.c:752 __cfg80211_connect_result+0x398/0x3e4 [cfg80211]
> Jan  4 07:14:45 NetFrog kernel: [21398.611248] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd asix uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
> Jan  4 07:14:45 NetFrog kernel: [21398.611376] CPU: 0 PID: 42 Comm: kworker/u8:1 Tainted: G        WC      4.14.90-v7+ #1183
> Jan  4 07:14:45 NetFrog kernel: [21398.611381] Hardware name: BCM2835
> Jan  4 07:14:45 NetFrog kernel: [21398.611571] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
> Jan  4 07:14:45 NetFrog kernel: [21398.611605] [<8010ff30>] (unwind_backtrace) from [<8010c174>] (show_stack+0x20/0x24)
> Jan  4 07:14:45 NetFrog kernel: [21398.611622] [<8010c174>] (show_stack) from [<8078b864>] (dump_stack+0xd4/0x118)
> Jan  4 07:14:45 NetFrog kernel: [21398.611639] [<8078b864>] (dump_stack) from [<8011dd70>] (__warn+0xf8/0x110)
> Jan  4 07:14:45 NetFrog kernel: [21398.611654] [<8011dd70>] (__warn) from [<8011de58>] (warn_slowpath_null+0x30/0x38)
> Jan  4 07:14:45 NetFrog kernel: [21398.611850] [<8011de58>] (warn_slowpath_null) from [<7f22ed4c>] (__cfg80211_connect_result+0x398/0x3e4 [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.612227] [<7f22ed4c>] (__cfg80211_connect_result [cfg80211]) from [<7f202ec4>] (cfg80211_process_wdev_events+0x10c/0x14c [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.612598] [<7f202ec4>] (cfg80211_process_wdev_events [cfg80211]) from [<7f202f40>] (cfg80211_process_rdev_events+0x3c/0x70 [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.612965] [<7f202f40>] (cfg80211_process_rdev_events [cfg80211]) from [<7f1fd294>] (cfg80211_event_work+0x24/0x2c [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.613158] [<7f1fd294>] (cfg80211_event_work [cfg80211]) from [<801379f0>] (process_one_work+0x158/0x454)
> Jan  4 07:14:45 NetFrog kernel: [21398.613173] [<801379f0>] (process_one_work) from [<80137d50>] (worker_thread+0x64/0x5b8)
> Jan  4 07:14:45 NetFrog kernel: [21398.613187] [<80137d50>] (worker_thread) from [<8013ddd4>] (kthread+0x13c/0x16c)
> Jan  4 07:14:45 NetFrog kernel: [21398.613205] [<8013ddd4>] (kthread) from [<801080ac>] (ret_from_fork+0x14/0x28)
> Jan  4 07:14:45 NetFrog kernel: [21398.613293] ---[ end trace 5eac627de54b823b ]---
> Jan  4 07:14:46 NetFrog kernel: [21399.528228] Unable to handle kernel NULL pointer dereference at virtual address 00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528253] pgd = a50b0000
> Jan  4 07:14:46 NetFrog kernel: [21399.528264] [00000000] *pgd=373d8835, *pte=00000000, *ppte=00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528293] Internal error: Oops: 17 [#1] SMP ARM
> Jan  4 07:14:46 NetFrog kernel: [21399.528304] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd asix uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
> Jan  4 07:14:46 NetFrog kernel: [21399.528432] CPU: 1 PID: 1176 Comm: wpa_supplicant Tainted: G        WC      4.14.90-v7+ #1183
> Jan  4 07:14:46 NetFrog kernel: [21399.528442] Hardware name: BCM2835
> Jan  4 07:14:46 NetFrog kernel: [21399.528453] task: b674cb00 task.stack: a790e000
> Jan  4 07:14:46 NetFrog kernel: [21399.528476] PC is at call_commit_handler.part.2+0x18/0x44
> Jan  4 07:14:46 NetFrog kernel: [21399.528490] LR is at ioctl_standard_call+0x270/0x4bc
> Jan  4 07:14:46 NetFrog kernel: [21399.528502] pc : [<80785b08>]    lr : [<80785da4>]    psr: 20000013
> Jan  4 07:14:46 NetFrog kernel: [21399.528513] sp : a790fde0  ip : a790fdf0  fp : a790fdec
> Jan  4 07:14:46 NetFrog kernel: [21399.528523] r10: b9272800  r9 : 00000034  r8 : 00008b1a
> Jan  4 07:14:46 NetFrog kernel: [21399.528535] r7 : 0000000e  r6 : a790fea0  r5 : 00000003  r4 : 00000002
> Jan  4 07:14:46 NetFrog kernel: [21399.528547] r3 : 00000000  r2 : 60000013  r1 : 60000013  r0 : b9272800
> Jan  4 07:14:46 NetFrog kernel: [21399.528560] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
> Jan  4 07:14:46 NetFrog kernel: [21399.528572] Control: 10c5383d  Table: 250b006a  DAC: 00000055
> Jan  4 07:14:46 NetFrog kernel: [21399.528584] Process wpa_supplicant (pid: 1176, stack limit = 0xa790e210)
> Jan  4 07:14:46 NetFrog kernel: [21399.528595] Stack: (0xa790fde0 to 0xa7910000)
> Jan  4 07:14:46 NetFrog kernel: [21399.528613] fde0: a790fe5c a790fdf0 80785da4 80785afc 806ab2ac a790fe6c a790fe6c a790feb0
> Jan  4 07:14:46 NetFrog kernel: [21399.528630] fe00: 00000000 a536aa80 80697fd0 636ab284 616e6968 6566696c 6669772d 8068bc69
> Jan  4 07:14:46 NetFrog kernel: [21399.528648] fe20: a790fe44 807a4150 a790fe4c a790fe38 807a4150 00008b1a a790fea0 80c7a840
> Jan  4 07:14:46 NetFrog kernel: [21399.528665] fe40: 7efe67b8 00000000 a790e000 00000036 a790fe94 a790fe60 807862e8 80785b40
> Jan  4 07:14:46 NetFrog kernel: [21399.528682] fe60: 7f249c5c 80c7a840 7efe67b8 00008b1a 00000004 00008b1a 80c7a840 7efe67b8
> Jan  4 07:14:46 NetFrog kernel: [21399.528699] fe80: 80c7a840 00000004 a790feec a790fe98 806b59b4 8078613c a790ff14 a790fea8
> Jan  4 07:14:46 NetFrog kernel: [21399.528717] fea0: 6e616c77 00000030 00000000 00000000 7efe67d8 0001000e 00000000 00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528734] fec0: a790ff14 00008b1a 0000012a 7efe67b8 80c7a840 00000004 a790e000 00000036
> Jan  4 07:14:46 NetFrog kernel: [21399.528752] fee0: a790ff0c a790fef0 80671330 806b5924 7efe67b8 abf6eaa0 b67fcd80 00008b1a
> Jan  4 07:14:46 NetFrog kernel: [21399.528769] ff00: a790ff7c a790ff10 8029f060 80671218 807a1834 80145efc 00000128 801081c4
> Jan  4 07:14:46 NetFrog kernel: [21399.528787] ff20: a790e000 00000128 a790ff94 a790ff38 808024bc 397d3000 807a1e98 ba361d40
> Jan  4 07:14:46 NetFrog kernel: [21399.528803] ff40: 00000000 00000000 a790fe88 a790e000 a790e010 b67fcd80 7efe67b8 b67fcd80
> Jan  4 07:14:46 NetFrog kernel: [21399.528820] ff60: 00008b1a 00000004 a790e000 00000036 a790ffa4 a790ff80 8029f7bc 8029efc0
> Jan  4 07:14:46 NetFrog kernel: [21399.528839] ff80: 00000016 0000000e 019773c0 7efe67d8 00000036 801081c4 00000000 a790ffa8
> Jan  4 07:14:46 NetFrog kernel: [21399.528855] ffa0: 80108000 8029f784 0000000e 019773c0 00000004 00008b1a 7efe67b8 00000016
> Jan  4 07:14:46 NetFrog kernel: [21399.528872] ffc0: 0000000e 019773c0 7efe67d8 00000036 01989704 00000000 001bc730 019773d0
> Jan  4 07:14:46 NetFrog kernel: [21399.528889] ffe0: 001bcbac 7efe67b4 00141478 76c7680c 20000010 00000004 00000000 00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528920] [<80785b08>] (call_commit_handler.part.2) from [<80785da4>] (ioctl_standard_call+0x270/0x4bc)
> Jan  4 07:14:46 NetFrog kernel: [21399.528942] [<80785da4>] (ioctl_standard_call) from [<807862e8>] (wext_handle_ioctl+0x1b8/0x20c)
> Jan  4 07:14:46 NetFrog kernel: [21399.528967] [<807862e8>] (wext_handle_ioctl) from [<806b59b4>] (dev_ioctl+0x9c/0x8e8)
> Jan  4 07:14:46 NetFrog kernel: [21399.528992] [<806b59b4>] (dev_ioctl) from [<80671330>] (sock_ioctl+0x124/0x2d8)
> Jan  4 07:14:46 NetFrog kernel: [21399.529016] [<80671330>] (sock_ioctl) from [<8029f060>] (do_vfs_ioctl+0xac/0x7c4)
> Jan  4 07:14:46 NetFrog kernel: [21399.529037] [<8029f060>] (do_vfs_ioctl) from [<8029f7bc>] (SyS_ioctl+0x44/0x6c)
[... cut ...]
  The syslog told kernel Internal error: Oops: 17 [#1] SMP ARM for BCM2835. After that the wpa_supplicant didn't work, the system cpu usage increased gradually which would reach 100 or higher(measured by uptime). And then the linux system didn't work . SSH service couldn't be reach, but icmp ping was still alive.

  I tried the above test several times on the same and different raspberry pi 3B boards. The kernel error problem could be reproduced. The kernel firmware was upgraded to lastest 4.14.90-v7+.

  Any suggestion is welcome. Thank you !

cxueqin, 2019-01-08:
@pelwell I used nodejs to send command to wpa_supplicant ctrl_interface to send wifi test command.  The test logic:
   1.  enable a  specific wifi network 
   2. setNetworkSSID
   3. setNetworkVariable key_mgmt
   4. setNetworkVariable eap  
   5. setNetworkVariable proto WPA RSN
   6. setNetworkVariable pairwise  CCMP
   7. setNetworkVariable group CCMP
   8. setNetworkIdentity 
   9. setNetworkVariable  password
  10. enableNetwork
  11. reassociate

The test logic is repeated every two minutes. Sometimes after hundreds of tests , the kernel gave many error messages, such as
[image]

And in some cases , kernel will failed. 
[image]

The test script and wpa_supplicant config file attached here.
Just unzip it and run node wifi_test.js to reproduce the problem.

[wifi_test.zip](https://github.com/raspberrypi/linux/files/2737469/wifi_test.zip)

Many thanks.

pelwell, 2019-01-08:
Does it fail quicker with a shorter interval?

cxueqin, 2019-01-08:
> Does it fail quicker with a shorter interval?

I will try to do the test with a shorter interval, and tell the result.  

But in syslog , there was an noteable line, "Unable to handle kernel NULL pointer deference at virtual address 000000".   Maybe some bug in wifi driver or other modules cause the problem?

cxueqin, 2019-01-08:
@pelwell ok, thank u for reply. The kernel message file , some kernel log message like this
[image]
Maybe the log message has some relation with the bug.

cxueqin, 2019-01-10:
@pelwell  The wifi kernel crash  could be easily using my test script in one minute interval. I reproduced the crash in two respberry pi 3B board. One kernel version is 4.14.79-v7+ #1159, and the other one is 4.14.90-v7+ #1183.   The crash backstrace is almost same. 

 4.14.79-v7+ #1159 crash log:
[image]

4.14.90-v7 crash log:
[image]

Is there any way to avoid this problem in wifi 802.1x mode? Thanks

pelwell, 2019-01-10:
I've been running a modified version of your script for about an hour now, reconnecting every 20 seconds. In that time the kernel log has accumulated 4 "scan error (-16)"s, and no crashes.

However, the script modification was to switch to WPA-PSK authentation rather than WPA-EAP, which I am unable to test. Are you able to try with WPA-PSK to see if it is more or less reliable?

cxueqin, 2019-01-10:
@pelwell  In my wifi 802.1x test,  sometimes the raspberry pi system was stable for serveral days, sometimes kernel crash in hours.  General speaking, the chance of kernel crash was high in my enviroment.  

Now I am doing some PSK authentation test. I will report the result later. Thanks.

cxueqin, 2019-01-11:
@pelwell   Since we have no idea the reason of the kernel crash ,  could some action to be taken to detect the kernel crash. When kernel crash  is detected, the system will be rebooted automatically to avoid the bad situaction. Any way to achieve that goal? Thanks.
 
The newest kernel crash log with official kernel version  4.14.79-v7+ #1159.

```
Jan 11 01:19:00 NetFrog kernel: [ 8360.000887] Unable to handle kernel NULL pointer dereference at virtual address 00000000
Jan 11 01:19:00 NetFrog kernel: [ 8360.000909] pgd = a7950000
Jan 11 01:19:00 NetFrog kernel: [ 8360.000920] [00000000] *pgd=3708a835, *pte=00000000, *ppte=00000000
Jan 11 01:19:00 NetFrog kernel: [ 8360.000949] Internal error: Oops: 17 [#1] SMP ARM
Jan 11 01:19:00 NetFrog kernel: [ 8360.000960] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
Jan 11 01:19:00 NetFrog kernel: [ 8360.001083] CPU: 2 PID: 802 Comm: wpa_supplicant Tainted: G         C      4.14.79-v7+ #1159
Jan 11 01:19:00 NetFrog kernel: [ 8360.001092] Hardware name: BCM2835
Jan 11 01:19:00 NetFrog kernel: [ 8360.001103] task: b6b03c00 task.stack: a79d4000
Jan 11 01:19:00 NetFrog kernel: [ 8360.001126] PC is at call_commit_handler.part.2+0x18/0x44
Jan 11 01:19:00 NetFrog kernel: [ 8360.001140] LR is at ioctl_standard_call+0x270/0x4bc
Jan 11 01:19:00 NetFrog kernel: [ 8360.001151] pc : [<8078324c>]    lr : [<807834e8>]    psr: 20000013
Jan 11 01:19:00 NetFrog kernel: [ 8360.001162] sp : a79d5de0  ip : a79d5df0  fp : a79d5dec
Jan 11 01:19:00 NetFrog kernel: [ 8360.001173] r10: b6853000  r9 : 00000034  r8 : 00008b1a
Jan 11 01:19:00 NetFrog kernel: [ 8360.001184] r7 : 0000000e  r6 : a79d5ea0  r5 : 00000007  r4 : 00000002
Jan 11 01:19:00 NetFrog kernel: [ 8360.001196] r3 : 00000000  r2 : 60000013  r1 : 60000013  r0 : b6853000
Jan 11 01:19:00 NetFrog kernel: [ 8360.001209] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jan 11 01:19:00 NetFrog kernel: [ 8360.001220] Control: 10c5383d  Table: 2795006a  DAC: 00000055
Jan 11 01:19:00 NetFrog kernel: [ 8360.001232] Process wpa_supplicant (pid: 802, stack limit = 0xa79d4210)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001243] Stack: (0xa79d5de0 to 0xa79d6000)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001261] 5de0: a79d5e5c a79d5df0 807834e8 80783240 806a8be0 a79d5e6c a79d5e6c a79d5eb0
Jan 11 01:19:00 NetFrog kernel: [ 8360.001278] 5e00: 00000000 a783f180 8069591c 636a8bb8 616e6968 6566696c 6669772d 80689569
Jan 11 01:19:00 NetFrog kernel: [ 8360.001295] 5e20: a79d5e44 807a1890 a79d5e4c a79d5e38 807a1890 00008b1a a79d5ea0 80c7a7c0
Jan 11 01:19:00 NetFrog kernel: [ 8360.001312] 5e40: 7e9c77b8 00000000 a79d4000 00000036 a79d5e94 a79d5e60 80783a2c 80783284
Jan 11 01:19:00 NetFrog kernel: [ 8360.001328] 5e60: 7f284c5c 80c7a7c0 7e9c77b8 00008b1a 00000004 00008b1a 80c7a7c0 7e9c77b8
Jan 11 01:19:00 NetFrog kernel: [ 8360.001345] 5e80: 80c7a7c0 00000004 a79d5eec a79d5e98 806b32e8 80783880 a79d5f14 a79d5ea8
Jan 11 01:19:00 NetFrog kernel: [ 8360.001362] 5ea0: 6e616c77 00000030 00000000 00000000 7e9c77d8 0001000e 00000000 00000000
Jan 11 01:19:00 NetFrog kernel: [ 8360.001379] 5ec0: a79d5f14 00008b1a 0000012a 7e9c77b8 80c7a7c0 00000004 a79d4000 00000036
Jan 11 01:19:00 NetFrog kernel: [ 8360.001396] 5ee0: a79d5f0c a79d5ef0 8066ecec 806b3258 7e9c77b8 b8b02e20 b9068c00 00008b1a
Jan 11 01:19:00 NetFrog kernel: [ 8360.001413] 5f00: a79d5f7c a79d5f10 8029ea5c 8066ebd4 8079ef74 80145ec0 00000128 801081c4
Jan 11 01:19:00 NetFrog kernel: [ 8360.001430] 5f20: a79d4000 00000128 a79d5f94 a79d5f38 808024bc 397e4000 8079f5d8 ba372d40
Jan 11 01:19:00 NetFrog kernel: [ 8360.001446] 5f40: 00000000 00000000 a79d5e88 a79d4000 a79d4010 b9068c00 7e9c77b8 b9068c00
Jan 11 01:19:00 NetFrog kernel: [ 8360.001463] 5f60: 00008b1a 00000004 a79d4000 00000036 a79d5fa4 a79d5f80 8029f1b8 8029e9bc
Jan 11 01:19:00 NetFrog kernel: [ 8360.001480] 5f80: 00000016 0000000e 020fe3c0 7e9c77d8 00000036 801081c4 00000000 a79d5fa8
Jan 11 01:19:00 NetFrog kernel: [ 8360.001497] 5fa0: 80108000 8029f180 0000000e 020fe3c0 00000004 00008b1a 7e9c77b8 00000016
Jan 11 01:19:00 NetFrog kernel: [ 8360.001513] 5fc0: 0000000e 020fe3c0 7e9c77d8 00000036 0211cf4c 00000000 001bc730 020fe3d0
Jan 11 01:19:00 NetFrog kernel: [ 8360.001530] 5fe0: 001bcbac 7e9c77b4 00141478 76c6880c 20000010 00000004 00000000 00000000
Jan 11 01:19:00 NetFrog kernel: [ 8360.001559] [<8078324c>] (call_commit_handler.part.2) from [<807834e8>] (ioctl_standard_call+0x270/0x4bc)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001582] [<807834e8>] (ioctl_standard_call) from [<80783a2c>] (wext_handle_ioctl+0x1b8/0x20c)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001605] [<80783a2c>] (wext_handle_ioctl) from [<806b32e8>] (dev_ioctl+0x9c/0x8e8)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001630] [<806b32e8>] (dev_ioctl) from [<8066ecec>] (sock_ioctl+0x124/0x2d8)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001653] [<8066ecec>] (sock_ioctl) from [<8029ea5c>] (do_vfs_ioctl+0xac/0x7c4)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001674] [<8029ea5c>] (do_vfs_ioctl) from [<8029f1b8>] (SyS_ioctl+0x44/0x6c)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001696] [<8029f1b8>] (SyS_ioctl) from [<80108000>] (ret_fast_syscall+0x0/0x28)
Jan 11 01:19:00 NetFrog kernel: [ 8360.001714] Code: e24cb004 e52de004 e8bd4000 e5903120 (e5933000) 
Jan 11 01:19:00 NetFrog kernel: [ 8360.001729] ---[ end trace a931f3365b4fc949 ]---
```

pelwell, 2019-01-23:
We can't fix the kernel bug until we can reproduce it, and so far that hasn't been possible. Your kernel is not a standard kernel, your environment is different, and your use case (repeatedly connecting and disconnecting) is very niche.

JamesH65, 2019-09-19:
@cxueqin Have you tried this with the latest kernel? Does that make any difference?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

cxueqin, 2019-09-21:
> This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

@JamesH65  I haven't test in the lastest kernel. Can you tell me which kernel version I should use for testing? Thanks

JamesH65, 2019-09-21:
Just the latest release one, from apt.

sudo apt update
sudo apt upgrade

jiscfoo, 2019-09-23:
I am seeing a similar WPA-EAP related issue but it's stopping it from connecting rather than causing it to fail later.

This is on a Raspberry Pi 3 running Alpine 3.10 (`Linux xx 4.19.67-0-rpi2 #1-Alpine SMP Tue Aug 20 08:39:08 UTC 2019 armv7l Linux`). I have also installed Wireguard from Alpine packages which taints the kernel.

When running `wpa_supplicant` using a known good WPA-EAP configuration (tested elsewhere) and both `-D wext` and `-D nl80211`, it fails complete an authentication and, eventually, throws the following to `dmesg`:

```
[  +0.301576] ------------[ cut here ]------------
[  +0.000342] WARNING: CPU: 1 PID: 1482 at net/wireless/sme.c:752 __cfg80211_connect_result+0x380/0x3ec [cfg80211]
[  +0.000007] Modules linked in: wireguard(O) ip6_udp_tunnel udp_tunnel ipv6 af_packet brcmfmac brcmutil sha256_generic cfg802
11 rfkill raspberrypi_hwmon hwmon bcm2835_codec(C) v4l2_mem2mem videobuf2_dma_contig bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_
sm_cma(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media fixed uio_pdrv_genirq
uio
[  +0.000146] CPU: 1 PID: 1482 Comm: kworker/u8:2 Tainted: G        WC O      4.19.67-0-rpi2 #1-Alpine
[  +0.000005] Hardware name: BCM2835
[  +0.000288] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  +0.000034] [<801100f4>] (unwind_backtrace) from [<8010c1e4>] (show_stack+0x10/0x14)
[  +0.000018] [<8010c1e4>] (show_stack) from [<80774cb4>] (dump_stack+0xb8/0x108)
[  +0.000018] [<80774cb4>] (dump_stack) from [<8011d4bc>] (__warn.part.3+0xa8/0xd4)
[  +0.000015] [<8011d4bc>] (__warn.part.3) from [<8011d640>] (warn_slowpath_null+0x40/0x4c)
[  +0.000286] [<8011d640>] (warn_slowpath_null) from [<7f134890>] (__cfg80211_connect_result+0x380/0x3ec [cfg80211])
[  +0.000573] [<7f134890>] (__cfg80211_connect_result [cfg80211]) from [<7f106d64>] (cfg80211_process_wdev_events+0xf4/0x150 [
cfg80211])
[  +0.000572] [<7f106d64>] (cfg80211_process_wdev_events [cfg80211]) from [<7f106df0>] (cfg80211_process_rdev_events+0x30/0x88
 [cfg80211])
[  +0.000563] [<7f106df0>] (cfg80211_process_rdev_events [cfg80211]) from [<7f1012a4>] (cfg80211_event_work+0x14/0x1c [cfg8021
1])
[  +0.000295] [<7f1012a4>] (cfg80211_event_work [cfg80211]) from [<80135f7c>] (process_one_work+0x220/0x4ec)
[  +0.000014] [<80135f7c>] (process_one_work) from [<80136ec4>] (worker_thread+0x28/0x5ac)
[  +0.000017] [<80136ec4>] (worker_thread) from [<8013be08>] (kthread+0x138/0x150)
[  +0.000015] [<8013be08>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[  +0.000006] Exception stack(0xb9e2dfb0 to 0xb9e2dff8)
[  +0.000009] dfa0:                                     00000000 00000000 00000000 00000000
[  +0.000011] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  +0.000011] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  +0.000007] ---[ end trace 8b9f861bdf3fb8bb ]---

[ +10.787136] ------------[ cut here ]------------
[  +0.000143] WARNING: CPU: 1 PID: 1558 at net/wireless/sme.c:945 brcmf_notify_roaming_status+0x158/0x1f4 [brcmfmac]
[  +0.000006] Modules linked in: wireguard(O) ip6_udp_tunnel udp_tunnel ipv6 af_packet brcmfmac brcmutil sha256_generic cfg802
11 rfkill raspberrypi_hwmon hwmon bcm2835_codec(C) v4l2_mem2mem videobuf2_dma_contig bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_
sm_cma(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media fixed uio_pdrv_genirq uio
[  +0.000143] CPU: 1 PID: 1558 Comm: kworker/1:0 Tainted: G        WC O      4.19.67-0-rpi2 #1-Alpine
[  +0.000004] Hardware name: BCM2835
[  +0.000116] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[  +0.000031] [<801100f4>] (unwind_backtrace) from [<8010c1e4>] (show_stack+0x10/0x14)
[  +0.000018] [<8010c1e4>] (show_stack) from [<80774cb4>] (dump_stack+0xb8/0x108)
[  +0.000018] [<80774cb4>] (dump_stack) from [<8011d4bc>] (__warn.part.3+0xa8/0xd4)
[  +0.000016] [<8011d4bc>] (__warn.part.3) from [<8011d640>] (warn_slowpath_null+0x40/0x4c)
[  +0.000119] [<8011d640>] (warn_slowpath_null) from [<7f1dfb90>] (brcmf_notify_roaming_status+0x158/0x1f4 [brcmfmac])
[  +0.000224] [<7f1dfb90>] (brcmf_notify_roaming_status [brcmfmac]) from [<7f1e8e50>] (brcmf_fweh_event_worker+0x1f8/0x498 [brcmfmac])
[  +0.000123] [<7f1e8e50>] (brcmf_fweh_event_worker [brcmfmac]) from [<80135f7c>] (process_one_work+0x220/0x4ec)
[  +0.000014] [<80135f7c>] (process_one_work) from [<80136ec4>] (worker_thread+0x28/0x5ac)
[  +0.000017] [<80136ec4>] (worker_thread) from [<8013be08>] (kthread+0x138/0x150)
[  +0.000016] [<8013be08>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[  +0.000006] Exception stack(0xb9703fb0 to 0xb9703ff8)
[  +0.000009] 3fa0:                                     00000000 00000000 00000000 00000000
[  +0.000011] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  +0.000009] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  +0.000007] ---[ end trace 8b9f861bdf3fb8bc ]---
```
Output from `wpa_supplicant`:

```
# wpa_supplicant -D wext -i wlan0 -c roamtest.conf  -t
1569242739.647865: Successfully initialized wpa_supplicant
1569242740.332581: wlan0: Trying to associate with xx:xx:xx:xx:xx:xx (SSID='xx' freq=2462 MHz)
1569242740.332799: Failed to add supported operating classes IE
1569242740.481315: wlan0: Associated with xx:xx:xx:xx:xx:xx
1569242740.481560: wlan0: CTRL-EVENT-EAP-STARTED EAP authentication started
1569242810.482272: wlan0: Authentication with xx:xx:xx:xx:xx:xx timed out.
1569242810.491701: wlan0: CTRL-EVENT-DISCONNECTED bssid=xx:xx:xx:xx:xx:xx reason=3 locally_generated=1
1569242810.592117: ioctl[SIOCSIWSCAN]: Resource temporarily unavailable
1569242810.592229: wlan0: CTRL-EVENT-SCAN-FAILED ret=-1 retry=1
```

chrispollitt, 2019-09-30:
I am also having this issue.

spaetow, 2021-02-15:
I have been able to reproduce this today. As with the OP, this effectively causes the RPi to stop responding to reboots (well, it shuts down but then hangs and a hard reset is required). 

```Feb 15 22:19:54 raspberrypi kernel: [   67.564580] Unable to handle kernel NULL pointer dereference at virtual address 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.564603] pgd = 39a51f6e
Feb 15 22:19:54 raspberrypi kernel: [   67.564622] [00000000] *pgd=06eba835, *pte=00000000, *ppte=00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.564679] Internal error: Oops: 17 [#1] SMP ARM
Feb 15 22:19:54 raspberrypi kernel: [   67.564692] Modules linked in: rfcomm fuse bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc nft_counter nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink joydev brcmfmac brcmutil sha256_generic cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) bcm2835_v4l2(C) snd_bcm2835(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer videodev snd vc_sm_cma(C) mc uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
Feb 15 22:19:54 raspberrypi kernel: [   67.565022] CPU: 2 PID: 503 Comm: hostapd Tainted: G         C        5.10.11-v7+ #1399
Feb 15 22:19:54 raspberrypi kernel: [   67.565033] Hardware name: BCM2835
Feb 15 22:19:54 raspberrypi kernel: [   67.565209] PC is at brcmf_p2p_send_action_frame+0x288/0xa60 [brcmfmac]
Feb 15 22:19:54 raspberrypi kernel: [   67.565368] LR is at brcmf_p2p_send_action_frame+0x270/0xa60 [brcmfmac]
Feb 15 22:19:54 raspberrypi kernel: [   67.565382] pc : [<7f470f34>]    lr : [<7f470f1c>]    psr: 80000013
Feb 15 22:19:54 raspberrypi kernel: [   67.565394] sp : 85bf9a88  ip : 00000000  fp : 85bf9aec
Feb 15 22:19:54 raspberrypi kernel: [   67.565406] r10: 85ae04e0  r9 : 00000004  r8 : 7f48b6a8
Feb 15 22:19:54 raspberrypi kernel: [   67.565419] r7 : 00000000  r6 : 8401500c  r5 : 858a1000  r4 : 84015000
Feb 15 22:19:54 raspberrypi kernel: [   67.565432] r3 : 00000000  r2 : 858a1000  r1 : 7f492504  r0 : 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565447] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Feb 15 22:19:54 raspberrypi kernel: [   67.565461] Control: 10c5383d  Table: 05bfc06a  DAC: 00000055
Feb 15 22:19:54 raspberrypi kernel: [   67.565476] Process hostapd (pid: 503, stack limit = 0x2963c431)
Feb 15 22:19:54 raspberrypi kernel: [   67.565489] Stack: (0x85bf9a88 to 0x85bfa000)
Feb 15 22:19:54 raspberrypi kernel: [   67.565507] 9a80:                   00000000 00000000 80e057c8 85ae04e0 84014d80 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565527] 9aa0: ffffffff 7f492538 7f492384 7f492504 7f491b48 ffffa52f 80e03d00 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565546] 9ac0: 85bf9aec 80e05008 0000006d 858a1000 9303002c 85bf9b80 84015000 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565564] 9ae0: 85bf9b44 85bf9af0 7f466980 7f470cb8 00000000 00000000 0000006d 0000099e
Feb 15 22:19:54 raspberrypi kernel: [   67.565583] 9b00: 7f324110 85ae04e0 86d00004 00000085 0000099e a1908b93 808d4454 7f4667fc
Feb 15 22:19:54 raspberrypi kernel: [   67.565602] 9b20: 85ae01c0 85ae0000 85bf9bac 85bf9b80 83a8db40 84e9f014 85bf9b74 85bf9b48
Feb 15 22:19:54 raspberrypi kernel: [   67.565622] 9b40: 7f2eb5a0 7f466808 808d9784 808d4424 85bf9c54 80e05008 85ae0000 86d00004
Feb 15 22:19:54 raspberrypi kernel: [   67.565640] 9b60: 00000000 83a8db40 85bf9bfc 85bf9b78 7f2e1c3c 7f2eb498 00000000 0000003b
Feb 15 22:19:54 raspberrypi kernel: [   67.565660] 9b80: 00000000 00000000 809e5c7c 80196f54 84019e30 00000000 0000099e 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565680] 9ba0: 00000000 00000000 00000000 84019e30 00000001 00000000 9303002c 00000085
Feb 15 22:19:54 raspberrypi kernel: [   67.565699] 9bc0: 00000000 00000000 00000000 a1908b93 85bf9cb0 00000000 7f324110 80e05008
Feb 15 22:19:54 raspberrypi kernel: [   67.565718] 9be0: 858a5000 83a8dc00 80f16400 85bf9cb0 85bf9cac 85bf9c00 808db758 7f2e1a0c
Feb 15 22:19:54 raspberrypi kernel: [   67.565736] 9c00: 00000001 809eb3b4 00000000 85bf9c18 00000000 80e05008 00000000 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565755] 9c20: 00000000 85bf9c24 85bf9c24 a1908b93 0000173f 8ca790c0 7f2e1a00 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565774] 9c40: 00000000 00000000 7f31a3cc 00000129 03101c3b 602af30f 2f0001ec 93030000
Feb 15 22:19:54 raspberrypi kernel: [   67.565793] 9c60: 93030010 93030014 858a5000 80f16400 85ae0000 86d00004 85bf9cb0 a1908b93
Feb 15 22:19:54 raspberrypi kernel: [   67.565813] 9c80: 85bf9cbc 83a8dc00 80e05008 808db598 93030000 000000b4 00000000 80e05008
Feb 15 22:19:54 raspberrypi kernel: [   67.565832] 9ca0: 85bf9cfc 85bf9cb0 808d903c 808db5a4 00000000 00000000 00000000 00000000
Feb 15 22:19:54 raspberrypi kernel: [   67.565850] 9cc0: 00000000 00000000 00000000 00000000 00000000 a1908b93 85bf9cfc 80f1cf54
Feb 15 22:19:54 raspberrypi kernel: [   67.565869] 9ce0: 83a8dc00 000000b4 83a8dc00 84efe184 85bf9d14 85bf9d00 808d98c8 808d8f80
Feb 15 22:19:54 raspberrypi kernel: [   67.565888] 9d00: 816b8000 84efe000 85bf9d4c 85bf9d18 808d86d4 808d98a0 7fffffff a1908b93
Feb 15 22:19:54 raspberrypi kernel: [   67.565907] 9d20: 000001c0 00000008 85bf9f40 80e05008 83a8dc00 84efe000 000000b4 00000000
[... cut ...]
Feb 15 22:19:54 raspberrypi kernel: [   67.568476] ---[ end trace 8d5fba1f93e042f8 ]---```

I hope this is helpful. This is on ```PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"```

Kernel version is `Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux`

spaetow, 2021-02-15:
As an addendum: When the line `hs20=1` is commented out, the problem no longer occurs. So it appears that the firmware as provided by Broadcom/Cypress does *not* support Hotspot 2.0/Passpoint, or at least freaks out substantially to cause a kernel hang. I'll have to check this on Fedora as a comparison.